open-sdr/openwifi #4: monitor mode

bd467913, 2019-12-26:
Hi
I'm trying to set monitor mode.
I used the 'user_space' _monitor_ch.sh_ but without success.

thanks

JiaoXianjun, 2019-12-26:
Could you show me the exact command with arguments and outputs?

You could also use:
iwconfig sdr0 mode monitor

bd467913, 2019-12-26:
this is working
but after i am getting the sdr0 up all is stack

ip link set sdr0 down
iwconfig sdr0 mode monitor
ip link set sdr0 up
**after this command the broad is stuck and i have to restart it**

JiaoXianjun, 2019-12-26:
Can you set a channel in 5GHz to listen? Such as, 36, 44, 48?
iwconfig sdr0 channel 44

Current version might have some issue in 2.4GHz because SIFS in 2.4GHz is shorter. We are working on it. There will be new version soon.

bd467913, 2019-12-26:
OK for 5Ghz its looks ok
so i will wait for the 2.4Ghz
Thanks for the help

bd467913, 2019-12-26:
i getting stack again in when it get to set the sdr0 up

JiaoXianjun, 2019-12-26:
So, for monitor mode in 5GHz you never run it successfully? (Stuck immediately or after a while?). You are using the original version without modifying anything?

hostapd is successful?

JiaoXianjun, 2019-12-26:
Before setting it to monitor mode, could you please run:
service network-manager stop

bd467913, 2019-12-26:
Ok 
hostapd with the file configuration attached is working and i see the network
when run the commands:

root@analog:~/openwifi# service network-manager stop
network-manager stop/waiting
root@analog:~/openwifi# sudo ip link set sdr0 down
root@analog:~/openwifi# sudo iwconfig sdr0 mode monitor
root@analog:~/openwifi# sudo ip link set sdr0 up
root@analog:~/openwifi# sudo iwconfig sdr0 channel 44

this gets stuck

JiaoXianjun, 2019-12-26:
Thanks. We will take a look ASAP after holiday. 

How about this sequence?
wgd.sh
iwlist sdr0 scan
iwconfig sdr0 channel 44
iwconfig sdr0 mode monitor

bd467913, 2019-12-26:
OK
but you have to set sdr0 down, before setting the monitor mode

JiaoXianjun, 2020-01-03:
Hello, I have fixed the monitor mode bug in the driver sdr.c. My test is OK.

Would you please just pull and regenerate sdr.ko by the make_all.sh script and see whether it is fixed at your side?

Thanks!

bd467913, 2020-01-05:
Hello, I have checked and it looks better for the 5Ghz. 
About the 2.4Ghz still sum problems, buy i understand that will be fix for it.

Also i checked and not all 5Ghz frequencies also not supported. Is there any chance to add more frequencies?

JiaoXianjun, 2020-01-05:
Hi, for 2.4GHz monitor mode, actually it is related to another known issue that has been fixed internally. We will have some commits next week to fix it. After that, monitor mode should work for all bands.

For more 5GHz channels, could you please create another issue? I will close this one.

closed 2020-01-05 by JiaoXianjun

bd467913, 2020-01-05:
Yes
Thank you for the quick response