open-sdr/openwifi #5: About the error of "Run Linux and do some post-config"

legendof1900, 2020-01-02:
As shown in the figure, when I do the process of "Run Linux and do some post-config", I meet with a problem about the command of "modprobe mac80211".
[image]

JiaoXianjun, 2020-01-02:
Can you show me the content of directory of /lib/modules/ and the content of its sub-directories?

In README, there is a step:

sudo mkdir $SDCARD_DIR/rootfs/lib/modules/4.14.0-g4220d5d24c6c

So, the directory name should be like "4.14.0-g4220d5d24c6c", not "4.14.0".

Make sure you have mkdir with the correct name.

legendof1900, 2020-01-02:
I use the command "ls". The sub-directories are 4.14.0-g4220d5d and 4.14.0-g4220d5d24c6c

JiaoXianjun, 2020-01-02:
Seems like your environment generated directory name is 4.14.0 instead of names (4.14.0-g4220d5d or 4.14.0-g4220d5d24c6c) found by us.
Anyway, the basic principle is creating (or make symbol link) the directory that modprobe asked, and put all files from lib/modules/4.14.0-g4220d5d24c6c into that directory. In your case, it is directory 4.14.0.

You can also do this by the step:

ln -s /lib/modules/4.14.0-g4220d5d24c6c /lib/modules/4.14.0

legendof1900, 2020-01-02:
Thank you. I solve the problem.

closed 2020-01-02 by JiaoXianjun

JiaoXianjun, 2020-07-31:
Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email? If you could also introduce yourself bit (student? Company? University?), that would be perfect. We might send out some questions to listen for user feed feedback.

Thanks.