open-sdr/openwifi #6: 5Ghz frequency

bd467913, 2020-01-05:
Hello,
I checked the frequency and there are 25 channels until now.
sdr0      25 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 14 : 2.484 GHz
          Channel 36 : 5.18 GHz
          Channel 38 : 5.19 GHz
          Channel 40 : 5.2 GHz
          Channel 42 : 5.21 GHz
          Channel 44 : 5.22 GHz
          Channel 46 : 5.23 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Current Frequency:5.22 GHz (Channel 44)

How can i add more frequencies to the 5Ghz?

JiaoXianjun, 2020-01-05:
regulation and frequency config section is added to the document. Would you please try to change/extend the channel by yourself? Thanks!

bd467913, 2020-01-06:
OK
thank you

JiaoXianjun, 2020-01-07:
Hello,

We just update the openwifi (openwifi-hw as well). Now the 2.4GHz monitor mode shouldn't have any issue according to our test. Please pull them. (Do the pull of openwifi-hw as submodule inside openwifi to use correct FPGA image in openwifi compiling procedure).

Then, you need to re-generate BOOT.BIN and put BOOT.BIN into the boot partition of SD card. Also please re-compile all drivers and put them into rootfs of SD card. Please find related section in README. 

If it works after update at your side, would you please let us know?

Thanks!

closed 2020-03-06 by JiaoXianjun

JiaoXianjun, 2020-07-31:
Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email? If you could also introduce yourself bit (student? Company? University?), that would be perfect. We might send out some questions to listen for user feed feedback.

Thanks.