open-sdr/openwifi #7: pre-built image openwifi-1.0.0-ghent.img on zc706 with  AD-FMCOMMS3-EBZ

dzoran1, 2020-01-08:
Problem to start pre-built image from  https://users.ugent.be/~xjiao/openwifi-1.0.0-ghent.zip
SDCARD is created using 
sudo dd bs=4M if=openwifi-1.0.0-ghent.img of=/dev/sdc

console out is:

root@analog:~# 
root@analog:~# 
root@analog:~# 
root@analog:~# cd openwifi/
root@analog:~/openwifi# service network-manager stop
network-manager stop/waiting
root@analog:~/openwifi# ./wgd.sh 
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
set RF frontend: ant0 rx, ant1 tx
rx bw
18000000
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
ad9361 spi0.0: ad9361_dig_tune_delay: Tuning RX FAILED!
ad9361 spi0.0: Calibration TIMEOUT (0x16, 0x10)
ad9361 spi0.0: Calibration TIMEOUT (0x16, 0x10)
ad9361 spi0.0: Calibration TIMEOUT (0x16, 0x10)
sh: echo: I/O error
17500000
tx_bw
18000000
37500000
rx rate
61440000
ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
40000000
tx rate
40000000
40000000
rx lo
2400000000
5240000000
tx lo
2450000000
5249999998
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
rx0 gain to 70
59.000000 dB
sh: echo: I/O error
59.000000 dB
rx1 gain to 70
45.000000 dB
sh: echo: I/O error
45.000000 dB
tx0 gain -89dB
[... 458 lines cut ...]
3ec0: 0006efdc 00000000 ef1f3f48 00000000 00000000 00000000 00000000 00000000
3ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
3f00: 00000000 00000000 7fffffff 00000000 00000003 0045d008 0000017b c0107b04
3f20: ef1f2000 00000000 bed90848 c0187dcc 7fffffff 00000000 00000003 00000000
3f40: 00000000 f29b1000 0006efdc 00000000 f29b5b66 f29b1000 0006efdc f2a1f794
3f60: f2a1f58c f2a051a4 00008000 00008330 00000000 00000000 00000000 00003600
3f80: 00000033 00000034 0000001e 00000018 00000014 00000000 b6e961e8 0000000b
3fa0: 0c05e300 c0107940 b6e961e8 0000000b 00000003 0045d008 00000000 bed90848
3fc0: b6e961e8 0000000b 0c05e300 0000017b 0045dda8 00000000 0045d008 bed90848
3fe0: bed90670 bed90660 00448177 b6f1f2b2 80070030 00000003 00000000 00000000
[<bf12bba4>] (ad9361_spi_to_phy [ad9361_drv]) from [<bf168ad8>] (openwifi_dev_probe+0xb8/0x718 [sdr])
[<bf168ad8>] (openwifi_dev_probe [sdr]) from [<c03b2880>] (platform_drv_probe+0x50/0xb0)
[<c03b2880>] (platform_drv_probe) from [<c03b0fc4>] (driver_probe_device+0x234/0x2e8)
[<c03b0fc4>] (driver_probe_device) from [<c03b1130>] (__driver_attach+0xb8/0xbc)
[<c03b1130>] (__driver_attach) from [<c03af52c>] (bus_for_each_dev+0x68/0x9c)
[<c03af52c>] (bus_for_each_dev) from [<c03b0484>] (bus_add_driver+0x100/0x20c)
[<c03b0484>] (bus_add_driver) from [<c03b19f4>] (driver_register+0x78/0xf4)
[<c03b19f4>] (driver_register) from [<c0101a20>] (do_one_initcall+0x44/0x168)
[<c0101a20>] (do_one_initcall) from [<c0188708>] (do_init_module+0x60/0x1f0)
[<c0188708>] (do_init_module) from [<c01875a8>] (load_module+0x1de8/0x23fc)
[<c01875a8>] (load_module) from [<c0187dcc>] (SyS_finit_module+0xa8/0xb8)
[<c0187dcc>] (SyS_finit_module) from [<c0107940>] (ret_fast_syscall+0x0/0x48)
Code: e58d3040 e300123a eafffed9 e5903054 (e5930018) 
---[ end trace dad18458bfe54443 ]---
root@analog:~/openwifi#

JiaoXianjun, 2020-01-09:
Since we don't have fmcomms3, hard to reproduce here.
The first error is: ad9361 spi0.0: Calibration TIMEOUT (0x5E, 0x80)
It is printed by ad9361_check_cal_done. 0x5E means REG_CH_1_OVERFLOW in ad9361_regs.h.

Could you do two things for the debug?
1. terminate all RF port to make sure no signal present there.
2. Run some example of Analog Devices basic image (http://swdownloads.analog.com/cse/2017_R1-2018_01_29.img.xz  at  https://wiki.analog.com/resources/tools-software/linux-software/zynq_images) to see whether ad9361 can be initialized successfully. If the basic image from ADI also has the issue, it would most probably to be with hardware. If there is no issue, I will check our modifications on ADI's basic image.

Thanks!

dzoran1, 2020-01-09:
We found HW related problem. Engineer who worked with AD-FMCOMMS3-EBZ removed Cristal oscillator Y101 and worked with external clock. He forget to inform me.  I suppose that everything will be OK when we provide appropriate clock.

JiaoXianjun, 2020-01-09:
Thanks for the update.
After you fix it and run the openwifi successfully, would you please tell us here.

dzoran1, 2020-01-09:
I will try to fix this if I find appropriate oscillator - originals are in another town - i need a few days to be there. will report about results...

JiaoXianjun, 2020-01-09:
Take your time.

Or you could make a quick test if you still have the external clock source.

Best regards,
--
Xianjun Jiao
Senior researcher, imec-Gent university
210.007, IDLab, iGent, Technologiepark 126, 9052, Ghent, Belgium
Mobile: +32 484 57 22 32<tel:+32%20484%2057%2022%2032>

dzoran1, 2020-01-13:
We made test with appropriate external clock source 40MHz without oscillator Y101 but with soldered C113 and ZC706 with AD-FMCOMMS3-EBZ with prebuilt image works perfect...  :)
After soldering original Y101 and removing C113 system works perfect again …  :)
Thank you so much for help...

JiaoXianjun, 2020-01-13:
Good news! Have fun. 

We close this issue then.

closed 2020-01-13 by JiaoXianjun

JiaoXianjun, 2020-07-31:
Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email (if you could also introduce yourself bit, that would be perfect). We might send out some questions to listen for user feed feedback.

Thanks.