open-sdr/openwifi #12: fail to run fosdem.sh

Leo-Cheung-CUHK, 2020-03-29:
I'm following the quick start and meet the error shown as below when running ./fosdem.sh

xilinx-vdma 43000000.axivdma: failed to get axi_aclk (-2)
xilinx-vdma: probe of 43000000.axivdma failed with error -2
SIOCSIFADDR: No such device
sdr0: ERROR while getting interface flags: No such device
SIOCADDRT: File exists
stop: Unknown instance: 
isc-dhcp-server start/running, process 3319
Configuration file: hostapd-openwifi.conf
Could not read interface sdr0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
Could not read interface sdr0 flags: No such device
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface sdr0 wasn't started

Also, I meet some "error message" when running dmesg command:

Booting Linux on physical CPU 0x0
Linux version 4.14.0-g4220d5d24c6c (jxj@jxj-xps) (gcc version 6.2.1 20161114 (Linaro GCC Snapshot 6.2-2016.11)) #1 SMP PREEMPT Mon Mar 2 19:20:30 CET 2020
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Xilinx Zynq ZC706
bootconsole [earlycon0] enabled
Memory policy: Data cache writealloc
cma: Reserved 128 MiB at 0x38000000
On node 0 totalpages: 262144
free_area_init_node: node 0, pgdat c0c4fd80, node_mem_map ef7f7000
  Normal zone: 1536 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 196608 pages, LIFO batch:31
  HighMem zone: 65536 pages, LIFO batch:15
random: fast init done
percpu: Embedded 16 pages/cpu @ef7c7000 s33036 r8192 d24308 u65536
pcpu-alloc: s33036 r8192 d24308 u65536 alloc=16*4096
pcpu-alloc: [0] 0 [0] 1
Built 1 zonelists, mobility grouping on.  Total pages: 260608
Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 rootwait
PID hash table entries: 4096 (order: 2, 16384 bytes)
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Memory: 895996K/1048576K available (7168K kernel code, 323K rwdata, 2364K rodata, 1024K init, 166K bss, 21508K reserved, 131072K cma-reserved, 131072K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
    lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc0800000   (8160 kB)
      .init : 0xc0b00000 - 0xc0c00000   (1024 kB)
      .data : 0xc0c00000 - 0xc0c50c20   ( 324 kB)
       .bss : 0xc0c56668 - 0xc0c7ff3c   ( 167 kB)
Preemptible hierarchical RCU implementation.
        RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
        Tasks RCU enabled.
RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
efuse mapped to f0802000
slcr mapped to f0804000
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
zynq_clock_init: clkc starts at f0804100
[... 285 lines cut ...]
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4491) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4501) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4511) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4521) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4531) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4541) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4551) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4561) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4571) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (4581) terminated with status 1
init: isc-dhcp-server respawning too fast, stopped
xilinx-vdma 43000000.axivdma: failed to get axi_aclk (-2)
xilinx-vdma: probe of 43000000.axivdma failed with error -2
xilinx-vdma 43000000.axivdma: failed to get axi_aclk (-2)
xilinx-vdma: probe of 43000000.axivdma failed with error -2

JiaoXianjun, 2020-03-29:
Did you do this:

overwrite the files in the base directory with the files in board_name directory of BOOT partiton.

Leo-Cheung-CUHK, 2020-03-29:
Yes, I've updated the hardware image following the section “Update FPGA” using BOARD_NAME=zc806_fmcs2. But I'm using zc706 with AD-FMCOMMS3-EBZ. I wonder whether it is because of it.

JiaoXianjun, 2020-03-29:
Please do not do “ Update FPGA”. This section is not part of quick start. It is only needed when you modify the FPGA design. If you only do the quick start (flash SD card, copy files in BOOT partition), was it successful?

Leo-Cheung-CUHK, 2020-03-29:
Thanks for the quick response.
I just flash SD card by using the restoring function provided in Disk manager (the image file is openwifi-1.1.0-taiyuan.img.xz )and copy the file  BOOT.bin and  devicetrrr.dtb under the folder zynq-zc706-adv7511-fmcomms2-3. And I still meet the problem as I run ./fosdem.sh

oot@analog:~# ~/openwifi/fosdem.sh 
hostapd: no process found
network-manager stop/waiting
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
99.75 dB
113.25 dB
rx0 gain to 70
37.000000 dB
sh: echo: I/O error
37.000000 dB
rx1 gain to 70
50.000000 dB
sh: echo: I/O error
50.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module rx_intf is not currently loaded
check rx_intf module is loaded or not
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
check openofdm_tx module is loaded or not
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
check openofdm_rx module is loaded or not
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
check xpu module is loaded or not
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
insert sdr.ko test_mode=0
[... 248 lines cut ...]
init: isc-dhcp-server main process (2536) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2547) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2558) terminated with status 1
init: isc-dhcp-server respawning too fast, stopped
xilinx-vdma 43000000.axivdma: failed to get axi_aclk (-2)
xilinx-vdma: probe of 43000000.axivdma failed with error -2
tx_intf: loading out-of-tree module taints kernel.
ad9361 spi0.0: ad9361_probe : enter (ad9361)
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.00.b) at 0x79020000 mapped to 0xf2820000, probed ADC AD9361 as MASTER
cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER (9.00.b) at 0x79024000 mapped to 0xf281c000, probed DDS AD9361
sdr: no symbol version for openofdm_rx_api
SIOCSIFADDR: No such device
sdr0: ERROR while getting interface flags: No such device
SIOCADDRT: File exists
stop: Unknown instance: 
isc-dhcp-server start/running, process 3217
Configuration file: hostapd-openwifi.conf
Could not read interface sdr0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
Could not read interface sdr0 flags: No such device
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface sdr0 wasn't started

JiaoXianjun, 2020-03-29:
The board name directory should be “zc706_fmcs2”.  Not the analog devices defined one. Openwifi has its own definition.

Leo-Cheung-CUHK, 2020-03-29:
My fault.  
Great project and Thanks for your help.

JiaoXianjun, 2020-03-29:
You have run the demo successfully?

Leo-Cheung-CUHK, 2020-03-29:
Yes, thanks for your help

JiaoXianjun, 2020-03-29:
Good to know. Enjoy. I close this issue.

closed 2020-03-29 by JiaoXianjun