open-sdr/openwifi #13: Kernel Version may mismatch

littlelid, 2020-04-09:
When running the file `post_config.sh` in my board, the shell yields
```
++ uname -r
+ '[' -d /lib/modules/4.9.0-g0fa1a63 ']'
++ uname -r
+ echo '/lib/modules/4.9.0-g0fa1a63 already exists.'
/lib/modules/4.9.0-g0fa1a63 already exists.
+ depmod
depmod: WARNING: could not open /lib/modules/4.9.0-g0fa1a63/modules.order: No such file or directory
depmod: WARNING: could not open /lib/modules/4.9.0-g0fa1a63/modules.builtin: No such file or directory
+ modprobe mac80211
modprobe: ERROR: could not insert 'mac80211': Exec format error
```
So I check the kernel verison by typing `uname -r`:
```
4.9.0-g0fa1a63
```
Then I switch to the `/lib/modules/4.9.0-g0fa1a63`, and check the `modinfo cfg80211 `:
```
filename:       /lib/modules/4.9.0-g0fa1a63/8021q.ko
version:        1.8
license:        GPL
alias:          rtnl-link-vlan
srcversion:     8F7369A60FF4BAC87D0D0DF
depends:        
intree:         Y
name:           8021q
vermagic:       4.14.0-g4220d5d24c6c-dirty SMP preempt mod_unload modversions ARMv7 p2v8 
root@analog:/lib/modules/4.9.0-g0fa1a63# modinfo cfg80211.ko
filename:       /lib/modules/4.9.0-g0fa1a63/cfg80211.ko
alias:          net-pf-16-proto-16-family-nl80211
description:    wireless configuration support
license:        GPL
author:         Johannes Berg
depends:        
intree:         Y
name:           cfg80211
vermagic:       4.14.0-g4220d5d24c6c-dirty SMP preempt mod_unload modversions ARMv7 p2v8 
parm:           bss_entries_limit:limit to number of scan BSS entries (per wiphy, default 1000) (int)
parm:           ieee80211_regdom:IEEE 802.11 regulatory domain code (charp)
parm:           cfg80211_disable_40mhz_24ghz:Disable 40MHz support in the 2.4GHz band (bool)
```
The `vermagic` above indicates that the module mac80211.ko  is complied with the `4.14.0-g4220d5d24c6c-dirty` version of kernel, which is mismatch with the current version `4.9.0-g0fa1a63`. 

I believe that I stick to the README, and I have tried two ways to burn my SD-Card, using the pre-build image and self-compiled image, and both failed.  So I doubt there is something wrong with the kernel version.

JiaoXianjun, 2020-04-09:
Hi,

Which SD card image file are you using? If you use the one we mentioned in the README: https://users.ugent.be/~xjiao/openwifi-1.1.0-taiyuan-1.img.xz

When you first-time run post_config.sh, there shouldn't be a directory like: /lib/modules/4.9.0-g0fa1a63

Maybe you have some steps mistaken?

Would you please just follow the Quick start section of readme? Or would you please tell me your whole steps from scratch?

littlelid, 2020-04-09:
Thanks for you quick response.
I figure it out !
There is an `uImage` file in the root of BOOT partition of the SD-card. And I replaced it with the `BOOT/zynq-common/uImage` as described in `BOOT/README`:
```
For the zynq projects - copy target/BOOT.BIN, target/devicetree.dtb
and zynq-common/uImage to the root of the BOOT FAT32 partition.
```

The solution is just not to replace the uImage. Maybe you need to particularly point it out.

JiaoXianjun, 2020-04-09:
That BOOT/READMEis from analog devices not from us.
But good to mention this potential issue, or we just remove the original README.

Thanks for the advice. 

I close this issue.

closed 2020-04-09 by JiaoXianjun