open-sdr/openwifi #16: sdr: no symbol version for openofdm_rx_api

skinnym, 2020-05-11:
Hello,

First I want to say thank you so much for all the effort and time you have put into this project! It is truly (at least in my eyes) a major milestone, and you should be proud of yourself... It isn't easy... I know. (at least dsp; fpga; digitial comms; etc; isn't easy for me..., but I am trying to learn :) )

I have a Xilinx ZC706 dev board + FMCOMMS2/3/4. I setup the SD-card as stated in your quick-install section. I can ssh into the system. When I try to run ~/openwifi/fosdem.sh the following output occurs:

root@analog:~# ~/openwifi/fosdem.sh
hostapd: no process found
network-manager stop/waiting
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
117.00 dB
112.50 dB
rx0 gain to 70
56.000000 dB
sh: echo: I/O error
56.000000 dB
rx1 gain to 70
50.000000 dB
sh: echo: I/O error
50.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module rx_intf is not currently loaded
check rx_intf module is loaded or not
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
check openofdm_tx module is loaded or not
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
check openofdm_rx module is loaded or not
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
check xpu module is loaded or not
[... 266 lines cut ...]
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0xf2868000, probed ADC AD9361 as MASTER
cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER (9.01.b) at 0x79024000 mapped to 0xf2862000, probed DDS AD9361
random: crng init done
sdr: no symbol version for openofdm_rx_api
SIOCSIFADDR: No such device
sdr0: ERROR while getting interface flags: No such device
stop: Unknown instance:
isc-dhcp-server start/running, process 2343
Configuration file: hostapd-openwifi.conf
Could not read interface sdr0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
Could not read interface sdr0 flags: No such device
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface sdr0 wasn't started_

----

I believe the error has to do with the "sdr: no symbol version for openofdm_rx_api" print output, but I am not for certain. It appears the ad9361 is loading successfully. But it appears to me the openofdm_rx_api driver is having issues which is causing the sdr device driver to not function properly (that is my interpretation).

Any help you can provide would be greatly appreciated.
Thank you for your time and for making this project possible.

Best,
-matt

JiaoXianjun, 2020-05-11:
Hi,

Thanks for giving it a shot.
The issue only happens once? Or you can reproduce it every time? (power cycle the board, and try again).
After power cycle, can you show me the output of ./wgd.sh?
In my memory, it could happen occasionally, rarely.

skinnym, 2020-05-11:
Hello,

Thank you for the super quick response!

The issue happens after power cycling every time I have tried it. So it is very reproducible. 

I did what I think you asked me to do. The following covers the output of these tests:

- Power cycled and ran: ~/openwifi/fosdem.sh
- Powered cycled again and ran: ./wgd.sh
- Didn't power cycle and ran  ~/openwifi/fosdem.sh followed by ./wgd.sh

Thanks again for your help and all your work!
Best,
-Matt

-------

Booted up and ran the following:
--------------------------------

root@analog:~/openwifi# ~/openwifi/fosdem.sh
hostapd: no process found
network-manager stop/waiting
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
119.75 dB
113.25 dB
rx0 gain to 70
61.000000 dB
sh: echo: I/O error
61.000000 dB
rx1 gain to 70
50.000000 dB
sh: echo: I/O error
50.000000 dB
tx0 gain -89dB
-10.000000 dB
[... 1314 lines cut ...]
init: isc-dhcp-server main process (2473) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2483) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2493) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2503) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2513) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2523) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2533) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2543) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2553) terminated with status 1
init: isc-dhcp-server main process ended, respawning
init: isc-dhcp-server main process (2563) terminated with status 1
init: isc-dhcp-server respawning too fast, stopped
root@analog:~/openwifi#

- 

-

uwbtag, 2020-05-11:
How about use modprobe to load openofdm_rx?

Could you tell me how do you do the sdcard flashing?

uwbtag, 2020-05-11:
Also modprobe to load sdr.ko

JiaoXianjun, 2020-05-11:
Could you please tell me how did you flash the SD card?

skinnym, 2020-05-11:
Sure,

I downloaded the 32 bit image sd-card for zc706_fmcs2:
       https://users.ugent.be/~xjiao/openwifi-1.1.0-taiyuan-2-32bit.img.xz 

I then extracted it to the full ~7.7 GB img file. I then used win32diskimager to write it to a sd-card.
I then followed the steps on the initial setup involving copying the openwifi/board_name (the one for zc706_fmcs2 in my case) to the base BOOT partition. I then copied openwifi/zynq-common/uImage to the base directory of the BOOT partition.

Hope that helps.
Again thanks for your help and time!

Best,
-Matt

JiaoXianjun, 2020-05-11:
The procedure seems ok except that I never try to flash a SD card in windows....

If you have Linux, could you try to flash it in Linux?

You can also try to build the sd card in Linux by my script.

skinnym, 2020-05-12:
Hello,

I made some progress with your help. First I learned my sd-card I was using was having write/read errors. I swapped it out with a fresh one. Furthermore I utilized a piece of software called balenaEtcher (compatible with Linux and windows) to burn the image file to the new sd-card. Note: I originally used dd to image the device, but had uncertainty about the blocksize to use and such so I just went with Etcher and it wrote the sdcard and verified the contents for me. 

Another mistake I had been making was I had been copying the files from "zynq-zc706-adv7511-fmcomms2-3" directory rather than the correct directory "openwifi/zc706_fmcs2" to the base BOOT directory.

With these changes I booted up the zc706 and executed "~/openwifi/fosdem.sh". I obtained the following output:

--------------------------------------------------------------------

root@analog:~# ~/openwifi/fosdem.sh
hostapd: no process found
network-manager stop/waiting
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
115.00 dB
114.75 dB
rx0 gain to 70
54.000000 dB
sh: echo: I/O error
54.000000 dB
rx1 gain to 70
53.000000 dB
sh: echo: I/O error
53.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module rx_intf is not currently loaded
check rx_intf module is loaded or not
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
check openofdm_tx module is loaded or not
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
[... 368 lines cut ...]
sdr,sdr openwifi_dev_probe: test_mode 0
sdr,sdr openwifi_dev_probe: rfkill radio on
sdr,sdr openwifi_dev_probe: band_2GHz.n_channels 14 n_bitrates 12 band_5GHz.n_channels 11 n_bitrates 12
sdr,sdr openwifi_dev_probe: mac_addr 66:55:44:33:22:e4
ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
sdr,sdr openwifi_dev_probe: ieee80211_register_hw 0
ieee80211 phy0: hwaddr 6655443322e4, ZYNQ + ad9361
sdr,sdr openwifi_rfkill_init: wireless switch is on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr fpga-axi@0:sdr sdr0: renamed from wlan0
stop: Unknown instance:
isc-dhcp-server start/running, process 2658
Configuration file: hostapd-openwifi.conf
sdr0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface sdr0 with hwaddr 66:55:44:33:22:e4 and ssid "openwifi"
sdr0: interface state COUNTRY_UPDATE->ENABLED
sdr0: AP-ENABLED

--------------------------------------------------------------------

This looked very promising to me! However when I went to my phone as well as my computer wifi network listing I didn't see any 'openwifi' access point available. I then thought I may have connected the WiFi antennas to the wrong ports on the fmcomm3-ebz board. I looked at the picture on the webserver hosted by the board and matched the antenna ports you utilized in your picture. This didn't change anything. I then placed the TX antenna on TX1A and the RX antenna on RX1A. This also didn't change anything. Note I have the other two coax ports connected to 50 ohm loads. I thought I would then try to change the channels the openwifi is utilizing and/or change it not to use 5 GHz mode (even though I know my phone and computer WiFi can handle both 2.4 and 5 GHz modes). Before I started to make these changes though I thought I would post my progress here, and see if you could already spot something else I am doing wrong. 

Thank you for your time and help again! I really do appreciate it!
Best,
-Matt

JiaoXianjun, 2020-05-12:
Indeed looks promising. 
You can do sth like this, after you run fosdem.sh:

cat /prof/interrupts

Run above several time to see whether RX interrupt and tx interrupt is changing. Normally the tx interrupt count should change because the beacon is sent periodically. 

use dmesg to see whether there are printing for RX and tx.

When you do this better make sure some WiFi traffic is there at the same channel as openwifi. 

You can change the openwifi channel in the hostapd configuration file.

skinnym, 2020-05-12:
Thank for for the additional advice. I believe I followed your directions and the following is the output (also note I changed the channel to 36 because I know there is wifi traffic happening there according to my home router wifi scanner):

**### root@analog:~/openwifi# ~/openwifi/fosdem.sh**
hostapd: no process found
network-manager stop/waiting
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
112.75 dB
112.50 dB
rx0 gain to 70
51.000000 dB
sh: echo: I/O error
51.000000 dB
rx1 gain to 70
50.000000 dB
sh: echo: I/O error
50.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module rx_intf is not currently loaded
check rx_intf module is loaded or not
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
check openofdm_tx module is loaded or not
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
check openofdm_rx module is loaded or not
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
check xpu module is loaded or not
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
insert sdr.ko test_mode=0
check sdr module is loaded or not
[... 1940 lines cut ...]
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BEACON_INT 64
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_TXPOWER 14
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BEACON_ENABLED
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BEACON_ENABLED 512 BSS_CHANGED_BEACON 256
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_CTS_PROT 0
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BASIC_RATES f0
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_SLOT 8 BSS_CHANGED_ERP_PREAMBLE 4 short slot 1
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rx_interrupt: WARNING jump idx target 0 old 1 diff 0f
sdr,sdr openwifi_rx_interrupt:2777bytes ht0  6M FC0743 DI3a6b addr1/2/3:c8158bbad277/72b8e40d810b/e95a17b9a833 SCaeb3 fcs0 sn2 i1 -92dBm
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rx_interrupt:  32bytes ht0 24M FC6c40 DI57d3 addr1/2/3:773282cefab3/8869bc14cf48/626f6f3328f3 SCd438 fcs0 sn3 i2 -85dBm
sdr,sdr openwifi_rx_interrupt:  32bytes ht0 24M FC39a4 DI0b8b addr1/2/3:13588613da92/3e67135e522d/597b92fd9a9d SCf464 fcs0 sn4 i3 -88dBm
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on

--------------------------------

Thank you!
-Matt

JiaoXianjun, 2020-05-13:
From the log, fpga works fine. But signal is quite weak. 

I suggest you check antenna and distance. 

Antenna should be tx1 and rx1. 
Make sure your antenna SMA connector and fmcommsX SMA connector Make/female pro poetry is matched. And your antenna should support 5GHz band. Some antenna support only 2.4GHx band. Some antenna supports dual band. 

If your antenna has issue, you can Move your cellphone very very close to the antenna port.

skinnym, 2020-05-13:
You are an amazing person!

It works! Thank you so much for your help and patience.

Also again - I think the work you are doing is absolutely amazing! I do not think it is an easy thing to do (relative to other and myself). You should be proud of yourself.

Thank you so much!
-Matt

closed 2020-05-13 by skinnym

JiaoXianjun, 2020-05-13:
Glad to know it works.
Would you please tell me what kind of change makes it work finally? Then next time we could have more efficient advise to other people.

skinnym, 2020-05-13:
Absolutely! This is the least I can do for your work and guidance.

I think the main thing was the antennas I was using. I had a VERT 2450 I think on the TX and an unknown antenna (believed to be wifi band) antenna on the RX. It is possible this RX antenna didn't support dual band. Not sure. Their are no markings on it as far as I can tell.

Many other reasons I didn't get it working right away:
     - I misread (too quickly read) the quick start section, just looked for the closest thing that resembeled my hardware board inside of the main /BOOT partition. I found the analog device folder for my device (as close as I could identify) and copied that to the root /BOOT location. I then saw another posting from the closed question/answer section on github about making the same mistake I made. MUST USE the openwifi/ directory on the /BOOT partition instead of all the folders that look similar in the root directory.

- I went to the web server hosted by the device on the (192.168.10.1) and saw the picture their with a board that looked just like mine. I rearranged my RF antennas to match what you did (as far as I could understand from the photo). I also tried to use the for lack of better words change_ant.sh or whatever it is called script you made to change the antennas to the known ports I had antennas connected to. I don't think it is TX1 and RX1 in the picture hosted on the website, but it could just be a different FMCOMM board for all I know.

- Like i said above I think the main reason was my RX antenna probably not being dual band.

- Oh... other thing I did wrong was I used win32diskimager in windows and wrote the image, but never bothered to verify the contents. Once I started to verify I noticed many I/O errors. Anyways I switched SD card as well as the SD-card writer/reader (because for some reason I had to disassemble it at one point due to ... something different) so just to rule out SD-card reader/writer I swapped it. Verify the image! Also I used something called etcher balenaEtcher (google etcher usb) anyways because I was (naively) worried about my setting flags on dd in Linux to burn the image.

-Another issue I had was I used "git clone" originally to download the repo from github and I had errors resolving the analog device repos (definitely separate, issue but something to maybe note) - I tried this both in Linux and Windows and had same errors on cloning the analog device components from openwif repo.

I don't remember anything else right now. If I do I will add further comments.

Again, thank you so much for your help and patience!
-Matt

JiaoXianjun, 2020-05-13:
Thanks for the comprehensive notes. Absolutely will be help for people later on!

JiaoXianjun, 2020-07-31:
Hello,

How are you doing with openwifi now? Do you still have any issue? The openwifi has been improved a lot during the last .5 year. It supports more boards (high end to low end) and becomes more stable. 

Would you please tell us your email? If you could also introduce yourself bit (student? Company? University?), that would be perfect. We might send out some questions to listen for user feed feedback.

Thanks.