open-sdr/openwifi #17: Kernel panic when connecting to AP

frestuc, 2020-05-13:
Hi,
I am getting a kernel panic when I am trying to associate to the AP after launching `fosdem.sh`.
It happens with both the default BOOT/devicetree and the ones compiled from scratch:

```
root@analog:~/openwifi# sdr0: STA 1c:36:bb:8e:9b:06 IEEE 802.11: did not acknowledge authentication response
sdr0: STA 1c:36:bb:8e:9b:06 IEEE 802.11: associated (aid 1)
sdr0: AP-STA-CONNECTED 1c:36:bb:8e:9b:06
sdr0: STA 1c:36:bb:8e:9b:06 RADIUS: starting accounting session 5EBC28FE-00000000
Unable to handle kernel NULL pointer dereference at virtual address 00000004
pgd = c0004000
[00000004] *pgd=00000000
Internal error: Oops: 817 [#1] PREEMPT SMP ARM
Modules linked in: 8021q arc4 sdr(O) xpu(O) openofdm_rx(O) openofdm_tx(O) rx_intf(O) ad9361_drv(O) tx_intf(O) xilinx_dma mac80211 cfg80211 mwipcore mwipcore_iio_streaming mwipcore_dma_streaming mathworks_ip_common
CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  O    4.14.0-g4fea7c58ad92-dirty #1
Hardware name: Xilinx Zynq Platform
task: c0c0da80 task.stack: c0c00000
PC is at skb_dequeue+0x54/0x6c
LR is at skb_dequeue+0x14/0x6c
pc : [<c05e337c>]    lr : [<c05e333c>]    psr: 200b0193
sp : c0c01e60  ip : 00000000  fp : 40000006
r10: c0c02080  r9 : ee07a898  r8 : bf0cd990
r7 : bf0cd938  r6 : ee07a894  r5 : ee07a888  r4 : eeabfc00
r3 : 00000000  r2 : 00000000  r1 : 600b0113  r0 : 600b0113
Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 18c5387d  Table: 2ebe804a  DAC: 00000051
Process swapper/0 (pid: 0, stack limit = 0xc0c00210)
Stack: (0xc0c01e60 to 0xc0c02000)
1e60: eeabfc00 ee07a888 ee07a460 bf074610 ee07a874 ee07a878 00000000 c0c5b200
1e80: c0b41294 ffffe000 c0c02080 c0120fb4 00000000 00000006 ffffe000 c0c02098
1ea0: c0c02080 00000101 c0c02080 c0101548 c0c0534c 00000000 ef003600 c0b412f8
1ec0: c0c5b200 0000000a ffff9a7f c0c02d00 c0801860 00200102 ef162364 c0b48280
1ee0: 00000035 00000000 00000000 ef003600 f8f01100 00000000 00000000 c01213cc
1f00: c0b48280 c015bc08 c0c05344 c0c1a114 f8f0010c f8f00100 c0c01f40 c0101424
1f20: c0108484 600b0013 ffffffff c0c01f74 c0c54c60 c0c00000 00000000 c010c20c
1f40: 00000001 00000000 00000001 c01179e0 ffffe000 c0c050f4 c0c050a8 c0b47a48
1f60: c0c54c60 00000000 00000000 00000000 2ec85000 c0c01f90 c0108480 c0108484
1f80: 600b0013 ffffffff 00000051 00000000 ffffe000 c0150e14 000000be c0c5a8c0
1fa0: c0c05080 ffffffff c0c5a8c0 efffcd40 c0b35a30 c0151148 c0c5a90c c0b00c6c
1fc0: ffffffff ffffffff 00000000 c0b00680 00000000 c0b35a30 c0c5ab54 c0c05098
1fe0: c0b35a2c c0c0ed04 0000406a 413fc090 00000000 0000807c 00000000 00000000
[<c05e337c>] (skb_dequeue) from [<bf074610>] (ieee80211_tasklet_handler+0x30/0xd4 [mac80211])
[<bf074610>] (ieee80211_tasklet_handler [mac80211]) from [<c0120fb4>] (tasklet_action+0x7c/0x118)
[<c0120fb4>] (tasklet_action) from [<c0101548>] (__do_softirq+0xe0/0x248)
[<c0101548>] (__do_softirq) from [<c01213cc>] (irq_exit+0xd0/0xd4)
[<c01213cc>] (irq_exit) from [<c015bc08>] (__handle_domain_irq+0x60/0xb4)
[<c015bc08>] (__handle_domain_irq) from [<c0101424>] (gic_handle_irq+0x4c/0x90)
[<c0101424>] (gic_handle_irq) from [<c010c20c>] (__irq_svc+0x6c/0xa8)
Exception stack(0xc0c01f40 to 0xc0c01f88)
1f40: 00000001 00000000 00000001 c01179e0 ffffe000 c0c050f4 c0c050a8 c0b47a48
1f60: c0c54c60 00000000 00000000 00000000 2ec85000 c0c01f90 c0108480 c0108484
1f80: 600b0013 ffffffff
[<c010c20c>] (__irq_svc) from [<c0108484>] (arch_cpu_idle+0x38/0x3c)
[<c0108484>] (arch_cpu_idle) from [<c0150e14>] (do_idle+0x16c/0x1f8)
[<c0150e14>] (do_idle) from [<c0151148>] (cpu_startup_entry+0x18/0x1c)
[<c0151148>] (cpu_startup_entry) from [<c0b00c6c>] (start_kernel+0x394/0x3a0)
Code: e5853008 e1c420d0 e584c000 e584c004 (e5823004) 
---[ end trace fc63c07a424ba159 ]---
Kernel panic - not syncing: Fatal exception in interrupt
CPU1: stopping
CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D W  O    4.14.0-g4fea7c58ad92-dirty #1
Hardware name: Xilinx Zynq Platform
[<c010edbc>] (unwind_backtrace) from [<c010b6f4>] (show_stack+0x10/0x14)
[<c010b6f4>] (show_stack) from [<c06edd44>] (dump_stack+0x8c/0xa0)
[<c06edd44>] (dump_stack) from [<c010d4dc>] (ipi_cpu_stop+0x64/0x74)
[<c010d4dc>] (ipi_cpu_stop) from [<c010dc1c>] (handle_IPI+0x6c/0x7c)
[<c010dc1c>] (handle_IPI) from [<c0101464>] (gic_handle_irq+0x8c/0x90)
[<c0101464>] (gic_handle_irq) from [<c010c20c>] (__irq_svc+0x6c/0xa8)
Exception stack(0xef05ff78 to 0xef05ffc0)
ff60:                                                       00000001 00000000
ff80: 00000001 c01179e0 ffffe000 c0c050f4 c0c050a8 c0b47a48 c0c54c60 00000000
ffa0: 00000000 00000000 2ec95000 ef05ffc8 c0108480 c0108484 60000013 ffffffff
[<c010c20c>] (__irq_svc) from [<c0108484>] (arch_cpu_idle+0x38/0x3c)
[<c0108484>] (arch_cpu_idle) from [<c0150e14>] (do_idle+0x16c/0x1f8)
[<c0150e14>] (do_idle) from [<c0151148>] (cpu_startup_entry+0x18/0x1c)
[<c0151148>] (cpu_startup_entry) from [<0010174c>] (0x10174c)
---[ end Kernel panic - not syncing: Fatal exception in interrupt
sdr,sdr openwifi_rx_interrupt:  84bytes ht0 54M FC0108 DI002c addr1/2/3:665544332236/1c36bb8e9b06/333300000002 SCa9d0 fcs1 sn29 i29 -56dBm
```

JiaoXianjun, 2020-05-13:
Which board are you using? I think it is most probably zcu102 board, right?
The issue can be reproduced steadily or randomly? Could you tell me how to reproduce it?

For zcu102 board, we do observe this crash when downlink (openwifi to other) traffic is too big.
Suspect that it is related to FPGA queue and Linux queue status unsync/conflict because zcu102 FPGA and Linux run in higher clock speed. We are working on this.

frestuc, 2020-05-13:
The board is ZC706. 
I just loaded the BOOT.bin and devicetree.dtb from the openwifi folder in the SD card. 
Same thing happens with BOOT.bin and devicetree.dtb built from scratch.

JiaoXianjun, 2020-05-13:
which RF board?
We know that many people succeed with zc706. You could check:
SD card flashing method. We use Ubuntu native tool to flash.
You need 32bit image instead of 64bit image.
copy file from openwifi/zc706_fmcs2/* to the base directory of BOOT partition.
copy openwifi/zynq-common/uImage to the base directory of BOOT partition.

JiaoXianjun, 2020-05-13:
You can also share me the output/dmesg of fosdem.sh

frestuc, 2020-05-13:
> which RF board?

FMCOMMS2

> We know that many people succeed with zc706. You could check:
> SD card flashing method. We use Ubuntu native tool to flash.
> You need 32bit image instead of 64bit image.
> copy file from [...]

Done, still getting the same output

> You can also share me the output/dmesg of fosdem.sh

This is the output I get 

```
root@analog:~/openwifi# ./fosdem.sh 
hostapd: no process found
network-manager stop/waiting
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
115.75 dB
115.50 dB
rx0 gain to 70
54.000000 dB
sh: echo: I/O error
54.000000 dB
rx1 gain to 70
53.000000 dB
sh: echo: I/O error
53.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module rx_intf is not currently loaded
[... 349 lines cut ...]
sdr,sdr openwifi_dev_probe: ad9361_ctrl_outs_setup en_mask 0xff index 0x16
sdr,sdr openwifi_dev_probe: ad9361_spi_read REG_CTRL_OUTPUT_POINTER 0x16
sdr,sdr openwifi_dev_probe: ad9361_spi_read REG_CTRL_OUTPUT_ENABLE 0xff
sdr,sdr custom_match_platform_dev 79024000.cf-ad9361-dds-core-lpc
sdr,sdr openwifi_dev_probe: cf-ad9361-dds-core-lpc dds_st->version 00090162 chip_info->name AD9361
sdr,sdr openwifi_dev_probe: cf_axi_dds_datasel DATA_SEL_DMA
sdr,sdr openwifi_dev_probe: priv->rf_bw == 40000000Hz. bool for 20000000 0, 40000000 1
sdr,sdr openwifi_dev_probe: test_mode 0
sdr,sdr openwifi_dev_probe: rfkill radio on
sdr,sdr openwifi_dev_probe: band_2GHz.n_channels 14 n_bitrates 12 band_5GHz.n_channels 11 n_bitrates 12
sdr,sdr openwifi_dev_probe: mac_addr 66:55:44:33:22:0f
ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
sdr,sdr openwifi_dev_probe: ieee80211_register_hw 0
ieee80211 phy0: hwaddr 66554433220f, ZYNQ + ad9361
sdr,sdr openwifi_rfkill_init: wireless switch is on
sdr,sdr openwifi_rfkill_poll: wireless radio switch turned on
sdr,sdr fpga-axi@0:sdr sdr0: renamed from wlan0
stop: Unknown instance: 
isc-dhcp-server start/running, process 2370
Configuration file: hostapd-openwifi.conf
sdr0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface sdr0 with hwaddr 66:55:44:33:22:0f and ssid "openwifi"
sdr0: interface state COUNTRY_UPDATE->ENABLED
sdr0: AP-ENABLED 
```

JiaoXianjun, 2020-05-13:
Can't see abnormal things. 

After running hostapd, can you see the "openwifi" SSID from other wifi device? At this step, is there a crash?

Crash always happen when you click connect on other device to connect openwifi?

I suggest use some 5GHz channel (like our default 44 hostapd configuration file) as start. Also could check some basic functionalities:

service network-manager stop
./wgd.sh
iwlist sdr0 scan
(See if some SSIDs can be found)

use wpa_supplicant or "iwconfig sdr0 essid SSID_NAME" to connect other AP from openwifi.

frestuc, 2020-05-13:
> Can't see abnormal things.
> 
> After running hostapd, can you see the "openwifi" SSID from other wifi device? At this step, is there a crash?

No. The crash happens when i connect to the openwifi network.

> 
> Crash always happen when you click connect on other device to connect openwifi?
> 

Yes

> I suggest use some 5GHz channel (like our default 44 hostapd configuration file) as start. Also could check some basic functionalities:
> 
> service network-manager stop
> ./wgd.sh
> iwlist sdr0 scan
> (See if some SSIDs can be found)
> 

iwlist works fine, unfortunately we get that error when connecting to openwifi as AP

JiaoXianjun, 2020-05-13:
Could you tell me your zc706 revision?

https://www.xilinx.com/support/answers/59748.html

frestuc, 2020-05-13:
> Could you tell me your zc706 revision?
> 
> https://www.xilinx.com/support/answers/59748.html

REV 2.0

JiaoXianjun, 2020-05-13:
Currently I can not figure out the reason.
From the crash message, something abnormal happens in mac80211, not in sdr driver. Looks like the Linux ( file system ) is abnormal. So, Change the SD card, or change the board to take a look.
We will also test again at our side with zc706.

frestuc, 2020-05-13:
> Currently I can not figure out the reason.
> From the crash message, something abnormal happens in mac80211, not in sdr driver. Looks like the Linux ( file system ) is abnormal. So, Change the SD card, or change the board to take a look.
> We will also test again at our side with zc706.

It looks indeed like a Linux problem. I want to say that with the earlier versions of openwifi (when you had only the zc706 code, back in late November 2019) this wasn't happening.

JiaoXianjun, 2020-05-13:
Can you follow this section (https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch) to make a new SD card?

frestuc, 2020-05-13:
> Can you follow this section (https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch) to make a new SD card?

I did it, still getting the kernel panic

JiaoXianjun, 2020-05-14:
Even after you fix your kernel issue in the closed driver compiling issue?

frestuc, 2020-05-14:
> Even after you fix your kernel issue in the closed driver compiling issue?

Yes. Is there any way we can debug this problem?

JiaoXianjun, 2020-05-14:
Share us the SDR card image you built by script. We can take a look and try our side.

JiaoXianjun, 2020-05-14:
We tested again. It works perfectly at our side on zc706 fmcomms2.
Maybe you should change your hardware to nail it down.

frestuc, 2020-05-14:
Sorry for the delay, please find the images here: 

https://www.dropbox.com/s/jbn9k4eumubie9f/sdcard_BOOT.img?dl=0 
https://www.dropbox.com/s/7xwt23mqjt1o30p/sdcard_rootfs.img?dl=0

JiaoXianjun, 2020-05-15:
How to flash two img files into a single SD card?
Can you make one whole image for the SD card?

frestuc, 2020-05-15:
> How to flash two img files into a single SD card?

Can you try with `dd if=path/to/sdcard_BOOT.img of=/dev/mount_for_BOOT bs=1024 ` and similar for rootfs?

JiaoXianjun, 2020-05-15:
To avoid extra uncertainty, we prefer to follow our procedure firstly to do the test.
After you create your SD card with our script, please try to generate a single img file from SD card as following:

sudo fdisk /dev/mmcblk0 -l
(mmcblk0 is the SD card device name in your computer! Find out the correct name!)

Disk /dev/sdc: 14,9 GiB, 15962472448 bytes, 31176704 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00096174

Device     Boot   Start      End  Sectors  Size Id Type
/dev/sdc1          8192  1056767  1048576  512M  c W95 FAT32 (LBA)
/dev/sdc2       1056768 XXXXXXX 13441024  6,4G 83 Linux
/dev/sdc3          4096     6143     2048    1M a2 unknown

(From above fdisk info, the SD card Sector size 512 and end at XXXXXXX. Then:)

sudo dd bs=512 count=XXXXXXX if=/dev/mmcblk0 of=openwifi-xxxx.img

JiaoXianjun, 2020-05-15:
Could you tell me how did you create SD card after download our pre-built image?

frestuc, 2020-05-15:
> Could you tell me how did you create SD card after download our pre-built image?

with Ubuntu's Startup Disk Creator

frestuc, 2020-05-15:
> To avoid extra uncertainty, we prefer to follow our procedure firstly to do the test.
> After you create your SD card with our script, please try to generate a single img file from SD card as [...]

There you go https://www.dropbox.com/s/9aeog7pfag8fzz0/openwifi-frank.img?dl=0

weiliu1011, 2020-05-17:
Hi Frank,
I just tested your image on my hardware zc706 rev 2.0, fmcoms4, it works perfectly, I can connect two phones to it and stream video several times to both phones without any sign of crash. So this means a high probability of hardware issue on your side.
You said that with earlier version of openwifi you did not experience of any issue, I would say test that earlier version, see if you experience issue, if same thing occurs then we are certain it is hardware related.
Br,
Wei

frestuc, 2020-05-17:
> I would say test that earlier version, see if you experience issue, if same thing occurs then we are certain it is hardware related.

Ok, do you know where I can get the old rootfs?

JiaoXianjun, 2020-05-17:
You can go back to the commit you think is OK, and tell me the image file name?
The very beginning one is here: https://drive.google.com/file/d/1tEL9RIES8PwsvBoqOjGoaFJ9_jflsNsI/view?usp=sharing

frestuc, 2020-05-17:
The openwifi-hw commit is `ec02b58990ff1fd89afe78853182ca3955866f28`, the openwifi commit is `20101e8202c7626fd0220b8b5022cf7b2632ba91`. Can you confirm the one you linked is the right one?

frestuc, 2020-05-17:
I do not experience the same issue when using the image you posted. So I think it's not a hw issue...

JiaoXianjun, 2020-05-17:
Thing is getting interesting.
We are working on more safe queue management method which I believe will reduce a lot of chance getting crash in Linux. It will be available in next few weeks.

Could you test with more types of client (iPhone, iPad, laptop, Andriod, dongle, etc)? Because different Wi-Fi chips have different characters (Timing, packet interval, etc.). The crash only happens for a single client? Or all kinds of client can trigger this crash?

Meanwhile, if you want, you can test more images: https://drive.google.com/drive/folders/1KadYvc-dRD6iBdpHE3XfPMI0mJxKJ9Np?usp=sharing
This will help us understand the issue.

Thanks!

frestuc, 2020-05-18:
> Thing is getting interesting.
> We are working on more safe queue management method which I believe will reduce a lot of chance getting crash in Linux. It will be available in next few weeks.
>
> Could [...]

I have tried different clients, same thing. I think it's definitely a kernel/software issue.

> 
> Meanwhile, if you want, you can test more images: https://drive.google.com/drive/folders/1KadYvc-dRD6iBdpHE3XfPMI0mJxKJ9Np?usp=sharing
> This will help us understand the issue.
 
Will do, let me get back to you.

> Thanks!

JiaoXianjun, 2020-06-11:
Hello,

Would you please try this new image (openwifi-1.1.0-taiyuan-3-32bit.img.xz) to see whether the crash issue is gone?
https://drive.google.com/file/d/1DuSlHjLq9WQGvUv-0VNElqemQL_k3dPc/view?usp=sharing

I have fixed the potential crash issue by improving FPGA and driver involving more safe and exception tolerant design.

If your test give me positive result, I will update the source code to this git repository soon.

Thanks!

frestuc, 2020-06-11:
Hi, 
I am getting the following error:

```
root@analog:~/openwifi# hostapd hostapd-openwifi.conf 
Configuration file: hostapd-openwifi.conf
nl80211: Could not configure driver mode
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface sdr0 wasn't started
```
However, sdr0 is up and works fine (was able to scan with iwlist).

JiaoXianjun, 2020-06-11:
Can you just flash the SD card with the image, and run fosdem.sh to start everything?

frestuc, 2020-06-11:
It works, I am able to associate without crash, however the link is very unstable and weak, I get disassociated after a short while. I think you fixed the bug though.

JiaoXianjun, 2020-06-11:
Good to know! In our local test with iPhone and laptop no big performance difference. Maybe you could adjust distance/location. For iphone you can use WiFiPerf Endpoint, and HE.NET Network Tools.

closed 2020-06-12 by JiaoXianjun

frestuc, 2020-07-18:
What is the commit where you fixed the bug? Is it `4ecf49bb3edf31ca09d11d9d5e397c3dbc353c0a`?

JiaoXianjun, 2020-07-18:
I think so. Also openwifi-hw is updated as well. It is an essential update of the FPGA, driver and Linux flow control mechanism on different queues.