open-sdr/openwifi #18: Error when building drivers

frestuc, 2020-05-13:
Getting this error when compiling the linux drivers:

```
frank@frank:~/Documents/xilinx_projects/openwifi/driver$ ./make_all.sh .. /opt/Xilinx/ 32
$OPENWIFI_DIR is found!
$XILINX_DIR is found!
$ARCH_OPTION is valid!
 setup linux kernel path ../adi-linux/
++ pwd
+ home_dir=/home/frank/Documents/xilinx_projects/openwifi/driver
+ cd ../driver/
+ make KDIR=../adi-linux/ ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf-
make -C ../adi-linux/ M=/home/frank/Documents/xilinx_projects/openwifi/driver modules
make[1]: Entering directory '/home/frank/Documents/xilinx_projects/openwifi/adi-linux'
  CC [M]  /home/frank/Documents/xilinx_projects/openwifi/driver/sdr.o
In file included from ./include/linux/notifier.h:16:0,
                 from ./include/linux/memory_hotplug.h:7,
                 from ./include/linux/mmzone.h:780,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/mm.h:10,
                 from ./include/linux/scatterlist.h:8,
                 from ./include/linux/dmapool.h:14,
                 from /home/frank/Documents/xilinx_projects/openwifi/driver/sdr.c:4:
./include/linux/srcu.h:64:2: error: #error "Unknown SRCU implementation specified to kernel configuration"
 #error "Unknown SRCU implementation specified to kernel configuration"
  ^~~~~
In file included from ./include/linux/memory_hotplug.h:7:0,
                 from ./include/linux/mmzone.h:780,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/mm.h:10,
                 from ./include/linux/scatterlist.h:8,
                 from ./include/linux/dmapool.h:14,
                 from /home/frank/Documents/xilinx_projects/openwifi/driver/sdr.c:4:
./include/linux/notifier.h:78:21: error: field ‘srcu’ has incomplete type
  struct srcu_struct srcu;
                     ^~~~
In file included from ./arch/arm/include/asm/unistd.h:16:0,
                 from ./include/uapi/linux/unistd.h:8,
                 from ./include/asm-generic/seccomp.h:14,
                 from ./arch/arm/include/generated/asm/seccomp.h:1,
                 from ./include/linux/seccomp.h:13,
                 from ./include/linux/sched.h:21,
                 from ./arch/arm/include/asm/tlbflush.h:208,
                 from ./arch/arm/include/asm/pgtable.h:29,
                 from ./include/linux/memremap.h:8,
                 from ./include/linux/mm.h:27,
                 from ./include/linux/scatterlist.h:8,
                 from ./include/linux/dmapool.h:14,
                 from /home/frank/Documents/xilinx_projects/openwifi/driver/sdr.c:4:
./arch/arm/include/uapi/asm/unistd.h:21:10: fatal error: asm/unistd-eabi.h: No such file or directory
 #include <asm/unistd-eabi.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
[...]
```

JiaoXianjun, 2020-05-13:
Could you use absolute directory instead of ".."

JiaoXianjun, 2020-05-13:
$OPENWIFI_DIR/user_space/prepare_kernel.sh $OPENWIFI_DIR $XILINX_DIR ARCH_BIT
Above is successful?

frestuc, 2020-05-13:
> $OPENWIFI_DIR/user_space/prepare_kernel.sh $OPENWIFI_DIR $XILINX_DIR ARCH_BIT
> Above is successful?

Yes

frestuc, 2020-05-13:
> Could you use absolute directory instead of ".."
I tried what you suggested but still errors.. check out https://www.dropbox.com/s/8g4ry43k7lui5ts/output.txt?dl=0

JiaoXianjun, 2020-05-13:
Can you give me "ls" result of directory:
openwifi/adi-linux
Xilinx installation directory

and alsot "git log" result in your openwifi/adi-linux directory.

JiaoXianjun, 2020-05-13:
And what is your Linux version and Vivado version?

frestuc, 2020-05-13:
> Can you give me "ls" result of directory:
> openwifi/adi-linux

```
frank@frank:~/Documents/xilinx_projects/openwifi/adi-linux$ ls
arch   ci       crypto         firmware  init    Kconfig  LICENSES     mm              net        samples   sound       usr      vmlinux-gdb.py
block  COPYING  Documentation  fs        ipc     kernel   MAINTAINERS  modules.order   README     scripts   System.map  virt     vmlinux.o
certs  CREDITS  drivers        include   Kbuild  lib      Makefile     Module.symvers  README.md  security  tools       vmlinux
```

> Xilinx installation directory

```
frank@frank:/opt/Xilinx/Vivado/2018.3$ ls
bin     dashboard  doc       fonts       ids_lite  layouts  lnx64   reportstrategies  settings64.csh  src         tps
common  data       examples  hybrid_sim  include   lib      msys64  scripts           settings64.sh   strategies
```

> and alsot "git log" result in your openwifi/adi-linux directory.

```
frank@frank:~/Documents/xilinx_projects/openwifi/adi-linux$ git log
commit 4fea7c58ad92283acb90f182821b51d72b6afefa (HEAD -> 2019_R1, origin/2019_R1)
Author: Alexandru Ardelean <alexandru.ardelean@analog.com>
Date:   Wed Jan 22 13:42:16 2020 +0200

    arch: microblaze: rename '/include/' -> '#include'
    
    The #include directive is a bit more flexible/permissive when including a
    mix of DTs and C header files with dt-binding macros.
    
    We did the same for ARM [Zynq] DTs.
    
    Signed-off-by: Alexandru Ardelean <alexandru.ardelean@analog.com>
```

> And what is your Linux version and Vivado version?

Vivado 2018.3.

JiaoXianjun, 2020-05-13:
The error message is really strange. Seems like the script can not identify many Linux native code for the sdr driver. I can run the script without any issue at my side.
My working environment is Ubuntu 18 LTS.
If no other thoughts, maybe time to switch to Ubuntu 18 LTS. This environment we have verified many times.

JiaoXianjun, 2020-05-13:
Another possibility is that your adi-linux kernel directory is not well prepared.
Can you give me output of:
$OPENWIFI_DIR/user_space/prepare_kernel.sh $OPENWIFI_DIR $XILINX_DIR ARCH_BIT

frestuc, 2020-05-13:
```
frank@frank:~/Documents/xilinx_projects/openwifi$ ./user_space/prepare_kernel.sh `pwd` /opt/Xilinx/ 32
$OPENWIFI_DIR is found!
$XILINX_DIR is found!
$ARCH_OPTION is valid!
+ cd /home/frank/Documents/xilinx_projects/openwifi/
+ git submodule init adi-linux
+ git submodule update adi-linux
+ cd /home/frank/Documents/xilinx_projects/openwifi/adi-linux
+ git checkout 2019_R1
M	drivers/dma/xilinx/xilinx_dma.c
Already on '2019_R1'
Your branch is up to date with 'origin/2019_R1'.
+ git pull origin 2019_R1
From https://github.com/analogdevicesinc/linux
 * branch                      2019_R1    -> FETCH_HEAD
Already up to date.
+ cp /home/frank/Documents/xilinx_projects/openwifi/driver/xilinx_dma/xilinx_dma.c /home/frank/Documents/xilinx_projects/openwifi/adi-linux/drivers/dma/xilinx/xilinx_dma.c
+ cp /home/frank/Documents/xilinx_projects/openwifi/kernel_boot/kernel_config ./.config
+ source /opt/Xilinx//SDK/2018.3/settings64.sh
++ source /opt/Xilinx/Vivado/2018.3/.settings64-Vivado.sh
+++ export XILINX_VIVADO=/opt/Xilinx/Vivado/2018.3
+++ XILINX_VIVADO=/opt/Xilinx/Vivado/2018.3
+++ '[' -n /opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/installs/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/var/lib/snapd/snap/bin ']'
+++ export PATH=/opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/installs/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/var/lib/snapd/snap/bin
+++ PATH=/opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/installs/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/var/lib/snapd/snap/bin
++ source /opt/Xilinx/SDK/2018.3/.settings64-SDK_Core_Tools.sh
+++ '[' -n /opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/installs/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/var/lib/snapd/snap/bin ']'
+++ export PATH=/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/installs/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/var/lib/snapd/snap/bin
[... cut ...]
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
+ make modules_prepare
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CHK     scripts/mod/devicetable-offsets.h
+ '[' 3 -gt 3 ']'
+ cd /home/frank/Documents/xilinx_projects/openwifi
```

frestuc, 2020-05-13:
> The error message is really strange. Seems like the script can not identify many Linux native code for the sdr driver. I can run the script without any issue at my side.
> My working environment is [...]

I will try and let you know

JiaoXianjun, 2020-05-13:
I don't know what is the issue.
Let's hope Ubuntu 18 resolves it.

frestuc, 2020-05-13:
Was able to compile the drivers with Ubuntu 19. There was some problem with
the kernel, I hadn't built it correctly. Thanks

closed 2020-05-13 by frestuc