open-sdr/openwifi #19: 802.11ax roadmap

buggy64, 2020-05-25:
Hello,
in the README of your amazing project it was mentioned that 802.11ax is on the roadmap. I wonder if you could share more details on the planed feature set and in particular on the time line.
Best regards

JiaoXianjun, 2020-05-25:
Hello,

Thanks for asking.
We are preparing some fundamental functionalities for the 11ax. I think after summer there will be a concrete plan on 11ax implementation.

JiaoXianjun, 2020-08-14:
I think we probably will have the 802.11ax carrier spacing by the end of this year.

imlibra, 2020-08-15:
Now, which parts of Verilog cores need to change for 802.11n/ax implementation?

JiaoXianjun, 2020-08-15:
Hello,

Thanks for reaching us.
Before I answer, would you please introduce a bit yourself? Like company/university name, email, which country are your from, what is your role (student/employee) etc.?

imlibra, 2020-08-16:
Not belong to any company/organization, just for Verilog and wireless study.
email: meow@imlibra.me

imlibra, 2020-08-16:
Why do wireless projects like this and OpenAirInterface need contributors to sign an agreements? For liability of patents in 3GPP and IEEE standards?

JiaoXianjun, 2020-08-16:
The basic reason is that once the contributor’s code join the repository, it will become our responsibility to maintain it for very long time. But in many cases, after long time we even can not reach the contributor anymore. Since we take the responsibility, we need to own the contributor’s code totally and make decision about their code by ourself without reaching the contributor. This is good for user.

JiaoXianjun, 2020-08-16:
Hello @buggy64 

Thanks for reaching us.
would you please introduce a bit yourself? Like company/university name, email, which country are your from, what is your role (student/employee) etc.?

imlibra, 2020-08-17:
Not belong to any organizations, students, just interested on Verilog and wireless, currently in China.

JiaoXianjun, 2020-08-17:
OK. thanks!

imlibra, 2020-09-02:
Is there any roadmaps for 802.11x and ac development currently? Like ongoing implementation each parts of FPGA cores?

JiaoXianjun, 2020-09-02:
We are working on 802.11n. And there will be some pieces ready for 802.11ax at the end of this year.

closed 2021-04-21 by JiaoXianjun

daishi-kondo, 2021-09-08:
Hello.
I would like to know whether the source code about 802.11ax will be publicly available soon.
Thank you.

JiaoXianjun, 2021-09-08:
I don't think it will be soon. Because the 11ax is quite complicated.

jonathancross, 2025-05-16:
2025: was ax abandoned?

JiaoXianjun, 2025-05-16:
We DO have ax! Currently only available in subscription: https://openwifi.tech/subscriptions/openwifi
After several years it will be public, if the subscription model rolling well.

jonathancross, 2025-05-20:
Oh, that is unfortunate that it isn't available as FOSS.
I understand you need to make money somehow, but my interest is in security / transparency.  Would it be possible for a FOSS project like OpenWRT to pay for a license, then integrate the code?  Or are licensees required to keep the source hidden from the public?