open-sdr/openwifi #24: Questions about AD9361/4 work modes

openhardwarefan, 2020-07-01:
Hello,

This is great project, I've got a few questions about some settings for AD9361/4.

1. How many radio channels are used for transmit and receive by default? I guess it's only one radio channel since the project also supports AD9364.
2. What's the working mode for AD936x by default, FDD or TDD? 
3. Is TDD controller being used? If so, where is TDD_ENABLE (from REG_TDD_CONTROL_0) programmed? And what signals trigger tx_intf and rx_intf to start transmit and receive packet respectively through TDD controller's control ENABLE and TXNRX pins
4. Is AD936x initialized into ALERT state when the driver is loaded and enabled?

Thank you,

ohf

JiaoXianjun, 2020-07-01:
1. Currently only one channel. For ad9361, we support channel/antenna switching from driver (please check set_ant.sh). In the future, the automatic antenna selection will be controlled by Linux mac80211

2. FDD.

3/4 are not valid, since we use fdd.

Because we use fdd full duplex mode, we can demodulate our own packet if you “unmute” the receiver. (Search unmute in this page: https://github.com/open-sdr/openwifi/tree/master/doc)

We have some digital IF design to reduce the residual full duplex self interference (leakage when no transmission): https://github.com/open-sdr/openwifi/tree/master/doc#analog-and-digital-frequency-design

The independent tx rx antenna also offer some isolation, then further reduce some residual self interference.

JiaoXianjun, 2020-07-02:
Do you need further clarification?
If not, I will close this issue.

openhardwarefan, 2020-07-02:
@JiaoXianjun thank you so much for quick response,yes, I got my questions answered.

closed 2020-07-02 by openhardwarefan