open-sdr/openwifi #25: Channel State Information extraction

mzakharocsc, 2020-07-07:
Hello,  how difficult is it to obtain CSI from this project, with device operating in monitor mode? Something similar to [nexmon_csi](https://github.com/seemoo-lab/nexmon_csi), [Intel 5300](https://dhalperi.github.io/linux-80211n-csitool/), and [Atheors](https://wands.sg/research/wifi/AtherosCSI/)?

JiaoXianjun, 2020-07-07:
Currently CSI can be gotten via fpga internal signal through FPGA tool (ILA/chipscope).
To make it more easy to be accessed, we need to setup a side channel to transfer this information to Linux from FPGA.

The feature is in our pipeline. But we need some time due to limited internal development resource. 

If anyone is interested in developing this feature, we will be glad to offer support.

pzanna, 2020-07-07:
I am looking to add CSI, RSSI and a couple of other metrics over the next month or so. Xianjun, your support would be much appreciated.

JiaoXianjun, 2020-07-08:
nice!

addrices, 2020-08-06:
hello, I'm trying to use Atheros-CSI-tool to obtain CSI (the board running openwifi is ap,and another computer is client),but I can not obtain any CSI. openwifi can not transmit HT packets should be the reason. Can you share your timeline about 802.11n with us? Any advice or solution would be appreciated. thank you very much.

JiaoXianjun, 2020-08-06:
We are working on 11n. It will be ready by the end of this year.

frestuc, 2020-10-05:
Hi guys, where exactly are CSI/RSSI computed in the FPGA code?

JiaoXianjun, 2020-10-06:
We will have csi/equalizer information feature (capture into Linux) this week. Stay tuned.

csi:

https://github.com/open-sdr/openofdm/blob/dot11zynq/verilog/equalizer.v

lts_i_out/lts_q_out when state == S_CALC_FREQ_OFFSET

rssi:

https://github.com/open-sdr/openwifi-hw/blob/master/ip/xpu/src/rssi.v

Please check how we use rssi.v in xpu.v

JiaoXianjun, 2020-10-08:
csi is done 
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md

closed 2020-10-08 by JiaoXianjun

sstarlung, 2021-03-04:
Hi, I'm so excited about your job. Finally, we can get pure PHY information from a  research wifi device able to communicate commercial wifi devices.
After a week of the survey, I found CSI and IQ can't be obtained simultaneously. Is there any reason? My research needs all information (CSI/RSSI/AGC gain) at the same time.

Thank you.

JiaoXianjun, 2021-03-04:
Hello,

It is just a implementation choice. Yes, we can have a mixed mode that outputs CSI and IQ at the same time, if we want it. Or you can check our code and document, and implement it yourself.

But before go further, may I get an introduction about you? (Student? University? Company? etc). Would you please send it to xianjun.jiao@ugent.be?

Thanks!