open-sdr/openwifi #26: Fail to let an Zynq-Client connect to an Zynq-AP on 5GHz band

Leo-Cheung-CUHK, 2020-07-11:
Xiaojun, could you help me solve this problem?

**I set an Zynq as an AP by hostapd, the hostapd-openwifi.conf is as follows:**
///////////////////////////////////////////////////////
interface=sdr0
driver=nl80211
country_code=BE
ssid=openwifi
hw_mode=a
channel=44
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 90 120 180

#logger_syslog=-1
#logger_syslog_level=2

#macaddr_acl=0
#auth_algs=1
#ignore_broadcast_ssid=1
wpa=2
wpa_passphrase=mytestpwd
wpa_key_mgmt=WPA-PSK
#wpa_pairwise=TKIP
#rsn_pairwise=CCMP
//////////////////////////////////////////////////
**And on another Zynq, I use wpa_supplicant -i sdr0 -c wpa-connect.conf to connect the "openwifi"**
**wpa-connect.conf  is as bellow:**
network={
        ssid="openwifi"
        psk="mytestpwd"
        #psk=11f68e99e79769d14354714c7f23d6a7bc3de8360f6f7965e949561f399a174d
}
////////////////////////////////////////////////
**But in the client's terminal, something wrong during the authentication:**

root@analog:~/openwifi# wpa_supplicant -i sdr0 -c wpa-connect.conf
Successfully initialized wpa_supplicant
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: SME: Trying to authenticate with 66:55:44:33:22:8c (SSID='openwifi' freq=5220 MHz)
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: SME: Trying to authenticate with 66:55:44:33:22:8c (SSID='openwifi' freq=5220 MHz)
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: SME: Trying to authenticate with 66:55:44:33:22:8c (SSID='openwifi' freq=5220 MHz)
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: SME: Trying to authenticate with 66:55:44:33:22:8c (SSID='openwifi' freq=5220 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="openwifi" auth_failures=1 duration=10
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SSID-REENABLED id=0 ssid="openwifi"
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: SME: Trying to authenticate with 66:55:44:33:22:8c (SSID='openwifi' freq=5220 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="openwifi" auth_failures=2 duration=20
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: CTRL-EVENT-SSID-REENABLED id=0 ssid="openwifi"
sdr0: SME: Trying to authenticate with 66:55:44:33:22:8c (SSID='openwifi' freq=5220 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="openwifi" auth_failures=3 duration=20

JiaoXianjun, 2020-07-11:
Hello,

Please try following instructions that have been verified. I believe that you can add more things according to your requirement later.

openwifi board 1 (AP):
```
cd openwifi
./fosdem.sh

```
openwifi board 2 (Client):
```
cd openwifi
service network-manager stop
./wgd.sh
iwconfig sdr0 mode managed
ifconfig sdr0 up
iwconfig sdr0 essid openwifi
dhclient sdr0
(Get IP from openwifi dhcp server. It could take a while)
./set_csma_normal.sh
(Turn on random backoff. Another script set_csma_high.sh will turn off backoff)
ping 192.168.13.1
```

Leo-Cheung-CUHK, 2020-07-11:
Hi Xianjun,

**Thanks for your kind reply.
I follow your instruction but it still doesn't work.
The client's terminal is like this:**

root@analog:~/openwifi# dhclient sdr0
PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.

--- 192.168.1.1 ping statistics ---
1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms

root@analog:~/openwifi# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:0a:35:00:01:22 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.122/24 brd 192.168.10.255 scope global eth0
       valid_lft forever preferred_lft forever
3: sdr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 66:55:44:33:22:83 brd ff:ff:ff:ff:ff:ff
    inet 169.254.5.142/16 brd 169.254.255.255 scope link sdr0:avahi
       valid_lft forever preferred_lft forever

**the ifconfig result from the client is as below:**

sdr0      Link encap:Ethernet  HWaddr 66:55:44:33:22:83  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sdr0:avahi Link encap:Ethernet  HWaddr 66:55:44:33:22:83  
          inet addr:169.254.5.142  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1

**Looking for your kind reply**

JiaoXianjun, 2020-07-11:
dhclient sdr0
After run above command you need to wait for it exit by itself. Do not curl+c to quit it. It takes while.
This command get your sdr0 IP address like 192.168.13.X

after iwconfig sdr0 essid openwifi
Could you tell me out put of the AP side?
And also output of iwconfig at client side? This checks whether your client is associated with AP.

you could also adjust the distance and location of your antenna. Due to the limited tx power and agc range, too close or too far are not good.

Are you using the latest git commit?

Leo-Cheung-CUHK, 2020-07-11:
The first printout is the result exiting by dhclient sdr0. And it did take a while. 

After "iwconfig sdr0 essid openwifi", it just returns with nothing at the client-side. And there is nothing output from the AP. 

I succeed in using Zynq to connect commercial-used AP. May try to adjust the distance as you suggested

JiaoXianjun, 2020-07-11:
Normally after you run “iwconfig sdr0 essid openwifi“, you should also see something at AP side printed by hostapd to show you that a client is associated. You run fosdem.sh at ap side, right? You can also paste me output of fosdem.sh.

Leo-Cheung-CUHK, 2020-07-11:
**Yes, I run fosdem.sh at ap side. I use the zc706 board for both AP and client. The output of fosdem.sh is:**

root@analog:~/openwifi# ./fosdem.sh 
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
hostapd: no process found
network-manager stop/waiting
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
num_arg  0
last_input xxxxxxxxx
test_mode 0
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module xilinx_dma is not currently loaded
check xilinx_dma module is loaded or not
xilinx_dma is loaded!
rmmod: ERROR: Module tx_intf is not currently loaded
check tx_intf module is loaded or not
tx_intf is loaded!
rmmod: ERROR: Module ad9361_drv is not currently loaded
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ cd /sys/bus/iio/devices/iio:device2/
+ set +x
FIR Rx: 32,1 Tx: 32,1
1
0
rx0 agc fast_attack
slow_attack
fast_attack
rx1 agc fast_attack
slow_attack
fast_attack
40000000
17500000
40000000
37500000
rssi
96.75 dB
111.00 dB
rx0 gain to 70
34.000000 dB
sh: echo: I/O error
34.000000 dB
rx1 gain to 70
47.000000 dB
sh: echo: I/O error
47.000000 dB
tx0 gain -89dB
-10.000000 dB
-89.000000 dB
tx1 gain 0dB
-10.000000 dB
0.000000 dB
rmmod: ERROR: Module rx_intf is not currently loaded
check rx_intf module is loaded or not
rx_intf is loaded!
rmmod: ERROR: Module openofdm_tx is not currently loaded
check openofdm_tx module is loaded or not
openofdm_tx is loaded!
rmmod: ERROR: Module openofdm_rx is not currently loaded
check openofdm_rx module is loaded or not
openofdm_rx is loaded!
rmmod: ERROR: Module xpu is not currently loaded
check xpu module is loaded or not
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
[... 356 lines cut ...]
Configuration file: hostapd-openwifi.conf
sdr0: interface state UNINITIALIZED->COUNTRY_UPDATE
Using interface sdr0 with hwaddr 66:55:44:33:22:b2 and ssid "openwifi"
sdr0: interface state COUNTRY_UPDATE->ENABLED
sdr0: AP-ENABLED 

**But nothing happened on AP side after running “iwconfig sdr0 essid openwifi“ on client side**

**In terms of distance, I can find the openwifi using "iwlist sdr0 scan" as bellows**
          Cell 34 - Address: 66:55:44:33:22:B2
                    Channel:44
                    Frequency:5.22 GHz (Channel 44)
                    Quality=41/70  Signal level=-69 dBm  
                    Encryption key:off
                    ESSID:"openwifi"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000000001178e3ed
                    Extra: Last beacon: 700ms ago
                    IE: Unknown: 00086F70656E77696669
                    IE: Unknown: 01088C9298A43048606C
                    IE: Unknown: 03012C
                    IE: Unknown: 050401020000
                    IE: Unknown: 7F080000000000000040

JiaoXianjun, 2020-07-11:
You run “ iwlist sdr0 scan” at openwifi client or commercial client. 
-69dbm is not good, maybe you could move two boards closer.

Leo-Cheung-CUHK, 2020-07-11:
**Yes, the signal strength is too low. But the distance between these two boards are already very close (maybe around 30cm). And under this setting, everything goes well when connecting my board to a commercial AP (  working on Frequency:5.22 GHz, Channel 44) with a distance much more than 2 meters. 
Bellow is the example when I connect to an AP on 5 GHz using
"wpa_supplicant -i sdr0 -c wpa-connect.conf & dhclient sdr0"**

**root@analog:~/openwifi# wpa_supplicant -i sdr0 -c wpa-connect.conf & dhclient sdr0**
[1] 2952
Successfully initialized wpa_supplicant
sdr0: CTRL-EVENT-SCAN-STARTED 
sdr0: SME: Trying to authenticate with c0:56:27:bb:e8:e1 (SSID='806Lab_5GHz' freq=5220 MHz)
sdr0: Trying to associate with c0:56:27:bb:e8:e1 (SSID='806Lab_5GHz' freq=5220 MHz)
sdr0: Associated with c0:56:27:bb:e8:e1
sdr0: WPA: Key negotiation completed with c0:56:27:bb:e8:e1 [PTK=CCMP GTK=CCMP]
sdr0: CTRL-EVENT-CONNECTED - Connection to c0:56:27:bb:e8:e1 completed [id=0 id_str=]
**root@analog:~/openwifi# ip a**
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:0a:35:00:01:22 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.122/24 brd 192.168.10.255 scope global eth0
       valid_lft forever preferred_lft forever
4: sdr0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 66:55:44:33:22:21 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.114/24 brd 192.168.1.255 scope global sdr0
       valid_lft forever preferred_lft forever
**root@analog:~/openwifi# ifconfig** 
eth0      Link encap:Ethernet  HWaddr 00:0a:35:00:01:22  
          inet addr:192.168.10.122  Bcast:192.168.10.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2165 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2048 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:130148 (130.1 KB)  TX bytes:444567 (444.5 KB)
          Interrupt:29 Base address:0xb000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:26 errors:0 dropped:0 overruns:0 frame:0
          TX packets:26 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2296 (2.2 KB)  TX bytes:2296 (2.2 KB)

sdr0      Link encap:Ethernet  HWaddr 66:55:44:33:22:21  
          inet addr:192.168.1.114  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:15 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:2209 (2.2 KB)  TX bytes:5025 (5.0 KB)

JiaoXianjun, 2020-07-11:
To rule out the hardware issue, can you switch the role of openwifi AP and client board?

And when you do the experiment (openwifi client to openwifi AP), could you please share me the dmesg output (open another ssh session) on both side, after you run dhclient.

You can also “cat /proc/interrupts” to see whether the number of tx and rx interrupts are growing.

Leo-Cheung-CUHK, 2020-07-12:
Hi, xianjun,

I rewrite the SD image by compiling the lastest commit of the driver and follow your previous steps as bellow:

```
openwifi board 1 (AP):
cd openwifi
./fosdem.sh

openwifi board 2 (Client):
cd openwifi
service network-manager stop
./wgd.sh
iwconfig sdr0 mode managed
ifconfig sdr0 up
iwconfig sdr0 essid openwifi
dhclient sdr0
```

This time, after the "wconfig sdr0 essid openwifi", the AP finally can show something:
```
root@analog:~/openwifi# sdr0: STA 66:55:44:33:22:a7 IEEE 802.11: did not acknowledge authentication response
sdr0: STA 66:55:44:33:22:a7 IEEE 802.11: disassociated due to inactivity
sdr0: STA 66:55:44:33:22:a7 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

It seems that the AP fails to decode the authentication response sent by STA

The dmesg on AP is

```
  Normal zone: 196608 pages, LIFO batch:31
  HighMem zone: 65536 pages, LIFO batch:15
random: fast init done
percpu: Embedded 16 pages/cpu @ef7c6000 s33036 r8192 d24308 u65536
pcpu-alloc: s33036 r8192 d24308 u65536 alloc=16*4096
pcpu-alloc: [0] 0 [0] 1 
Built 1 zonelists, mobility grouping on.  Total pages: 260608
Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait cpuidle.off=1
PID hash table entries: 4096 (order: 2, 16384 bytes)
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Memory: 895976K/1048576K available (7168K kernel code, 339K rwdata, 2388K rodata, 1024K init, 165K bss, 21528K reserved, 131072K cma-reserved, 131072K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
    lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc0800000   (8160 kB)
      .init : 0xc0b00000 - 0xc0c00000   (1024 kB)
      .data : 0xc0c00000 - 0xc0c54ea0   ( 340 kB)
       .bss : 0xc0c5a894 - 0xc0c83f7c   ( 166 kB)
Preemptible hierarchical RCU implementation.
	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
	Tasks RCU enabled.
RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
efuse mapped to f0802000
slcr mapped to f0804000
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[... 804 lines cut ...]
sdr0: moving STA 66:55:44:33:22:73 to state 3
sdr0: moving STA 66:55:44:33:22:73 to state 4
sdr,sdr openwifi_conf_tx: WARNING [queue 0], aifs: 2, cw_min: 3, cw_max: 7, txop: 47
sdr,sdr openwifi_conf_tx: WARNING [queue 1], aifs: 2, cw_min: 7, cw_max: 15, txop: 94
sdr,sdr openwifi_conf_tx: WARNING [queue 2], aifs: 3, cw_min: 15, cw_max: 1023, txop: 0
sdr,sdr openwifi_conf_tx: WARNING [queue 3], aifs: 7, cw_min: 15, cw_max: 1023, txop: 0
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_SLOT 8 BSS_CHANGED_ERP_PREAMBLE 0 short slot 1
sdr0: associated
sdr0: disassociated from 66:55:44:33:22:73 (Reason: 7=CLASS3_FRAME_FROM_NONASSOC_STA)
sdr0: moving STA 66:55:44:33:22:73 to state 3
sdr0: moving STA 66:55:44:33:22:73 to state 2
sdr0: moving STA 66:55:44:33:22:73 to state 1
sdr0: Removed STA 66:55:44:33:22:73
sdr0: Destroyed STA 66:55:44:33:22:73
sdr,sdr openwifi_bss_info_changed BSS_CHANGED_BSSID 000000000000
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_CTS_PROT 0
sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_SLOT 8 BSS_CHANGED_ERP_PREAMBLE 4 short slot 0
sdr,sdr openwifi_conf_tx: WARNING [queue 0], aifs: 2, cw_min: 15, cw_max: 1023, txop: 0
sdr,sdr openwifi_conf_tx: WARNING [queue 1], aifs: 2, cw_min: 15, cw_max: 1023, txop: 0
sdr,sdr openwifi_conf_tx: WARNING [queue 2], aifs: 2, cw_min: 15, cw_max: 1023, txop: 0
sdr,sdr openwifi_conf_tx: WARNING [queue 3], aifs: 2, cw_min: 15, cw_max: 1023, txop: 0
sdr,sdr ad9361_rf_set_channel 5220M rssi_correction 145 (change flag 0)
sdr,sdr openwifi_config changed flag 00000100

```

JiaoXianjun, 2020-07-12:
Seems like an issue we are working on internally. If you try more, there will be a chance to connect.

Have you tried to “switch the role of openwifi AP and client board”. ?

Leo-Cheung-CUHK, 2020-07-12:
Yes, I switch the role of openwifi AP and client board, but nothing changes.  I try many times, but so far, I haven't got a chance to connect it.
I think it may because that the Zynq-client somehow did not send back an ACK during the association procedure with a Zynq-AP.

JiaoXianjun, 2020-07-12:
But both openwifi AP and client can work with commercial device. Right?

If so, it could be the same issue we are tracking internally. Could be because that the client send out ack, but somehow the AP receiver doesn’t decode it successfully. 

I will send you a temporary work around tomorrow.

Leo-Cheung-CUHK, 2020-07-12:
Yes, both openwifi AP and client can work with the commercial devices. 

Looking for your kind reply!!!

JiaoXianjun, 2020-07-13:
Hi,

Attached is a quick and dirty patch to avoid the issue temporarily. 
[patch_public_issue_26.txt](https://github.com/open-sdr/openwifi/files/4911420/patch_public_issue_26.txt)

Patch is very simple. At AP side, just replace (sdr.c):
`if ((tx_result_report&0x10)==0)`
by:
`if ((tx_result_report&0x10)==0 || priv->drv_xpu_reg_val[0])`

Then recompile the driver (you could use driver/make_all.sh), and put/load the new driver on AP board. For client, you don't need to update the driver.

Usage:
1. Before your openwifi client tries to connect openwifi AP, issue the follow command at AP side:
`./sdrctl dev sdr0 set reg drv_xpu 0 1`
This command will always tell Linux mac80211 (AP side) the packet is sent successfully, even the ACK is not decoded well.

2. After your client is connected successfully, you could disable this "fake ACK status" at AP side by:
 `./sdrctl dev sdr0 set reg drv_xpu 0 0`

Good luck!

Leo-Cheung-CUHK, 2020-07-13:
Xianjun,  it works!!! Thanks a lot for your kind help.

JiaoXianjun, 2020-07-13:
Hi,

Nice to know. 

This simple patch also reveal many connected information: how does sdrctl channel work, how to pass the tx status to Linux, etc. I believe you can modify the design further according to your idea. 

I will leave this issue open until we have formal/decent solution for this issue.

Have fun!

Leo-Cheung-CUHK, 2020-07-13:
Yes, it takes me some time to figure out these related parts in the driver.  I May ask you some detailed questions later, and thanks for your help again.

DatPhungThe, 2020-08-11:
> Yes, both openwifi AP and client can work with the commercial devices.
> 
> Looking for your kind reply!!!
> 
> FYI: 1) Everything goes well when setting up an ad-hoc network among Zynqs.
> 2) When connecting to a commercial AP on 2.4GHz (802.11g) rather than 5GHz (802.11a), Zynq-STA cannot pass the
> authentication and fail to connect to this AP.

Hello Leo-Cheung-CUHK,
Did you resolve the 2nd issue in  your above comment ?
I am involved in the same problem. My Zynq-STA client (is implemented in Zedboard) is only connect to 2 commercial APs at channels: Channel 1 (2.412 GHz) and Channel 11 (2.462 GHz). It cannot find the other commercial APs (at Channel 4, Channel 6, Channel 9,...)
If you know the reason and solution of this issue, please share it with me.

Thank you very very much !!

JiaoXianjun, 2020-09-05:
Hello,

We have performance update recently. Would you please help us do a test to check whether the issue has gone in our latest update (you can pull the code and build or use the image directly) at your side (do not apply the work around)

Thanks!

DatPhungThe, 2020-10-04:
Hello Xianjun,

Thanks for your great efforts !

I have tried to connect an Zynq-Client (Zedboard) to an Zynq-AP (Zedboard) on 2.4 GHz and 5GHz band (AP-Client mode).
But the connection process always fail in Authentication or Association phase (most of them are **Authentication Failed**).
I followed your instruction in the file: _openwifi/doc/app_notes/ap-client-two-sdr.md_ 

In AP Board:
```
cd openwifi
./fosdem.sh
./sdrctl dev sdr0 set reg drv_xpu 0 1
./sdrctl dev sdr0 set reg drv_tx 7 3
./sdrctl dev sdr0 set reg drv_rx 7 3
```

In Client Board:
```
service network-manager stop
cd openwifi
./wgd.sh
ifconfig sdr0 up
./sdrctl dev sdr0 set reg drv_xpu 0 1
./sdrctl dev sdr0 set reg drv_tx 7 3
./sdrctl dev sdr0 set reg drv_rx 7 3
iwlist sdr0 scan
=> The "openwifi" AP is shown in the scanning results.
iwconfig sdr0 essid openwifi
```

When I observed the "_dmesg_" output on both side, I saw that:
- The first Authentication Request packet from Client is always not seen at AP. 
- The 2nd and 3rd Authentication Request packet (retries packets) are captured and seen at AP. And the AP sent responsive Authentication Response packets for the purpose of handshaking. 
But the Client can not receive that reponses.

The _dmesg_ on Client is:

```
sdr0: Inserted STA 66:55:44:33:22:d5
sdr0: send auth to 66:55:44:33:22:d5 (try 1/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:6655443322d5/66554433226e/6655443322d5 SC1b90 flag40010012 retr1 ack1 prio0 q0 wr57 rd56
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr0: send auth to 66:55:44:33:22:d5 (try 2/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:6655443322d5/66554433226e/6655443322d5 SC1ba0 flag40010012 retr1 ack1 prio0 q0 wr58 rd57
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr0: send auth to 66:55:44:33:22:d5 (try 3/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:6655443322d5/66554433226e/6655443322d5 SC1bb0 flag40010012 retr1 ack1 prio0 q0 wr59 rd58
sdr0: authentication with 66:55:44:33:22:d5 timed out
sdr0: Removed STA 66:55:44:33:22:d5
sdr0: Destroyed STA 66:55:44:33:22:d5
```

The _dmesg_ on AP is:

```
sdr,sdr openwifi_tx:   72bytes  6M FC0050 DI00ac addr1/2/3:66554433226e/6655443322d5/6655443322d5 SC0410 flag4000001e retr1 ack0 prio0 q0 wr20 rd19
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_tx:   72bytes  6M FC0050 DI00ac addr1/2/3:66554433226e/6655443322d5/6655443322d5 SC0420 flag4000001e retr1 ack0 prio0 q0 wr52 rd51
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_rx_interrupt:  34bytes  6M FC08b0 DI013a addr1/2/3:6655443322d5/66554433226e/6655443322d5 SC1ba0 fcs1 buf_idx15 -45dBm
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:66554433226e/6655443322d5/6655443322d5 SC0430 flag4020001b retr1 ack1 prio0 q0 wr63 rd62
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_rx_interrupt:  34bytes  6M FC08b0 DI013a addr1/2/3:6655443322d5/66554433226e/6655443322d5 SC1bb0 fcs1 buf_idx0 -46dBm
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:66554433226e/6655443322d5/6655443322d5 SC0440 flag4020001b retr1 ack1 prio0 q0 wr1 rd0
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_tx_interrupt: WARNING loop_count 0
```

I tried to switch the role of openwifi AP and client board, but the result did not change.

Could you give me some advices on how to solve the above problem? Thank you very much !

JiaoXianjun, 2020-10-04:
Are you using the image provided by us?
The issue only happens when you use two SDR boards? Each board can communicate with commercial devices?

DatPhungThe, 2020-10-05:
Hi Xianjun,

I am using the SD Cards whose contents I built myself from your project, based on your instructions in the section: "Build openwifi Linux img from scratch".
The issue appears when I try to connect 2 Openwifi Zedboards. Openwifi AP and client can communicate with the commercial devices seperately. 

BTW, I am going to download your latest image and check whether the problem happens with it. 
 
Best Regards,

DatPhungThe, 2020-10-06:
I have tested with your image. Now I can connect 2 Openwifi boards at the certain time, but it looks like unstable. There are some unexpected cases I met during testing:  

1. It is still authenticated Failed, same as my above description. 

The dmesg output on Client:

```
sdr0: Inserted STA 66:55:44:33:22:d2
sdr0: send auth to 66:55:44:33:22:d2 (try 1/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI00ac addr1/2/3:6655443322d2/6655443322dc/6655443322d2 SC0160 flag40010012 retr1 ack1 prio0 q0 wr21 rd20
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr22 rd21
sdr0: send auth to 66:55:44:33:22:d2 (try 2/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI00ac addr1/2/3:6655443322d2/6655443322dc/6655443322d2 SC0170 flag40010012 retr1 ack1 prio0 q0 wr22 rd21
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr23 rd22
sdr0: send auth to 66:55:44:33:22:d2 (try 3/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI00ac addr1/2/3:6655443322d2/6655443322dc/6655443322d2 SC0180 flag40010012 retr1 ack1 prio0 q0 wr23 rd22
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr24 rd23
sdr0: authentication with 66:55:44:33:22:d2 timed out
sdr0: Removed STA 66:55:44:33:22:d2
sdr0: Destroyed STA 66:55:44:33:22:d2

```
The dmesg output on AP:

```
sdr,sdr openwifi_tx:   69bytes  6M FC0050 DI00ac addr1/2/3:ba3bf50fd1ae/6655443322d2/6655443322d2 SC42e0 flag4000001e retr1 ack0 prio0 q0 wr45 rd44
sdr,sdr openwifi_rx_interrupt:  34bytes ht0  6M FC00b0 DI00ac addr1/2/3:6655443322d2/6655443322dc/6655443322d2 SC0170 fcs1 buf_idx6 -64dBm
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI00ac addr1/2/3:6655443322dc/6655443322d2/6655443322d2 SC4530 flag4020001b retr1 ack1 prio0 q0 wr18 rd17
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr19 rd18
sdr,sdr openwifi_rx_interrupt:  34bytes ht0  6M FC00b0 DI00ac addr1/2/3:6655443322d2/6655443322dc/6655443322d2 SC0180 fcs1 buf_idx7 -64dBm
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI00ac addr1/2/3:6655443322dc/6655443322d2/6655443322d2 SC4550 flag4020001b retr1 ack1 prio0 q0 wr20 rd19
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr21 rd20

```
 When I repeated the command: "iwconfig sdr0 essid openwifi" 2-3 times, AP and STA were handshaked again and could be associated successfully.

2. AP and STA were authenticated and associated successfully, but then DHCP process can not allocate IP to STA.

In Client Openwifi Board:

```
dhclient sdr0 
=> Wait for it exiting by itself and check ifconfig.
It shows: sdr0:avahi .... with IP Address: 169.254.x.x
```

The dmesg output on Client:

```
sdr0: Inserted STA 66:55:44:33:22:a9
sdr0: send auth to 66:55:44:33:22:a9 (try 1/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:6655443322a9/66554433220a/6655443322a9 SC06a0 flag40010012 retr1 ack1 prio0 q0 wr41 rd40
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr42 rd41
sdr0: send auth to 66:55:44:33:22:a9 (try 2/3)
sdr,sdr openwifi_tx:   30bytes  1M FC00b0 DI013a addr1/2/3:6655443322a9/66554433220a/6655443322a9 SC06b0 flag40010012 retr1 ack1 prio0 q0 wr42 rd41
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr43 rd42
sdr,sdr openwifi_rx_interrupt:  34bytes ht0  6M FC00b0 DI013a addr1/2/3:66554433220a/6655443322a9/6655443322a9 SC6570 fcs1 buf_idx2 -50dBm
sdr0: authenticated
sdr0: moving STA 66:55:44:33:22:a9 to state 2
sdr,sdr fpga-axi@0:sdr sdr0: disabling HT as WMM/QoS is not supported by the AP
sdr,sdr fpga-axi@0:sdr sdr0: disabling VHT as WMM/QoS is not supported by the AP
sdr0: associate with 66:55:44:33:22:a9 (try 1/3)
sdr,sdr openwifi_tx:   58bytes  1M FC0000 DI013a addr1/2/3:6655443322a9/66554433220a/6655443322a9 SC06c0 flag40010012 retr1 ack1 prio0 q0 wr43 rd42
sdr,sdr openwifi_tx: WARNING sn 43 skb_headroom(skb)<LEN_PHY_HEADER
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr44 rd43
sdr,sdr openwifi_rx_interrupt:  54bytes ht0  6M FC0010 DI013a addr1/2/3:66554433220a/6655443322a9/6655443322a9 SC6580 fcs1 buf_idx3 -50dBm
sdr0: RX AssocResp from 66:55:44:33:22:a9 (capab=0x401 status=0 aid=1)
sdr0: moving STA 66:55:44:33:22:a9 to state 3
sdr0: moving STA 66:55:44:33:22:a9 to state 4
sdr,sdr openwifi_conf_tx: WARNING [queue 0], aifs: 2, cw_min: 3, cw_max: 7, txop: 47
[... 63 lines cut ...]
sdr,sdr openwifi_rx_interrupt: 364bytes ht0  6M FC0108 DI0040 addr1/2/3:6655443322a9/66554433220a/ffffffffffff SC0750 fcs1 buf_idx12 -47dBm
sdr,sdr openwifi_tx:   26bytes  1M FC00a0 DI013a addr1/2/3:66554433220a/6655443322a9/6655443322a9 SC0000 flag4020001b retr1 ack1 prio0 q0 wr59 rd58
sdr,sdr openwifi_rx_interrupt:1986bytes ht0 54M FCb442 DI2449 addr1/2/3:ad75573daef8/ac8993bc69c8/1191e6b45cf0 SC9712 fcs0 buf_idx13 -64dBm
sdr,sdr openwifi_rx_interrupt: WARNING content!
sdr,sdr openwifi_rx_interrupt:  11bytes ht0  6M FC94fc DI67e2 addr1/2/3:9604f3e409da/000000000000/000000000000 SC0000 fcs0 buf_idx9 -64dBm
sdr,sdr openwifi_rx_interrupt:3825bytes ht0 12M FC7d43 DIb74e addr1/2/3:e56eb3f4222a/ca90602efdeb/b7fe2656f3e2 SC64b3 fcs0 buf_idx10 -59dBm
sdr,sdr openwifi_rx_interrupt:1084bytes ht0 18M FC98a4 DI05d9 addr1/2/3:91f88a32ee2f/61e3e44232dc/d0443c3e8cd1 SC46fb fcs0 buf_idx11 -60dBm
sdr,sdr openwifi_rx_interrupt: 241bytes ht0 12M FC1640 DI1e92 addr1/2/3:399ad5b698c9/3338ce396dd7/0197891af25e SC86db fcs0 buf_idx12 -60dBm
sdr,sdr openwifi_tx:   26bytes  1M FC00a0 DI013a addr1/2/3:66554433220a/6655443322a9/6655443322a9 SC0000 flag4020001b retr1 ack1 prio0 q0 wr33 rd32
sdr,sdr openwifi_tx_interrupt: WARNING tx_result 10 prio0 wr34 rd33
[... cut ...]
sdr,sdr openwifi_tx:   26bytes  1M FC00c0 DI013a addr1/2/3:66554433220a/6655443322a9/6655443322a9 SC0000 flag4020001b retr1 ack1 prio0 q0 wr51 rd50

```

3. When AP and STA see each other, I can ping between them. However, if I use iperf tool to measure the throughput of wireless channel, I can not transfer TCP packets and trace the throughput, as is described in the attached screenshots.

[image]
[image]

Have you ever seen the same situations ? Please give me some advices or explanations about the above results.
Thank you so much !!

Best Regards,

JiaoXianjun, 2020-10-06:
which board (type/model) are you using?

DatPhungThe, 2020-10-07:
I am using Zynq Zedboard with AD-FMCOMMS2-EBZ.
Thank you !

JiaoXianjun, 2020-10-07:
make sure per board your two antennas are separated enough. far away to each other or vertical/orthogonal to each other.
You can also make them closer.
You can also try the quick and dirty patch here: https://github.com/open-sdr/openwifi/issues/26#issuecomment-657430988

DatPhungThe, 2020-10-08:
Yes, I added the dirty patch #26 in AP side.
I am placing 2 boards as the following photo:
https://drive.google.com/file/d/1P72sepJQnLEJoPFR5GFj9LfnSDs8WdnZ/view?usp=sharing
Is this Ok ? I think the distance between AP and STA is already enough close.

JiaoXianjun, 2020-10-08:
1. You need only two antennas per board, not four antennas. One rx  and one tx. Please make tx rx antenna vertical/orthogonal to each other. (Not closely parallel to each other like in your photo!)

2. When you have difficulty on connection, can you try script: set_csma_normal.sh? (Both sides or one side)? To see whether the connection becomes easier.

DatPhungThe, 2020-10-18:
Hi Xianjun,

Thank you very much !
Now, I can connect easier thanks to your tips.

I have looked for the way to go through the issue: "_IEEE 802.11: did not acknowledge authentication response_"

I think the dirty patch: [](https://github.com/open-sdr/openwifi/issues/26#issuecomment-657430988) is only a temporary solution, not an official solution, so it can not resolve the root cause.

I set up an AP Openwifi on Zedboard (with AD-FMCOMMS2-EBZ) without using the above dirty patch. Meanwhile, I put involved signals in `xpu/tx_control` (tx_control_state,tx_fail_lock,ack_timeout_count) to ILA core for monitoring and debugging. 
Then I try connecting a Client Linux-PC to AP Openwifi.  

In ILA Monitor, I captured the rising edge of `tx_fail_lock` signal and observed that `tx_control_state` is `RECV_ACK_WAIT_SIG_VALID` and `ack_timeout_count` reaches to `recv_ack_sig_valid_timeout_top_scale` (5050) at that time.

I am wondering whether the timeout value is enough long to detect ACK packets, and whether ``COUNT_SCALE` factor is correct. 

I still confuse about the assumption: ``ASSUMED_COUNTER_CLK_MHZ  = 10 MHz` in `board_def.v`. Why is not it another number (5 MHz or 20 MHz) ?
    
Have you any ideas or approaches ? Please share it with me to discuss and overcome this problem together. 

Best Regards,

JiaoXianjun, 2020-10-19:
Yes. We know there is some issue with short packet rx performance sometimes. We are working on it.

DatPhungThe, 2020-10-19:
Hi Xianjun,
I am also researching on this problem.
Could you please share some hints or ideas about your way ? Let's find the solution together. 
Best Regards,

JiaoXianjun, 2020-10-20:
You can try our latest IQ sample tool to try to debug it.

DatPhungThe, 2020-10-20:
Ok, I got it. I will use your Side Channel modules for debugging.
Thank you for your great effort !

Best Regards,

JiaoXianjun, 2020-12-18:
Hello @Leo-Cheung-CUHK ,

Since our latest release (1.2.0) has improved the low MAC, it should improve this issue a lot.
Would you please try the new release, and tell us whether the problem is still serious or not?

Thanks!

Leo-Cheung-CUHK, 2020-12-18:
Dear Xianjun,

The latest release is very existing.

I will do the testing in a few days and let you know the result.

Best,
Lihao

Leo-Cheung-CUHK, 2020-12-22:
> Hello @Leo-Cheung-CUHK ,
> 
> Since our latest release (1.2.0) has improved the low MAC; it should improve this issue a lot.
> Would you please try the new release, and tell us whether the problem is still serious or not?
> 
> Thanks!

Hi @JiaoXianjun,

I used one ZC706 as AP, creating a WLAN. The hostapd.conf is:
`interface=sdr0
driver=nl80211
country_code=BE
ssid=openwifi
hw_mode=g/a
channel=4/44
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 90 120 180
ieee80211n=1
ht_capab=[SHORT-GI-20]
require_ht=1
ieee80211d=1
ieee80211h=1
#wpa=3
#wpa_passphrase=myrabbit
#wpa_key_mgmt=WPA-PSK
`

Then, my PC working as STA with NIC supporting 802.11agn tried to connect to this AP. Here is the testing result:

1) When the AP uses 802.11a on the No.44 channel, STA can connect to the AP.
2) When using 802.11g, STA can see the beacon sent from AP. But the establishment fails. 

I think it's because openwifi does not support DSSS. I tried to patch the wpa_supplicant as "Special note for 11b" part in Readme. But it still cannot work.

JiaoXianjun, 2020-12-22:
In channel 44, the dsss shouldn’t be there because it is in 5GHz.
So far we almost always test in 5GHz. 

My question is: does this issue (didn’t not acknowledge) disappear at your side?

Leo-Cheung-CUHK, 2020-12-30:
> In channel 44, the dsss shouldn’t be there because it is in 5GHz.
> So far we almost always test in 5GHz.
> 
> My question is: does this issue (didn’t not acknowledge) disappear at your side?

Yes, the problem has gone.

JiaoXianjun, 2020-12-30:
thanks. finally we solve it! I believe it is a low mac issue (collision) that has been solved recently.

closed 2020-12-30 by JiaoXianjun