open-sdr/openwifi #27: Support for USRP boards

aygunbaltaci, 2020-07-21:
Hello, 

Are USRP devices (e.g. B210) supported or is there any plan to support it in the future? Following the conversation in issue #14, I guess that B210 would also not be supported but I am not clear. 

Thanks!

JiaoXianjun, 2020-07-21:
Hello,

Thanks for reaching our project.
Currently we don’t have a plan in the near future to support USRP B series.

May I ask what is the purpose for you to support USRP?
The cheapest USRP (like the B200mini) has the same/similar price as adrv9364z7020 or zedboard+fmcomms4, but USRP B200mini has much smaller FPGA than the Zynq platform I mentioned. And in USRP B200mini, there is no ARM processor.

The Zynq platform I mentioned (adrv9364z7020 or zedboard+fmcomms4) offer you a whole embedded Linux and FPGA WiFi card in the same chip. From this perspective, USRP B200mini is not competitive at all for WiFi research compared with the Zynq platform I mentioned.

aygunbaltaci, 2020-07-22:
Hi JiaoXianjun, 

Thanks for the information. I understand your argument regarding the Zynq platform. The reason why I asked is just because I have B210 in stock and it is quite common SDR on the market. It is just very convenient as many open source projects support this board.

Best Regards
Aygün

closed 2020-07-22 by aygunbaltaci

ohault, 2024-06-21:
Could you please support USRP SDR devices ?

Ref : https://files.ettus.com/manual/md_fpga.html

JiaoXianjun, 2024-06-21:
> Could you please add support USRP SDR devices ?
> 
> Ref : https://files.ettus.com/manual/md_fpga.html

yes we can, but need efforts.
And no big motivation. reasons:
1. usrp is for IQ streaming between hardware and computer, openwifi is a wifi NIC. even we put openwifi on USRP, USRP will become a wifi NIC, and it can not be used as IQ streaming anymore.
2. the hardware we currently supported already much cheaper (and more powerful) than usrp. so no need to go for usrp. such as this, only 329 USD: https://www.crowdsupply.com/microphase-technology/antsdr-e200#products
3. differentiation. openwifi's design is unique. ARM linux and wifi (FPGA fabric) are in the single chip (Xilinx SoC), so it has ultra low latency, and many other benefit. going for usrp will just lose all these uniqueness and benefits.

ohault, 2024-06-21:
Here are also some others arguments:
- USRP is a reference SDR hardware platform in almost all academic and research circles.
- Why you would buy a new SDR hardware when you are already using USRP
- The goal here is not to create an open hardware WIFI custom ASIC for the mass as a nic, but for engineering, research, education and development.

JiaoXianjun, 2024-06-21:
then feel free to try to do the porting.
we would glad to answer question about know-how of our current non-usrp designs.

VikiAgung, 2025-01-16:
Hi @JiaoXianjun 

I have an issue on my project to make USRP B210 as access point and ended up here. From your previous explanation, does it mean USRP will be WiFi NIC permanently and can't be back as IQ streaming?