open-sdr/openwifi #28: Openwifi-Client cannot see some commercial APs

DatPhungThe, 2020-08-11:
Hello Xianjun,

Thank you very much for your great project. It is really useful for me.
I've read, researched and run your project. And I met some problems. So I've got the following questions. Please instruct me to solve it, and understand your project more clearly.

1. When I run Openwifi as Client STA in Zedboard, Openwifi-Client only finds APs operating at channel 1 (2.412 GHz) and channel 11 (2.462 GHz). Openwifi-Client cannot see the other commercial APs. I'm sure that the other APs are operating because I can scan and see it in a Linux PC by the command: "iwlist scan".

I set Openwifi-Client in Zedboard as follows:

cd ~/openwifi/
./wgd.sh
iwlist sdr0 scan

And the output of "iwlist sdr0 scan" is :
Cell 01 - Address: xx:xx:xx:xx:xx:xx
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
...................................
 Cell 02 - Address: xx:xx:xx:xx:xx:xx
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)

Meanwhile, in the Linux PC, the output of "iwlist wlp2s0 scan" is:
Cell 01 - Address:xx:xx:xx:xx:xx:xx
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
...................................

  Cell 02 - Address: xx:xx:xx:xx:xx:xx
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
.............................
Cell 03 - Address: xx:xx:xx:xx:xx:xx
                    Channel:4
                    Frequency:2.427 GHz (Channel 4)
.............................
Cell 04 - Address: xx:xx:xx:xx:xx:xx
                    Channel:4
                    Frequency:2.427 GHz (Channel 4)
.............................
Cell 05 - Address: xx:xx:xx:xx:xx:xx
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
.............................
Cell 06 - Address: xx:xx:xx:xx:xx:xx
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
.............................
  Cell 11 - Address: xx:xx:xx:xx:xx:xx
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
...................................

(*) Why do not Openwifi-Client see Channel 4, Channel 6, Channel 9,..., same as Linux PC? 
Please help me to explain about it and resolve it !!

2. In the Verilog source code files: "short_preamble_rom.v", "long_preamble_rom.v", and "dot11_tx.v" (belong to ofdm_tx module), I see that you describe the value of complex numbers. 
Examples:
32'h40000000 = 1 + 0j
32'hC0000000 = -1 + 0j
32'h05E305E3 = 0.046 + 0.046j
32'h0BC70000 = 0.092 + 0j

I can only understand : MSB bit is sign bit, the upper 16 bits are real part, and the lower 16 bits are image part. 
Is that right ?? 
How do you convert float numbers into hexadecimal number ?  What is standard used here ?

weiliu1011, 2020-08-11:
Hi DatPhungThe,
Can you tell if the other APs have similar transmit power and distance to the openwifi client? Please verify if it is caused by bad reception link
Can you also try iw sdr0 scan?

JiaoXianjun, 2020-08-11:
Hello,

Thanks for reaching us. 
Before I answer, would you please introduce a bit yourself? Like company/university name, email, where are your from, etc.?

DatPhungThe, 2020-08-11:
Hello Weiliu and Xianjun,

Thank you for your immediate reply.
My name is: Phung The Dat. I come from Vietnam. I'm an under-graduated researcher in Danang University.
My email address: phungthedat07dt2@gmail.com

I am glad to connect with you.
Thank you very much for your supporting !

JiaoXianjun, 2020-08-11:
Hello,

For the AP scan result, you could check those AP (can not be discovered by openwifi) beacon format to see whether it is in 802.11b modulation (wireshark or other sniffer software can tell you). Some AP always sends out beacon in 802.11b modulation to have the most broad compatibility. Unfortunately openwifi only works in OFDM mode which means that it can not decode 802.11b packet. 

Or the beacon is too new (has new fields that is not supported by openwifi). Currently the openwifi support only 802.11a/g. We are working on 802.11n. Not finished yet. 

For the relationship between floating point and fixed point number, the integer number you have shown is the result of: floating number times 2^15. 0.046 = 0x5e3/32768 = 1507/32768

DatPhungThe, 2020-08-12:
Hi Xianjun,

Thank you for your detailed answer.
Unfortunately, my PC interface Card does not support Monitor Mode to capture and display beacon frames from APs. 
I will try to search another adapter enabling Monitor Mode, and test again later.
BTW, I understood the method of converting floating point number to hexadecimal number.

Thank you very much for your valuable help !

JiaoXianjun, 2020-08-12:
You can also use monitor_ch.sh to force openwifi to monitor a specified channel, then use tcpdump to see what kind of frame can be seen. Or use dmesg (check project document) to see the received frame. 

You can also search on internet how to specify cell parameter (like 48bit MAC address) when connect a AP. Each AP should have such a parameter when a network is created. There is an example in sdr-ad-hoc-join.sh. But that is ad-hoc mode, you need to find out the command in station mode.

JiaoXianjun, 2020-08-12:
And I just send you a questionnaire via email. Would you please take a look? 

If you feel that you are not the correct person to answer some questions, could you please check with your teacher/professor?

Thanks!

DatPhungThe, 2020-08-13:
> And I just send you a questionnaire via email. Would you please take a look?
> 
> If you feel that you are not the correct person to answer some questions, could you please check with your teacher/professor?
> 
> Thanks!

I have just filled your questionnaire, and sent it to you. Please check it, and reply to me if you have any questions or interest in my opinions.

JiaoXianjun, 2020-08-14:
Has your issue been solved?

DatPhungThe, 2020-08-14:
Hi Xianjun,
Thanks for your regards.
As your instruction, I ran "monitor_ch.sh" for each separate channel, and used tcpdump to capture packets in both of Openwifi and Linux PC (for reference).
It seems that only AP at Channel 1 transmitted 802.11g Beacon frames. The others transmitted 802.11b frames. I can trace in Linux PC, and it shown "11b".
I observed that .11b frames received are marked "bad-fcs" in Openwifi-Client.

But I am confusing about Channel 11 (AP is my iPhone). I saw that it also transmitted 802.11b Beacon frames, but Openwifi-Client can find it by scan command. By tcpdump, I realized that my iPhone AP transmitted 802.11g Probe Response frames.  
Did Openwifi-Client base on this frames to detect my iPhone AP ? 

Thank you very much !!

JiaoXianjun, 2020-08-14:
Hi, good catch. 

Indeed there two mechanisms about ap scan.
Passive: only try to receive beacon of ap
Active: send probe request (not sure the concrete name), then ap send response. 
If the response is in ofdm, then we can see the ap. if the response is in 11b, we can not see.

DatPhungThe, 2020-08-14:
Hi Xianjun,

That's great. You reminded me basic theory of 802.11. 
I have just accessed to research 802.11 WLAN, so I do not have much experience with it. I am sorry about that. I will try my best.

Thank you very much !

closed 2020-08-14 by JiaoXianjun