open-sdr/openwifi #29: make-wifi-fast leverage?

dtaht, 2020-08-29:
to what extent are your leveraging the make-wifi-fast work? (see https://lwn.net/Articles/705884/ or https://www.usenix.org/system/files/conference/atc17/atc17-hoiland-jorgensen.pdf

there was a lot left over from that project (sadly unfunded) that would have been great to co-design a hw implementation around)

JiaoXianjun, 2020-09-05:
It is a very nice work that address the pain point.

I noticed that in the video you mentioned that your work has been integrated into the ath9k driver, so I am thinking if someone from your group could take a look at our driver and propose how to optimise it, that would be great.

dtaht, 2021-01-21:
Zero funding presently, no devs. It also went into the ath10k, mt76, and iwl. Yes, I'd love ot look at it.