open-sdr/openwifi #32: Detail document of register

littlelid, 2020-09-17:
I am walking through the code in ./driver/sdr.c, and I encounter some registers and I cannot figure out the meaning of each bit. 
e.g., the register read by tx_intf_api->TX_INTF_REG_PKT_INFO_read().
Although I notice that there are a few registers that have been explained in the README in ./doc folder,  I am wondering where I can get the more detailed explanation of other registers.

JiaoXianjun, 2020-09-17:
You need to figure out the register functionality in this way:

You can find TX_INTF_REG_PKT_INFO_read() in tx_intf.c, where a register address is: TX_INTF_REG_PKT_INFO_ADDR.
That address is defined in hw_def.h: 

#define TX_INTF_REG_PKT_INFO_ADDR                  (22*4)

This means slv_reg22 in tx_intf.v (in openwifi-hw project). Finally you will find out that this slv_reg22 register reads value from the module tx_status_fifo_i (tx_status_fifo.v). For its meaning, you need to understand the verilog code of the openwifi design.

Currently we don't have such a document yet that describes all registers.
We will do it in the future when we believe those registers have become necessary, clean and clear.

JiaoXianjun, 2020-09-17:
Would you please tell me more about you? For example, students? employee? University name or Company name? You could send the information to xianjun.jiao@ugent.be

People could try out the code without telling us, but when people seek help from us we prefer to know who are we helping.

Thanks!

littlelid, 2020-09-18:
Thank you，I will send the information via mail.

closed 2020-09-30 by JiaoXianjun