open-sdr/openwifi #34: Where do you set AD9361 trx_clock_chain_freq and update_rf_bandwidth

frestuc, 2020-10-05:
Hi all,

I was wondering where you set the RF bandwidth of the A9361?
I see that you commented out the lines where you can configure the AD9361 for 40MHz of bandwidth:

https://github.com/open-sdr/openwifi/blob/525f977bb91ce7072860ccf118331323ab817257/driver/sdr.c#L1949
https://github.com/open-sdr/openwifi/blob/525f977bb91ce7072860ccf118331323ab817257/driver/sdr.c#L1951

So I was wondering where you set that -- I couldn't find it anywhere in the code...

Thanks,
Francesco

JiaoXianjun, 2020-10-06:
Because the BW is static (finally we decide this and comment it out from .c), the setting remains in rf_init.sh

frestuc, 2020-10-06:
Got it. 

Another question: by setting
https://github.com/open-sdr/openwifi/blob/525f977bb91ce7072860ccf118331323ab817257/user_space/rf_init.sh#L38
Do you also get a 40Msps clock for `adc_clk` in your design? 
I see that you decimate the input to get a 20Msps signal here?

https://github.com/open-sdr/openwifi-hw/blob/98601d9cf69ffaaef3601d6eefdd6ab252c7e563/ip/rx_intf/src/adc_intf.v#L107

In general, do you get a clock frequency `adc_clk` of X when you set `in_voltage_sampling_frequency` to X?

JiaoXianjun, 2020-10-06:
in_voltage_sampling_frequency is sampling rate not the adc_clk.

adc_clk could be the same or not (higher). The relationship is complicated under different rate setting. You need to capture fpga ila (adc_clk and sampling data) to confirm the relationship.

frestuc, 2020-10-06:
Will do that -- but if it's not the same, how come you are able to decimate by simply taking one sample out of two in `adc_intf.v`?

JiaoXianjun, 2020-10-06:
adc_clk is just a clk. It doesn't mean there is a data every clock. adc_valid matters.

closed 2020-10-06 by frestuc

DatPhungThe, 2020-10-07:
Hi Xianjun and Frank,
I wonder what will happen if I set up the sampling rate is 20 MSps (e.g: `echo 20000000 >  in_voltage_sampling_frequency`) and do not decimate input samples in `adc_intf.v`.
Can we do that ?

Best Regards,

JiaoXianjun, 2020-10-07:
just do it!