open-sdr/openwifi #43: 2 hours running time related with viterbi

shy2028, 2020-11-25:
Hi, Xianjun

I have zynq706-fmcomms3. It is working well as AP
But the running time is 2 hours only, because of the Viterbi license as your below comment. 
("Note 2: After ~2 hours, the Viterbi decoder will halt (Xilinx Evaluation License). Just power cycle the board if it happens. (If the output of "./sdrctl dev sdr0 get reg rx 20" is always the same, it means the decoder halts")

I want to know what am I fix for 24 running times without a power cycle if I have the Viterbi license (not evaluation). 
Just new compiling after adopting Viterbi license in VIVADO?

JiaoXianjun, 2020-11-25:
Yes. If you have a valid license, just remove the old evaluation license from your vivado and add the new valid license to your vivado (via license manager from the vivado help menu, some name like this). Then you delete the old top level FPGA project with evaluation license and re-create the whole top level FPGA project (check openwifi-hw readme on github). 

Now you can generate the new FPGA bit-stream without the viterbi time limit. 

Then check the “Update FPGA” section of the openwifi readme (make sure your FPGA bit stream and sdk files are in correct location openwifi/openwifi-hw/..... please check “./sdk_update.sh board_name” in openwifi-hw readme), generate the new boot.bin which contains your new FPGA bit steam and update it into the board. 

Finally power cycle the board.

shy2028, 2020-11-26:
I changed Viterbi license evaluation to bought.
I checked readme in openwifi-hw directory.
But I've got the error messages for openwifi.tcl and sdk_updata.sh

In linux and In vivado openwifi.tcl is not working. 

-----------------openwifi.tcl --------------------------

**hazero@hazero:~/openwifi/openwifi-hw/boards$  vivado -mode tcl**

****** Vivado v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

Vivado% ls
WARNING: [Common 17-259] Unknown Tcl command 'ls' sending command to the OS shell for execution. It is recommended to use 'exec' to send the command to the OS shell.
adrv9361z7035  sdk_update.sh	       vivado_3309.backup.log  vivado.log   zc706_fmcs2   zed_fmcs2
adrv9364z7020  vivado_3309.backup.jou  vivado.jou	       zc702_fmcs2  zcu102_fmcs2

**Vivado% cd zc706_fmcs2/**
Vivado% ls
WARNING: [Common 17-259] Unknown Tcl command 'ls' sending command to the OS shell for execution. It is recommended to use 'exec' to send the command to the OS shell.
openwifi_rev.coe  openwifi.tcl	sdk  src

**Vivado% source ./openwifi.tcl**
# set  fd  [open  "openwifi_rev.coe"  

fatal: not a git repository (or any of the parent directories): .git
    while executing
"exec git log -1 --pretty=%h"
    invoked from within
"set HASHCODE [exec git log -1 --pretty=%h]"
    (file "./openwifi.tcl" line 23)

-----------------./sdk_update.sh---------------------------------

hazero@hazero:~/openwifi/openwifi-hw/boards$ ./sdk_update.sh zc706_fmcs2
+ BOARD_NAME=zc706_fmcs2
+ echo zc706_fmcs2
zc706_fmcs2
+ test -d zc706_fmcs2/sdk
+ echo 'Found zc706_fmcs2/sdk'
Found zc706_fmcs2/sdk
+ cp zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.sdk/system_top_hw_platform_0 zc706_fmcs2/sdk/ -rf
cp: cannot stat 'zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.sdk/system_top_hw_platform_0': No such file or directory
hazero@hazero:~/openwifi/openwifi-hw/boards$ vivado &

JiaoXianjun, 2020-11-26:
We can help on this. But before that, can you send me some introductions about yourself (company? University? Name? Etc) to Xianjun.jiao@ugent.be?

shy2028, 2020-11-26:
Okay, I sent the e-mail with some information.

JiaoXianjun, 2020-11-26:
At the beginning of openwifi.tcl, we try to get git revision by:

`set HASHCODE [exec git log -1 --pretty=%h]`

and put it into FPGA register.

But in your case:

`fatal: not a git repository (or any of the parent directories): .git`

So looks like you unzip the downloaded package instead of clone/pull the git repository?

You can either make a local git repository or change the beginning part of the openwifi.tcl to avoid the step. (You need to create a fake openwifi_rev.coe by your self. Normally it is created by our .tcl)

shy2028, 2020-12-01:
Hi, 
Sorry, I'm late. 

I changed the begining part of the openwif.tcl (generate openwifi_rev.coe parts are comented out)

After that vivado run.

> Vivado% source ./openwifi.tcl
# set origin_dir "."
# if { [info exists ::origin_dir_loc] } {
#   set origin_dir $::origin_dir_loc
# }
# set _xil_proj_name_ "openwifi_zc706_fmcs2"
# if { [info exists ::user_project_name] } {
#   set _xil_proj_name_ $::user_project_name
# }
# variable script_file
# set script_file "openwifi.tcl"
# proc print_help {} {
#   variable script_file
#   puts "\nDescription:"
#   puts "Recreate a Vivado project from this script. The created project will be"
#   puts "functionally equivalent to the original project for which this script was"
#   puts "generated. The script contains commands for creating a project, filesets,"
#   puts "runs, adding/importing sources and setting properties on various objects.\n"
#   puts "Syntax:"
#   puts "$script_file"
#   puts "$script_file -tclargs \[--origin_dir <path>\]"
#   puts "$script_file -tclargs \[--project_name <name>\]"
#   puts "$script_file -tclargs \[--help\]\n"
#   puts "Usage:"
#   puts "Name                   Description"
#   puts "-------------------------------------------------------------------------"
#   puts "\[--origin_dir <path>\]  Determine source file paths wrt this path. Default"
#   puts "                       origin_dir path value is \".\", otherwise, the value"
#   puts "                       that was set with the \"-paths_relative_to\" switch"
#   puts "                       when this script was generated.\n"
#   puts "\[--project_name <name>\] Create project with the specified name. Default"
#   puts "                       name is the name of the project from where this"
#   puts "                       script was generated.\n"
#   puts "\[--help\]               Print help information for this script"
#   puts "-------------------------------------------------------------------------\n"
#   exit 0
# }
# if { $::argc > 0 } {
#   for {set i 0} {$i < $::argc} {incr i} {
#     set option [string trim [lindex $::argv $i]]
#     switch -regexp -- $option {
#       "--origin_dir"   { incr i; set origin_dir [lindex $::argv $i] }
#       "--project_name" { incr i; set _xil_proj_name_ [lindex $::argv $i] }
#       "--help"         { print_help }
#       default {
#         if { [regexp {^-} $option] } {
#           puts "ERROR: Unknown option '$option' specified, please type '$script_file -tclargs --help' for usage info.\n"
#           return 1
#         }
#       }
#     }
#   }
# }
# set orig_proj_dir "[file normalize "$origin_dir/"]"
# create_project ${_xil_proj_name_} ./${_xil_proj_name_} -part xc7z045ffg900-2
# set proj_dir [get_property directory [current_project]]
# set obj [current_project]
# set_property -name "board_part" -value "xilinx.com:zc706:part0:1.2" -objects $obj
# set_property -name "compxlib.activehdl_compiled_library_dir" -value "$proj_dir/${_xil_proj_name_}.cache/compile_simlib/activehdl" -objects $obj
# set_property -name "compxlib.funcsim" -value "1" -objects $obj
# set_property -name "compxlib.ies_compiled_library_dir" -value "$proj_dir/${_xil_proj_name_}.cache/compile_simlib/ies" -objects $obj
# set_property -name "compxlib.modelsim_compiled_library_dir" -value "$proj_dir/${_xil_proj_name_}.cache/compile_simlib/modelsim" -objects $obj
# set_property -name "compxlib.overwrite_libs" -value "0" -objects $obj
[... 894 lines cut ...]
# set_property -name "view.orientation" -value "Horizontal" -objects $obj
# set_property -name "view.type" -value "Graph" -objects $obj
# move_dashboard_gadget -name {utilization_1} -row 0 -col 0
# move_dashboard_gadget -name {power_1} -row 1 -col 0
# move_dashboard_gadget -name {drc_1} -row 2 -col 0
# move_dashboard_gadget -name {timing_1} -row 0 -col 1
# move_dashboard_gadget -name {utilization_2} -row 1 -col 1
# move_dashboard_gadget -name {methodology_1} -row 2 -col 1
# current_dashboard default_dashboard 
default_dashboard
Vivado% exit

I am not goot at git.
How to make local git repository? 
------------------------------
But sdk file is not working. 
hazero@hazero:~/openwifi/openwifi-hw/boards$ ./sdk_update.sh zc706_fmcs2
+ BOARD_NAME=zc706_fmcs2
+ echo zc706_fmcs2
zc706_fmcs2
+ test -d zc706_fmcs2/sdk
+ echo 'Found zc706_fmcs2/sdk'
Found zc706_fmcs2/sdk
+ cp zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.sdk/system_top_hw_platform_0 zc706_fmcs2/sdk/ -rf
cp: cannot stat 'zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.sdk/system_top_hw_platform_0': No such file or directory

shy2028, 2020-12-01:
Hi, 

I've got new git from openwifi-hw site using git clone.
And try again followed README. after that Implemantation, generation Bitstream, export HW,  Launch SDK is success I think.
Because no error massage while run and export.

I run sdk file and I've got comment below.
But I don't understand how to use git commit command.
How to apply the new FPGA img to zynq board?

----------------------------------------------------------------------
hazero@hazero:~/test/openwifi-hw/boards$ ./sdk_update.sh zc706_fmcs2 
+ BOARD_NAME=zc706_fmcs2
+ echo zc706_fmcs2
zc706_fmcs2
+ test -d zc706_fmcs2/sdk
+ echo 'Found zc706_fmcs2/sdk'
Found zc706_fmcs2/sdk
+ cp zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.sdk/system_top_hw_platform_0 zc706_fmcs2/sdk/ -rf
+ git add zc706_fmcs2/sdk/system_top_hw_platform_0 zc706_fmcs2/sdk/system_top.ltx
+ cp zc706_fmcs2/openwifi_zc706_fmcs2/openwifi_zc706_fmcs2.runs/impl_1/system_top.ltx zc706_fmcs2/sdk/
+ git add zc706_fmcs2/sdk/system_top_hw_platform_0 zc706_fmcs2/sdk/system_top.ltx

--------------------------------------------------

hazero@hazero:~/test/openwifi-hw/boards$ git commit -a -m "new openwifi"

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'hazero@hazero.(none)')
hazero@hazero:~/test/openwifi-hw/boards$

weiliu1011, 2020-12-01:
Hi,
Good to hear that you succeeded to generate bit stream. 
Now please run "$OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR zc706_fmcs2" to create a new BOOT.BIN file. Check read me to set the environment variables such as OPENWIFI_DIR and XILINX_DIR first. If the command succeeded, you will find the BOOT.BIN file in $OPENWIFI_DIR/kernel_boot/boards/zc706_fmcs2/output_boot_bin/ folder. Copy the BOOT.BIN file to the BOOT partition of your SD card, boot the board to use the new hardware.
Br,
Wei

shy2028, 2020-12-04:
Hi, 

I made BOOT.BIN file and It working well

Thank you.

closed 2020-12-04 by JiaoXianjun

shy2028, 2020-12-04:
Hi. 
After making BOOT.BIN file, I've got some error when the device connect to the openwifi board.

The openwifi board transmission the beacon frame for over 2hours and I saw the SSID on the phone too.
But, the phone couldn't connect to the openwifi AP.

When I try to connect to the openwifi board using phone, the host PC terminal doesn't show any response message such as a starting account session, authenticated and association messages. 
Few socond later, It is connect I think because "No internet conncetion" notice under openwifi SSID in the phone.
(before ssh, I set the ethernet forwarding to board and confirmed the phone using internet when I using default boot.bin file) 

When I turn off the wifi in the phone while trying to connect or after connected, then host pc terminal shown the disassociated and the unauthenticated message.

Am I making Boot.bin file again for fix these problem?

JiaoXianjun, 2020-12-04:
This seems like an issue that is irrelevant to the viterbi license. Can you confirm it is relevant or not. If it is irrelevant, could you please create a new issue and describe how to reproduce it steadily?