open-sdr/openwifi #44: error in running openwifi ap with country_code=CN

charlesLuoc, 2020-11-28:
Hi, Xianjun
'openwifi' AP runs well on my zed board + FMCOMMS3 when i operate according to Section Quick start of README.md.
But after changing the country_code from BE to CN, I found the following errors:
"
155.617286: 155.617291: sdr0: IEEE 802.11 Configured channel (44) not found from the channel list of current mode (2) IEEE 802.11a
155.617592: 155.617601: sdr0: IEEE 802.11 Hardware does not support configured channel
"
The modified hostapd-openwifi_5G_try.txt and entire log are given in attachment.
[Serial_11-28-145518.log](https://github.com/open-sdr/openwifi/files/5610279/Serial_11-28-145518.log)
[hostapd-openwifi_5G_try.txt](https://github.com/open-sdr/openwifi/files/5610283/hostapd-openwifi_5G_try.txt)

weiliu1011, 2020-12-01:
Hi Charles,
I think this issue is not related to openwifi itself. Basically if you set country code to CN, if you run iw reg get, you will see below:
root@analog:~/openwifi_cap# iw reg get
country CN:
	(2402 - 2482 @ 40), (N/A, 20)
	(5735 - 5835 @ 40), (N/A, 30)
	(57240 - 59400 @ 2160), (N/A, 28)
	(59400 - 63720 @ 2160), (N/A, 44)
	(63720 - 65880 @ 2160), (N/A, 28)
If you set country code for US, you will see:
root@analog:~/openwifi_cap# iw reg get
country US:
	(2402 - 2472 @ 40), (3, 27)
	(5170 - 5250 @ 40), (3, 17)
	(5250 - 5330 @ 40), (3, 20), DFS
	(5490 - 5600 @ 40), (3, 20), DFS
	(5650 - 5710 @ 40), (3, 20), DFS
	(5735 - 5835 @ 40), (3, 30)
	(57240 - 63720 @ 2160), (N/A, 40)
By comparing the output, we see the channel 44 (5220 MHz) is not among the enabled channel for country CN, but it is enabled in US or BE. Further google results show that this is determined by regulation database file /lib/crda/regulatory.bin, This is a binary file read by linux, more information see https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb#mailing_list_for_regulatory_updates.

I think because the linux distribution is slightly old, so the regulatory database is not up to date. If you must use CN code, then you need to update the database correspondingly. 
Wei

charlesLuoc, 2020-12-06:
But for country_code CN,I changed channel to 159. Similar error occurs:

60.024549: 60.024554: sdr0: IEEE 802.11 Configured channel (159) not found from the channel list of current mode (2) IEEE 802.11a
60.024847: 60.024855: sdr0: IEEE 802.11 Hardware does not support configured channel
60.024917: Could not select hw_mode and channel. (-3)

JiaoXianjun, 2020-12-06:
Would you please check this part to add support (country, channel, etc) by yourself?
https://github.com/open-sdr/openwifi/tree/master/doc#Regulation-and-channel-config

closed 2021-01-28 by JiaoXianjun