open-sdr/openwifi #45: Setting ACK interval

lucabaldesi, 2020-12-02:
Hi,
first, thank you for this project, it is really amazing! I am experimenting with a ZC706 board plus FMCOMM2 and I would
like to change the ACK threshold time (SIFS interval).
I don't know verilog, but from the sources in the IP directory, it seems to me that SIFS is read from a register (e.g., "input wire [0:6] sifs_time").

In the driver sources (sdr.c:932, tx_intf.c:280) I think you set the SIFS register through a call to TX_INTF_REG_CTS_TOSELF_WAIT_SIFS_TOP_write which results in a 10 us SIFS time.

My questions are:
 - Can I change the ACK (SIFS) interval by simply changing the value in the driver files (sdr.c, tx_intf.c)?
 - why this interval is set both during initialization (tx_intf.c) and during each transmission (sdr.c)?
 - Would you be interested in extending the sdrctl tool to tweak the SIFS interval?

Thanks,
Luca

weiliu1011, 2020-12-02:
Hi Luca,
As far as I am aware, you can indeed adjust the SIFS interval. However the registers are different than the ones you found. 
To adjsut SIFS waiting time to send an ack, you need to change value in bit 30:16 of reg 18 of XPU. To adjust SIFS waiting time to receive an ack, you need to change bit 30:16 of reg 16  for 2.4 GHz, or bit 30:16 of reg 17 for 5 GHz. 

The lines 927 - 930 in sdr.c are the current value of the registers. You can find how the bit of these registers are used in xpu.v between line 265 - 269. 

If you read the lines in sdr.c you will see the value written in the registers is not exactly 10 us or 16 us. From receive ack point of view, we first wait to for a signal that indicate valid header is received, and then wait for the fcs valid of the ack frame. So the threshold longer than the time the the wireless medium is being idle. 

Currently you can read and write any register with sdrctl, for instance to read the value of register 8 in xpu, simply do the following:
./sdrctl dev sdr0 get reg xpu 8

To write a value
./sdrctl dev sdr0 set reg xpu 8 130   

Unfortunately I did not write this piece of code, so I also don't know how exactly the register 16 and 17 values are calculated, however I think you can adjust the value based on the existing ones, first read the value and then try making it shorter or longer. Please be careful that there is also a scale factor. We assume in software that the hardware counter runs at 10 MHz. 
Br,
Wei

lucabaldesi, 2020-12-04:
Hi Wei,
thank you for your reply. It is pretty cool sdrctl already allows the change of registry values.
In the tx_control.v:199 the value of the registry is read, multiplied by a constant and saved in another variable (recv_ack_sig_valid_timeout_top_scale).

If I understand the code correctly, the variable ack_timeout_count is incremented by one at each cycle until either a valid signal is received, (line 383) or it reaches the value of recv_ack_sig_valid_timeout_top_scale, i.e., the timeout expires, (line 392).

The scaling factor, COUNT_SCALE is defined as NUM_CLK_PER_US/ASSUMED_COUNTER_CLK_MHZ, where NUM_CLK_PER_US can be 250, 200 or 100 (depending on the board) and ASSUMED_CLK_MHZ is 10 (MHz).
Hence, recv_ack_sig_valid_timeout_top_scale = 505*100/10 = 5,050. I would assume now that this value is in nano seconds (i.e., 5 us)

The tx interface board options are: "high", "slow" and "ultra_scale" (board_def.v). Since I am using a ZC706+FMCOMM2 which has a Zynq7000 chip, do I have a "slow" FPGA?

Best,
Luca

JiaoXianjun, 2020-12-04:
It is “low”, not “slow”. 
For 7z020 (zed, zc702, adrv9364), please use low. 
For the rest (except zcu102), please use high.

High means high volume fpga. (7z035, 7z045 (zc706 uses this))

Low mean low volume fpga.

weiliu1011, 2020-12-05:
Hi Luca,
As Xianjun already said, indeed zc706 +fmcs2 falls into the "high" category, the rest of your understanding seems correct to me, except that the unit of the top_scale depends on the clock speed. As you already pointed out, the ack_timeout_count increases by each clock, so top scale's unit depends on the speed of the clock. 
Best,
Wei

lucabaldesi, 2020-12-07:
Hi and thank you for your help,
I am still a bit confused about the interpretation of these variables; as a recap:

+-------------------------------------------------------------------------------------------------------+
|                 name                                     |     type              |   file             | Abbrv.  |
+-------------------------------------------------------------------------------------------------------+
| recv_ack_sig_valid_timeout_top           |     registry 16    | xpul.v           |     t      |
| recv_ack_sig_valid_timeout_top_scale | FPGA variable | tx_control.v  |     T     |
| ASSUMED_COUNTER_CLK_MH        | FPGA constant | board_def.v |     f      |
| NUM_CLK_PER_US                            | FPGA constant | board_def.v |     F     |
| COUNT_SCALE                                   | FPGA constant | board_def.v |     C     |
+-------------------------------------------------------------------------------------------------------+

I added abbreviation symbols to simplify the presentation.
C = F/f is the scale factor for the interval t.
T = t*C is the actual number of ticks that FPGA has to count before declaring the ACK gone lost (SIFS but expressed in ticks).

I expect the ticks to be counted by the FPGA, hence accumulated in the variable ack_timeout_count (file tx_control.v:382) with frequency F.
In the code, t=505 and, assumimg f=10MHz, F=200MHz, then T = 10100.
Counting the ticks: T*(1/F) = 50.5 useconds. <-- resulting ACK time treshold (SIFS)

The time t is expressed in ticks with respect to f, not F; why is that the case?
If the tick duration depends on F, why defining t with respect to f?
Is it okay the SIFS results so big?

Best,
Luca

weiliu1011, 2020-12-07:
Hi Luca,
I believe the expression of time scale in f not F is a design choice. Open wifi supports several boards, the hardware runs on different clock speed. Using 10 MHz assumption simplifies the software as it always writes the same value, but hardware running on different boards can interpret this value according to the clock speed.

Regarding the value you calculated that is longer than SIFS in the standard, intuitively, when you receive the signal field (header), you have already received STF, LTF. STF + LTF + signal is already more than 20 us. Then we add 10 us of the standard. I have 30 us explained. The rest is explained in the comment of sdr.c: "extra 300 clocks are needed when rx core fall into fake ht detection phase (rx mcs 6M)". This is some margin to handle special situation of receiver thinks it is decoding a 802.11n packet. 

Best,
Wei

lucabaldesi, 2020-12-08:
Thank you all for your help and patience, it is much clearer now.

Best,
Luca

closed 2020-12-08 by lucabaldesi