open-sdr/openwifi #46: No 'associated' message in host PC terminal and 'no internet' in the phone after copy  .BIN file (made myself)

shy2028, 2020-12-04:
Hi.
After making BOOT.BIN file, I've got some error when the device connect to the openwifi board.

The openwifi board transmission the beacon frame for over 2hours and I saw the SSID on the phone too.
But, the phone couldn't connect to the openwifi AP.

When I try to connect to the openwifi board using phone, the host PC terminal doesn't show any response message such as a starting account session, authenticated and association messages.
Few socond later, It is connect I think because "No internet conncetion" notice under openwifi SSID in the phone.
(before ssh, I set the ethernet forwarding to board and confirmed the phone using internet when I using default boot.bin file)

When I turn off the wifi in the phone while trying to connect or after connected, then the host pc terminal shown the disassociated and the unauthenticated message.

Just before, I made the Boot.bin file again. But still same. No associated message in PC, No internet in Phone

JiaoXianjun, 2020-12-04:
So can you confirm it happens always. And after you change the boot.bin back, it disappeares?

shy2028, 2020-12-04:
When I boot using the file I created, the same problem mentioned above continues to occur.
When I revert to the previous one (download from https://users.ugent.be/~xjiao/openwifi-1.1.0-taiyuan-6-32bit.img.xz), the internet connection problem disappears.

JiaoXianjun, 2020-12-04:
I think you need to make sure your boot.bin building procedure is correct.
I suggest you clone a clean openwifi-hw code, use the evaluation viterbi license instead of your own's, build the fpga project and finally build the boot.bin.
Then use this boot.bin instead of the one in our image to see whether there is a problem.
And please also record and show us all the commands/operations since you do the clean clone openwifi-hw, then we can analyze the command sequence to see whether there is an issue.

shy2028, 2020-12-07:
Hi, 

Here is the commands/operation record
It is start from downloading openwifi-hw directory.
It is in order Linux -> Vivado -> Linux command.
My viterbi evaluation license is already expired. So, I make the boot file using my own license.  (I attach the vivado license status)

   ---------------linux--------------------------------------------
> hazero@hazero:~/openwifi git clone https://github.com/open-sdr/openwifi-hw
> Cloning into 'openwifi-hw'...
> remote: Enumerating objects: 2530, done.
> remote: Total 2530 (delta 0), reused 0 (delta 0), pack-reused 2530
> Receiving objects: 100% (2530/2530), 223.48 MiB | 6.14 MiB/s, done.
> Resolving deltas: 100% (1433/1433), done.
> hazero@hazero:~/openwifi cd openwifi-hw/
> hazero@hazero:~/openwifi/openwifi-hw git status 
> On branch master
> Your branch is up to date with 'origin/master'.
> 
> nothing to commit, working tree clean
> 
> hazero@hazero:~/openwifi/openwifi-hw git checkout b3bd6e2
> Note: checking out 'b3bd6e2'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> HEAD is now at b3bd6e2 track latest dot11zynq
> hazero@hazero:~/openwifi/openwifi-hw git status 
> HEAD detached at b3bd6e2
> nothing to commit, working tree clean
> hazero@hazero:~/openwifi/openwifi-hw git fetch
> hazero@hazero:~/openwifi/openwifi-hw git pull
> You are not currently on a branch.
> Please specify which branch you want to merge with.
> See git-pull(1) for details.
> 
>     git pull <remote> <branch>
> 
> hazero@hazero:~/openwifi/openwifi-hw source /opt/Xilinx/Vivado/2018.3/settings64.sh
> hazero@hazero:~/openwifi/openwifi-hw export SDCARD_DIR=/media/hazero/
> hazero@hazero:~/openwifi/openwifi-hw export XILINX_DIR=/opt/Xilinx
> hazero@hazero:~/openwifi/openwifi-hw export OPENWIFI_DIR=//home/hazero/openwifi 
> hazero@hazero:~/openwifi/openwifi-hw export BOARD_NAME=zc706_fmcs2
> hazero@hazero:~/openwifi/openwifi-hw 
> hazero@hazero:~/openwifi/openwifi-hw ./prepare_adi_lib.sh $XILINX_DIR
> $XILINX_DIR is found!
> ++ git submodule init adi-hdl
> Submodule 'adi-hdl' (https://github.com/analogdevicesinc/hdl.git) registered for path 'adi-hdl'
> ++ git submodule update adi-hdl
> Cloning into '/home/hazero/openwifi/openwifi-hw/adi-hdl'...
> Submodule path 'adi-hdl': checked out 'f61d9707eb0a62533efd6facab59ab2444da94c9'
> ++ cd ./adi-hdl/library
> ++ git reset --hard f61d9707eb0a62533efd6facab59ab2444da94c9
> HEAD is now at f61d9707 ad_fmclidar1_ebz: Update the IO constraints to revB
> ++ cd /home/hazero/openwifi/openwifi-hw
> hazero@hazero:~/openwifi/openwifi-hw$ ./prepare_adi_board_ip.sh $XILINX_DIR $BOARD_NAME
> $XILINX_DIR is found!
> ./adi-hdl/projects/fmcomms2/zc706/
> ++ source /opt/Xilinx/Vivado/2018.3/settings64.sh
> +++ source /opt/Xilinx/DocNav/.settings64-DocNav.sh
> ++++ '[' -n /opt/Xilinx/Model_Composer/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'
> ++++ export PATH=/opt/Xilinx/DocNav:/opt/Xilinx/Model_Composer/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
> ++++ PATH=/opt/Xilinx/DocNav:/opt/Xilinx/Model_Composer/2018.3/bin:/opt/Xilinx/SDK/2018.3/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/opt/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/opt/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/opt/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/opt/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/opt/Xilinx/Vivado/2018.3/bin:/opt/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
[... 654 lines cut ...]
>     ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.
> 
> + rm ./boards/zc706_fmcs2/build_boot_bin -rf
[... cut ...]
> hazero@hazero:~/openwifi/kernel_boot/boards$ cd zc706_fmcs2/
> hazero@hazero:~/openwifi/kernel_boot/boards/zc706_fmcs2$ ls
> devicetree.dts  output_boot_bin  u-boot.elf
> hazero@hazero:~/openwifi/kernel_boot/boards/zc706_fmcs2$ cd output_boot_bin/
> hazero@hazero:~/openwifi/kernel_boot/boards/zc706_fmcs2/output_boot_bin$ ls
> BOOT.BIN  fsbl.elf  system.hdf  system_top.bit  u-boot.elf  zynq.bif

> -------------------------------------------------------------------------------------------------------------------------------

JiaoXianjun, 2020-12-10:
to me the boot.bin generation method is ok.

Can you show us some dmesg info during initial connection by the debug method: https://github.com/open-sdr/openwifi/tree/master/doc#debug-methods

You can also “cat /proc/interrupts” to see whether the number of tx and rx interrupts are growing.

Above method to check whether the tx and rx are running.

shy2028, 2020-12-11:
Here is the dmesg info during inital cooncetion, run fosdem...

Welcome to Linaro 14.04 (GNU/Linux 4.14.0-g4fea7c58ad92-dirty armv7l)

 * Documentation:  https://wiki.analog.com/ https://ez.analog.com/
Last login: Thu Jan  1 00:00:09 1970

### ----------------dmesg after login-------------------

> root@analog: dmesg -c
> Booting Linux on physical CPU 0x0
> Linux version 4.14.0-g4fea7c58ad92-dirty (jxj@jxj-xps) (gcc version 7.3.1 20180314 (Linaro GCC 7.3-2018.04-rc3)) 1 SMP PREEMPT Thu Jun 11 22:51:11 CEST 2020
> CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
> CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
> OF: fdt: Machine model: Xilinx Zynq ZC706
> OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
> Memory policy: Data cache writealloc
> cma: Reserved 128 MiB at 0x38000000
> On node 0 totalpages: 262144
> free_area_init_node: node 0, pgdat c0c54000, node_mem_map ef7f7000
>   Normal zone: 1536 pages used for memmap
>   Normal zone: 0 pages reserved
>   Normal zone: 196608 pages, LIFO batch:31
>   HighMem zone: 65536 pages, LIFO batch:15
> random: fast init done
> percpu: Embedded 16 pages/cpu @ef7c6000 s33036 r8192 d24308 u65536
> pcpu-alloc: s33036 r8192 d24308 u65536 alloc=16*4096
> pcpu-alloc: [0] 0 [0] 1 
> Built 1 zonelists, mobility grouping on.  Total pages: 260608
> Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait cpuidle.off=1
> PID hash table entries: 4096 (order: 2, 16384 bytes)
> Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
> Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
> Memory: 895976K/1048576K available (7168K kernel code, 339K rwdata, 2388K rodata, 1024K init, 165K bss, 21528K reserved, 131072K cma-reserved, 131072K highmem)
> Virtual kernel memory layout:
>     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
>     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
>     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
>     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
>     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
>     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
>       .text : 0xc0008000 - 0xc0800000   (8160 kB)
>       .init : 0xc0b00000 - 0xc0c00000   (1024 kB)
>       .data : 0xc0c00000 - 0xc0c54ea0   ( 340 kB)
>        .bss : 0xc0c5a894 - 0xc0c83f7c   ( 166 kB)
> Preemptible hierarchical RCU implementation.
> 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
> 	Tasks RCU enabled.
> RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
> NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
> efuse mapped to f0802000
> slcr mapped to f0804000
> L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
> L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
> L2C-310 erratum 769419 enabled
> L2C-310 enabling early BRESP for Cortex-A9
> L2C-310 full line of zeros enabled for Cortex-A9
> L2C-310 ID prefetch enabled, offset 1 lines
> L2C-310 dynamic clock gating enabled, standby mode enabled
> L2C-310 cache controller enabled, 8 ways, 512 kB
> L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
> zynq_clock_init: clkc starts at f0804100
> Zynq clock init
> sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
> clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
> Switching to timer-based delay loop, resolution 3ns
> clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
> timer 0 at f080c000, irq=17
> Console: colour dummy device 80x30
> Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
[... 1279 lines cut ...]
>  39:          0          0     GIC-0  75 Level     f8003000.dmac
>  40:          0          0     GIC-0  40 Level     f8007000.devcfg
>  46:          0          0     GIC-0  53 Level     e0002000.usb
>  47:          0          0     GIC-0  41 Edge      f8005000.watchdog
>  48:         96          0     GIC-0  90 Level     41600000.i2c
>  49:          0          0     GIC-0  91 Edge      43000000.dma
>  50:          0          0     GIC-0  89 Edge      7c400000.dma
>  51:          0          0     GIC-0  88 Edge      7c420000.dma
>  53:        834          0     GIC-0  62 Edge      sdr,rx_pkt_intr
>  55:          0          0     GIC-0  66 Edge      sdr,tx_itrpt1
>  56:       2975          0     GIC-0  67 Level     xilinx-dma-controller
>  57:          0          0     GIC-0  68 Level     xilinx-dma-controller
>  58:          0          0     GIC-0  63 Level     xilinx-dma-controller
>  59:        834          0     GIC-0  64 Level     xilinx-dma-controller
>  60:          0          0  zynq-gpio  58 Edge      Left
>  61:          0          0  zynq-gpio  60 Edge      Right
>  62:          0          0  zynq-gpio  59 Edge      Select
> IPI1:          0          0  Timer broadcast interrupts
> IPI2:       8980      15769  Rescheduling interrupts
> IPI3:          5          4  Function call interrupts
> IPI4:          0          0  CPU stop interrupts
> IPI5:          0          0  IRQ work interrupts
> IPI6:          0          0  completion interrupts
> Err:          0
> root@analog:/openwifi

JiaoXianjun, 2020-12-11:
This means some kernel panic like problem.

Strange.

Could you share us the boot.bin for zc706? Or do you have any other type of board?

------------[ cut here ]------------
WARNING: CPU: 1 PID: 0 at kernel/softirq.c:161 __local_bh_enable_ip+0xbc/0xf4
Modules linked in: 8021q arc4 sdr(O) xpu(O) openofdm_rx(O) openofdm_tx(O) rx_intf(O) ad9361_drv(O) tx_intf(O) xilinx_dma mac80211 cfg80211 mwipcore mwipcore_iio_streaming mwipcore_dma_streaming mathworks_ip_common
CPU: 1 PID: 0 Comm: swapper/1 Tainted: G O 4.14.0-g4fea7c58ad92-dirty 1
Hardware name: Xilinx Zynq Platform
[] (unwind_backtrace) from [] (show_stack+0x10/0x14)
[] (show_stack) from [] (dump_stack+0x8c/0xa0)
[] (dump_stack) from [] (__warn+0xe8/0x100)
[] (__warn) from [] (warn_slowpath_null+0x20/0x28)
[] (warn_slowpath_null) from [] (__local_bh_enable_ip+0xbc/0xf4)
[] (__local_bh_enable_ip) from [] (ieee80211_beacon_add_tim+0x4c/0x60 [mac80211])
[] (ieee80211_beacon_add_tim [mac80211]) from [] (__ieee80211_beacon_get+0x3ac/0x45c [mac80211])
[] (__ieee80211_beacon_get [mac80211]) from [] (ieee80211_beacon_get_tim+0x30/0x12c [mac80211])
[] (ieee80211_beacon_get_tim [mac80211]) from [] (openwifi_hr_timer+0x14c/0x53c [sdr])
[] (openwifi_hr_timer [sdr]) from [] (__hrtimer_run_queues+0x10c/0x198)
[] (__hrtimer_run_queues) from [] (hrtimer_interrupt+0xa0/0x214)
[] (hrtimer_interrupt) from [] (twd_handler+0x30/0x40)
[] (twd_handler) from [] (handle_percpu_devid_irq+0x78/0x138)
[] (handle_percpu_devid_irq) from [] (generic_handle_irq+0x24/0x34)
[] (generic_handle_irq) from [] (__handle_domain_irq+0x5c/0xb4)
[] (__handle_domain_irq) from [] (gic_handle_irq+0x4c/0x90)
[] (gic_handle_irq) from [] (__irq_svc+0x6c/0xa8)
Exception stack(0xef05ff78 to 0xef05ffc0)
ff60: 00000001 00000000
ff80: 00000001 c01179e0 ffffe000 c0c050f4 c0c050a8 c0b47a48 c0c54c60 00000000
ffa0: 00000000 00000000 2ec95000 ef05ffc8 c0108480 c0108484 60000013 ffffffff
[] (__irq_svc) from [] (arch_cpu_idle+0x38/0x3c)
[] (arch_cpu_idle) from [] (do_idle+0x16c/0x1f8)
[] (do_idle) from [] (cpu_startup_entry+0x18/0x1c)
[] (cpu_startup_entry) from [<0010174c>] (0x10174c)
---[ end trace 57f9519ce1d1808a ]---

shy2028, 2020-12-14:
Here is my boot.bin file.
[output_boot_bin.zip](https://github.com/open-sdr/openwifi/files/5686462/output_boot_bin.zip)

This file working well Boot file. It if from sd card IMG in open-sdr git page(2hours running).
[BOOT.zip](https://github.com/open-sdr/openwifi/files/5686467/BOOT.zip)

I have only zc706 boards....

weiliu1011, 2020-12-14:
Hi,
I used your BOOT.BIN file and I don't observe any issue in booting, or association. I believe your BOOT.BIN file is okay. I used iphone and dell laptop xps 15 to test. 
Apart from your BOOT.BIN file, the rest of the booting files (uImage, devicetree.dtb) and driver files are generated from this commit (e8237378f8d27daa937ede00688c0a603b99c71b) in the master branch. 
Maybe you can try to test with alternative client devices? and also double check the driver version? 
I will update you whether your BOOT.BIN still works after the two hour time lap.
Br,
Wei

weiliu1011, 2020-12-14:
Hi, your BOOT.BIN has been running for 2h10min on my local setup. So viterbi license is okay as well. I would remove the binaries so that people don't start using it for free.

shy2028, 2020-12-14:
Hi,
I'm really thanks for checking the boot bin file to see if it works over 2 hours.
And I removed the zip file, actuarlly I don't know the remove the zip files so I delete the comment.

I have some questions about your answer.

Then you host PC terminal shows the association messages? 
And the board running well as AP?

What is the meaning of try to test with alternative client devices? 
Is it means using another phone? or a laptop? 
Before I gave the Boot file, I used two devices as client.
one is iphone, another is thinkpad laptop x1.
The host pc connect to with AP(board) is desktop.

As your 3 days ago answer, you said kernel panic problem.
So, you check the boot file of mine. (It is working well) then, how fix the kernel panic problem?
After downloading the latest version of the master branch and rerun the board??

I can not 100% understand the commit (e823737) comment.

Driver version? which driver?  Linux mac80211 subsystem? (how to upgrad?) 
what is your driver version?

weiliu1011, 2020-12-14:
Hi,
Yes,  I used my del laptop as client, it is running windows 10. So nothing was shown, but it is connected and I can ping the AP (openwifi board). 

I was indeed thinking that you could try with different commercial devices as clients. However if both your laptop and phone can connect to the board when it was using the prebuild SD card image, and they failed to connect when you switch to the new BOOT.BIN file, I think this means there is a miss match between hardware and software.

Therefore, can you checkout openwifi master branch, and checkout that commit. Then all files will be set to that commit. Please check read me (https://github.com/open-sdr/openwifi#Update-Driver) to update driver.  This is the openwifi driver, not mac80211 subsystem. 
 Let us know how it goes,
Wei

shy2028, 2020-12-15:
Hi, 
I changed the openwifi all files to openwifi master branch, And update driver, boot.bin file also changed of my Boot file.
After that the board works well as AP.
The iPhone can connect to the board and use the internet at same time.

After two hours later, I'll update about board are still work or not.

Thank you.

shy2028, 2020-12-15:
Hi, 
The board is running as AP for over 3 hours without any problems.
The Internet connection is also good. 

I simply checked throughput through fast.com, but not fast. (7.5Mbps) 
To increase throughput, Do I change the settings in hostapd-openwifi.conf?

Thank you

weiliu1011, 2020-12-15:
Okay good to know that it is working now!
Regarding speed, can you enable regular printing (check https://github.com/open-sdr/openwifi/tree/master/doc#Debug-methods) and then show us some output of dmesg? 
I am suspecting your link quality is not good. openwifi frontend does not have power amplifier, so make sure your client device is close to the board could increase speed.
Also, can you run iperf between client and AP? fast.com will also depending on the ethernet port and some other stuff. So better only look at the wireless link.
best,
Wei

shy2028, 2020-12-16:
Hi, 
here is the output of dmesg and iperf.

### output of dmesg
> root@analog:~# ./openwifi/fosdem.sh 
> stop: Unknown instance: 
> num_arg  0
> last_input xxxxxxxxx
> test_mode 0
> check xilinx_dma module is loaded or not
> xilinx_dma is loaded!
> check tx_intf module is loaded or not
> tx_intf is loaded!
> rmmod: ERROR: Module ad9361_drv is in use
> insmod: ERROR: could not insert module ad9361_drv.ko: File exists
> check ad9361_drv module is loaded or not
> ad9361_drv is loaded!
> set RF frontend
> Found openwifi_ad9361_fir.ftr
> + test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
> + test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
> + test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
> + cd /sys/bus/iio/devices/iio:device2/
> + set +x
> FIR Rx: 32,1 Tx: 32,1
> 1
> 0
> rx0 agc fast_attack
> fast_attack
> fast_attack
> rx1 agc fast_attack
> fast_attack
> fast_attack
> 40000000
> 17500000
> 40000000
> 37500000
> rssi
> 118.75 dB
> 120.75 dB
> rx0 gain to 70
> 62.000000 dB
> sh: echo: I/O error
> 62.000000 dB
> rx1 gain to 70
> 62.000000 dB
> sh: echo: I/O error
> 62.000000 dB
> tx0 gain -89dB
> -89.750000 dB
> -89.000000 dB
> tx1 gain 0dB
> -89.750000 dB
> 0.000000 dB
> check rx_intf module is loaded or not
> rx_intf is loaded!
> check openofdm_tx module is loaded or not
> openofdm_tx is loaded!
> check openofdm_rx module is loaded or not
> openofdm_rx is loaded!
> check xpu module is loaded or not
> xpu is loaded!
> rmmod: ERROR: Module sdr is not currently loaded
> insert sdr.ko test_mode=0
> check sdr module is loaded or not
> sdr is loaded!
> the end
> sdr,sdr openwifi_tx:   69bytes  6M FC0050 DI00ac addr1/2/3:c6519a289f18/66554433228b/66554433228b SC2f80 flag4000001e retr1 ack0 prio0 q0 wr55 rd54
> sdr,sdr openwifi_tx:   69bytes  6M FC0050 DI00ac addr1/2/3:f4d108a654be/66554433228b/66554433228b SC3030 flag4000001e retr1 ack0 prio0 q0 wr2 rd1
> sdr,sdr openwifi_tx:   69bytes  6M FC0050 DI00ac addr1/2/3:66fc6de84b50/66554433228b/66554433228b SC3170 flag4000001e retr1 ack0 prio0 q0 wr22 rd21
[... 1161 lines cut ...]
> root@analog:~/openwifi# iperf -s -i 1
>  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> Server listening on TCP port 5001
> TCP window size: 85.3 KByte (default)
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> [  4] local 192.168.13.1 port 5001 connected with 192.168.13.8 port 56724
> [ ID] Interval       Transfer     Bandwidth
> [  4]  0.0- 1.0 sec  2.57 MBytes  21.6 Mbits/sec
> [  4]  1.0- 2.0 sec  2.43 MBytes  20.4 Mbits/sec
> [  4]  2.0- 3.0 sec  2.54 MBytes  21.3 Mbits/sec
> [  4]  3.0- 4.0 sec  2.58 MBytes  21.6 Mbits/sec
> [  4]  4.0- 5.0 sec  2.39 MBytes  20.1 Mbits/sec
> [  4]  5.0- 6.0 sec  2.75 MBytes  23.1 Mbits/sec
> [  4]  6.0- 7.0 sec  2.82 MBytes  23.7 Mbits/sec
> [  4]  7.0- 8.0 sec  2.80 MBytes  23.5 Mbits/sec
> [  4]  8.0- 9.0 sec  2.75 MBytes  23.1 Mbits/sec
> [  4]  9.0-10.0 sec  2.72 MBytes  22.8 Mbits/sec
> [  4]  0.0-10.1 sec  26.6 MBytes  22.1 Mbits/sec
> 
> [ ID] Interval       Transfer     Bandwidth
> [  3]  0.0-10.0 sec  32.1 MBytes  26.9 Mbits/sec
> 
> 
> 
>

weiliu1011, 2020-12-16:
Hi
From dmesg output I see that your link quality is good between openwifi AP and client. Also iperf up and down link throughput is good. I believe the speed limitation is elsewhere, not due to phy or mac of openwifi. 
Wei

JiaoXianjun, 2021-01-28:
Due to the limited TX power of ad9361 (for openwifi configuration, it is around 0 or several dBm), it won't reach long distance or reach good throughput at a distance. You need PA (Power Amplifier) to boost the distance. Such as:

https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/ad-trxboost1-ebz.html#eb-overview

closed 2021-01-28 by JiaoXianjun