open-sdr/openwifi #49: failed to Build openwifi Linux img from scratch

wolkenmeer911, 2020-12-14:
After compiling adi-linux and openwifi with zc706 project, I follow the step in Build openwifi Linux img from scratch. However, I can't run Linux successfully when burning it in SD card

JiaoXianjun, 2020-12-14:
Hello,

Yes we can help. But firstly would you please send me (xianjun.jiao@ugent.be) information about you? where are you from? Student? (MS? PhD?), or employee of which company? Thanks!

JiaoXianjun, 2020-12-14:
Can you tell us:

1 the git/revision of openwifi/openwifi-hw you are using. 

2 the error message.

wolkenmeer911, 2020-12-14:
Hi，

Thanks for your replying,
I'm sorry that I don't know what the first question means. I saw openwifi-hw @ b3bd6e2 in the github if you mean it?
And the error message is that hostapd:command not found; webfsd:command not foune.

JiaoXianjun, 2020-12-14:
what is the output of:

cd ~/openwifi && ./post_config.sh

wolkenmeer911, 2020-12-14:
> what is the output of:
>
> cd ~/openwifi &amp;&amp; ./post_config.sh

There's no such file in openwifi, but there's a post_config.sh in user_space, do you mean this file?

JiaoXianjun, 2020-12-14:
You need to run that in board. Can you check the openwifi directory on board? You need to check and follow this section carefully:

https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch

wolkenmeer911, 2020-12-14:
I successfully ran the post_config.sh. When running fosdem.sh it occurs Interface sdr0 wasn't started.

JiaoXianjun, 2020-12-14:
Can you share the full log of fosdem.sh?

wolkenmeer911, 2020-12-15:
Sorry for replying you so late, I save the log as a txt file.

wolkenmeer911, 2020-12-16:
Sorry to bother you, I've solved the problem. I think the cause of it is that I didn't downald the complete project code due to the poor net. Apologize！

closed 2020-12-16 by wolkenmeer911