open-sdr/openwifi #53: What is the approximate max-disance of openwifi that packages do not loss?

CaptainiFalcon, 2021-01-20:
When I put my laptop(with Linux) 100cm away from the  sender (board : zcu102_fmcs2), no package loss.
[image]

200cm away from the  sender (board : zcu102_fmcs2)：
[image]

300cm away from the  sender (board : zcu102_fmcs2)：
[image]

I don't know if there is something wrong with my experimental environment or that's the way it should be.

Thanks！

JiaoXianjun, 2021-01-20:
tcp or udp? can you share our your test command?

CaptainiFalcon, 2021-01-20:
> tcp or udp? can you share our your test command?

iperf -c 192.168.13.5 -u -i 1 -t 30 -l 100 -b 400k
and
iperf -s -u -i 1

JiaoXianjun, 2021-01-26:
Due to the limited TX power of ad9361 (for openwifi configuration, it is around 0 or several dBm), it won't reach long distance. You need PA (Power Amplifier) to boost the distance. Such as: 

https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/ad-trxboost1-ebz.html#eb-overview

closed 2021-01-28 by JiaoXianjun

winfelin, 2023-11-20:
If we use an ideal PA, can openwifi theoretically reach 5km within the line of sight?