open-sdr/openwifi #56: Cost-related questions

Alpxcx, 2021-02-21:
Hi. I am a begining self learner in telecom. I have read relative articles and found this project interesting. However, I'm concerned about a few points.
1. 802.11n and the future ax with 40 Mhz is difficult or impossible to achieve on ad9361/3/4 as they only allow 61.44 MSPS of sampling rate, is this correct?
2. Will 802.11ax be fully implemented on XC7Z020? I wonder because of the FPGA size.
3. Will 802.11ax be supported on ad9361/3/4? If I understand the specs correctly the maximun data rate is 122.88 MHz per channel which may not be enough with 1024 QAM.
All questions are related to the estimate cost of an 802.11ax openwifi dev board.

JiaoXianjun, 2021-02-22:
Hello,
1. 61.44Msps is enough for 40MHz channel bandwidth, because it is IQ (dual channel) sampling. If you have only single channel (ADC), indeed you will need 80Msps to handle 40MHz.

2. 802.11ax is possible to be implemented in 7020 FPGA, if we select a tiny/minimum sub-set of the standard.

3. 1024QAM indeed could be difficult for ad9361/3/4, but it is possible still. ad9361 signal quality is quite good. We don't know, until we try it (1024QAM).

Alpxcx, 2021-02-24:
Thank you! I may consider purchasing such a dev board when the major work on 802.11ax is completed.

closed 2021-02-24 by JiaoXianjun