open-sdr/openwifi #58: I Q的采样信息在PC上找不到？采样信息传到PC是会自动保存到一个文件夹吗？

18845725896peng, 2021-03-16:
您好，我在用"insmod side_ch.ko" and "./side_ch_ctl g"开始进行IQ采样，可以在终端看到触发条件被满足，对IQ进行了采样。但是打开另一个终端，找不到采样信息保存的文件夹，显示文件夹不存在。想咨询您IQ采样信息流入PC是会自动创建文件夹保存吗？
      期待您的回复！

JiaoXianjun, 2021-03-16:
Hello,

Thanks for reaching us. Would you please describe your question in English next time? In this way we can help more people worldwide.

Now I translate your question:

"I am using insmod side_ch.ko and ./side_ch_ctl g to do IQ sampling. The trigger condition is met as I can see in the terminal, and IQ is being captured. But I can't find the IQ file when I open another terminal. It says that the folder doesn't exist. May I ask that will a folder be created and IQ be saved automatically when the IQ is being streammed to PC?"

To answer your question, I need to know the following information:
1. Which application note are you following (There are two application notes about IQ capture)? Can you put the link here?
2. Would you please paste each step of your operation (for instance, in the PC terminal or in the ssh login of the board) and its outputs here? Please also point out which directory/file are you looking for?

Then we will have a better understanding of your question and try to find out what is the issue.

Thanks!

18845725896peng, 2021-03-17:
thank you for your reply，
  I use zc706 and ad_fmcomms2-ebz,  I explain your two questions

 1.https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq.md
 2.in the terminal，the command I entered is as follows：
    ssh root@192.168.10.122
   cd openwifi
   ./monitor_ch.sh sdr0 11
   insmod side_ch.ko iq_len_init=8187
   ./side_ch_ctl g
   its outputs:
   loop 64 side info count 6
   loop 128 side info count 8
  loop 192 side info count 14
  loop 256 side info count 18
  loop 320 side info count 25
  loop 384 side info count 28
  loop 448 side info count 35
  loop 512 side info count 41
  loop 576 side info count 47
  loop 640 side info count 52
  loop 704 side info count 59
  loop 768 side info count 62
  loop 832 side info count 66
  loop 896 side info count 72
  loop 960 side info count 75
   this is normal,and then I open the another terminal ,
   input:cd openwifi/user_space/side_ch_ctl_src
   output:/openwifi/user_space/side_ch_ctl_src$ 
  input: python3 iq_capture.py
  output:Assume iq_len = 8187! (Max UDP 65507 bytes; (65507/8)-1 = 8187)
^CUser quit
close()
Traceback (most recent call last):
  File "iq_capture.py", line 121, in <module>
    side_info_fd.close()
NameError: name 'side_info_fd' is not defined
I think this has something to do with setting the ip address to 192.168.10.5，so I change the IP as 192.168.10.1 and the problem is solved.

18845725896peng, 2021-03-17:
hello,
    I am  a student now , I have diffculty in working with this project and I need to learn too many things to best understand this project,thanks for your help

18845725896peng, 2021-03-17:
In the process I found another problem，numpy must be updated to a higher version .The terminal's error report about iq_capture.py is gone when I upgrade the version of numpy to 1.20.1(1.13before)

JiaoXianjun, 2021-03-17:
Glad to know that you solve it already!

closed 2021-03-17 by JiaoXianjun