open-sdr/openwifi #59: Is auto acknowledgements enabled in monitor mode?

Matheus-Garbelini, 2021-03-23:
Dear all, It appears acknowledgments are configured to a default value and never changed again:
https://github.com/open-sdr/openwifi/blob/266e956643d4d12434932d60df55164f7cd49611/driver/sdr.c#L1024-L1025

**When packets are injected in monitor mode, will the SDR wait for acknowledgments and retransmit if necessary based on the target MAC address?**

JiaoXianjun, 2021-03-23:
Hello,

The ACK and re-transmission state machine is in FPGA xpu (tx_control.v). Regarding ACK, there are two things:

1. After FPGA transmits a packet, wait for ACK. If ACK is received in time, report to ARM --> Linux. If ACK is not received in time, do re-transmission according to the packet re-transmission configuration, and report finally to ARM --> Linux (how many times re-transmission is tried, etc)

Above behavior is configured per packet in driver (sdr.c, openwifi_tx() ):
```
dma_reg = ( (( ((prio<<(NUM_BIT_MAX_NUM_HW_QUEUE+NUM_BIT_MAX_PHY_TX_SN))|(ring->bd_wr_idx<<NUM_BIT_MAX_NUM_HW_QUEUE)|queue_idx) )<<18)|(retry_limit_hw_value<<14)|(pkt_need_ack<<13)|num_dma_symbol );
```
If pkt_need_ack is 0, FPGA won't wait for ACK and do re-transmission. retry_limit_hw_value controls the maximum number of re-transmission. 
You need to check these two parameters in your mode (monitor mode), to foresee the FPGA behavior.

2. After FPGA receives a packet, FPGA will send back ACK according to the packet type and CRC result. In tx_control.v:
```
if ( fcs_valid && (is_data||is_management||is_blockackreq||is_blockack||is_pspoll||(is_rts&&(!cts_torts_disable))) 
               && (self_mac_addr==addr1)) // send ACK will not back to this IDLE until the last IQ sample sent.
  begin
      tx_control_state  <= SEND_ACK; //we also send cts (if rts is received) in SEND_ACK status
  end
```
Indeed this part is not affected by the monitor mode currently, so even in monitor mode FPGA still sends ACK for the incoming packet (if the condition is met).

In all the ACK related processing (if happens), the target MAC address is checked/set.

Should we disable the FPGA sending ACK in monitor mode? Or add a switch (register) to give user more control from driver/strctl? If you could elaborate more about your use case and consideration, we can do the implementation (shouldn't be heavy I think).

JiaoXianjun, 2021-03-25:
If no further question, I will close this issue.

Matheus-Garbelini, 2021-03-25:
Hi @JiaoXianjun thanks a lot for your detailed answer. Actually, for my application auto acknowledgment responses during monitor mode is already what we want. Generally, commercial Wi-Fi hardware disables auto acknowledgments in monitor mode under Linux and we need to enable it by setting a hardware register bit, so OpenWi-FI already has this enabled by default.

However, I see that to maintain behavior compatibility with Linux, adding an ack register so we can disable or enable this behavior would be ideal since in monitor mode we may want to be as transparent as possible and not auto-reply nor auto-retry when injecting packets.

Regards.

JiaoXianjun, 2021-03-25:
To disable (or confirm it is already the case from Linux config per packet) “waiting for ack and auto retry”, you can set (or print) pkt_need_ack  and retry_limit_hw_value to 0 in driver.

To disable “auto reply ack” in monitor mode, we will add a register for user to control.

Matheus-Garbelini, 2021-03-25:
Thanks a lot, @JiaoXianjun . And once again, thanks a lot for this great project. I'm going to close the issue and would suggest that the information of auto acknowledgments may be added to the monitor mode documentation so people know of these side effects when running OpenWi-Fi in monitor mode.

JiaoXianjun, 2021-04-05:
Hello,

We have lots of update just now in driver and FPGA. Two are related to the WiFi fuzzing:
- the ACK disable register is added. 
- a bug of retransmission when packet is injected under monitor mode is fixed.
  - According to the comments on "struct ieee80211_tx_rate" in mac80211.h, mac80211 should tell driver that the packet should be transmitted "up to" how many times. However when packet is injected under monitor mode, this "up to" number is 0, which is abnormal/make-no-sense. In this case our original code uses 15 instead of 0. Now the code uses 1 instead of 0. You can also override it.

See more explanations: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md

You can use the latest SD card image directly or pull the latest driver/FPGA.

Matheus-Garbelini, 2021-04-06:
@JiaoXianjun thanks a lot, this will definitely suit our use.

closed 2021-04-06 by JiaoXianjun

JiaoXianjun, 2021-05-03:
> Hi @JiaoXianjun thanks a lot for your detailed answer. Actually, for my application auto acknowledgment responses during monitor mode is already what we want. Generally, commercial Wi-Fi hardware [...]

Hello, just curious, would you please tell us about which COTS WiFi support the auto ACK in monitor mode? (by the register setting as you mentioned)

Thanks!