open-sdr/openwifi #60: fail to run fosdem.sh

qgzboy, 2021-04-08:
hello:
I'm want to use "Update FPGA" function, but I'm following the quick start and meet the error shown as below when running ./fosdem.sh
root@analog:~/openwifi# hostapd -dd hostapd-openwifi.conf &sleep 5
[1] 2239
random: Trying to read entropy from /dev/random
Configuration file: hostapd-openwifi.conf
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Using driver-based off-channel TX
nl80211: interface sdr0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Try mode change after setting interface down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Mode change succeeded while interface is down
nl80211: Setup AP(sdr0) - device_ap_sme=0 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0xb3dc0
nl80211: Register frame type=0xb0 nl_handle=0xb3dc0 match=
nl80211: Register frame type=0x0 nl_handle=0xb3dc0 match=
nl80211: Register frame type=0x20 nl_handle=0xb3dc0 match=
nl80211: Register frame type=0xa0 nl_handle=0xb3dc0 match=
nl80211: Register frame type=0xc0 nl_handle=0xb3dc0 match=
nl80211: Register frame type=0xd0 nl_handle=0xb3dc0 match=
nl80211: Register frame type=0x40 nl_handle=0xb3dc0 match=
nl80211: Add own interface ifindex 3
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
sdr0: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code 00, new country code BE 
Continue interface setup after channel list update
ctrl_iface not configured!
RTM_NEWLINK: ifi_index=3 ifname=sdr0 operstate=2 linkmode=0 ifi_flags=0x1002 ()
nl80211: Ignore interface down event since interface sdr0 is up
RTM_NEWLINK: ifi_index=3 ifname=sdr0 operstate=2 linkmode=0 ifi_flags=0x1003 ([UP])
nl80211: Event message available
nl80211: Drv Event 36 (NL80211_CMD_REG_CHANGE) received for sdr0
nl80211: Regulatory domain change
sdr0: Event CHANNEL_LIST_CHANGED (30) received
Channel list updated - continue setup
nl80211: Regulatory information - country=BE
nl80211: 2402-2482 @ 40 MHz 20 mBm
nl80211: 5170-5250 @ 40 MHz 20 mBm
nl80211: 5250-5330 @ 40 MHz 20 mBm
nl80211: 5490-5710 @ 40 MHz 27 mBm
nl80211: 57240-65880 @ 2160 MHz 40 mBm
nl80211: Added 802.11b mode based on 802.11g information
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=20 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=38 freq=5190 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=42 freq=5210 MHz max_tx_power=20 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=20 dBm
[... 119 lines cut ...]
[<c061128c>] (linkwatch_event) from [<c0133358>] (process_one_work+0x1d8/0x410)
[<c0133358>] (process_one_work) from [<c0134064>] (worker_thread+0x4c/0x598)
[<c0134064>] (worker_thread) from [<c0138de0>] (kthread+0x14c/0x154)
[<c0138de0>] (kthread) from [<c0107a30>] (ret_from_fork+0x14/0x24)
Code: e597371c e5911000 e0833182 e5910040 (e5934000) 
---[ end trace e23cc7b0cfa0c6e2 ]---
Kernel panic - not syncing: Fatal exception in interrupt
CPU1: stopping
CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D    O    4.14.0-gb6e379910a11-dirty #1
Hardware name: Xilinx Zynq Platform
[<c010edbc>] (unwind_backtrace) from [<c010b6f4>] (show_stack+0x10/0x14)
[<c010b6f4>] (show_stack) from [<c06f1e04>] (dump_stack+0x8c/0xa0)
[<c06f1e04>] (dump_stack) from [<c010d4dc>] (ipi_cpu_stop+0x64/0x74)
[<c010d4dc>] (ipi_cpu_stop) from [<c010dc1c>] (handle_IPI+0x6c/0x7c)
[<c010dc1c>] (handle_IPI) from [<c0101464>] (gic_handle_irq+0x8c/0x90)
[<c0101464>] (gic_handle_irq) from [<c010c20c>] (__irq_svc+0x6c/0xa8)
Exception stack(0xef05ff78 to 0xef05ffc0)
ff60:                                                       00000001 00000000
ff80: 00000001 c01179e0 ffffe000 c0c050f4 c0c050a8 c0b47a48 c0c54fa0 00000000
ffa0: 00000000 00000000 2ec97000 ef05ffc8 c0108480 c0108484 60040013 ffffffff
[<c010c20c>] (__irq_svc) from [<c0108484>] (arch_cpu_idle+0x38/0x3c)
[<c0108484>] (arch_cpu_idle) from [<c0150e14>] (do_idle+0x16c/0x1f8)
[<c0150e14>] (do_idle) from [<c0151148>] (cpu_startup_entry+0x18/0x1c)
[<c0151148>] (cpu_startup_entry) from [<0010174c>] (0x10174c)
---[ end Kernel panic - not syncing: Fatal exception in interrupt

qgzboy, 2021-04-08:
my board is adrv9361z7035

JiaoXianjun, 2021-04-08:
Thanks for trying openwifi. Could you please send your information to my email xianjun.jiao@ugent.be to let us know who we are helping? Thanks!

qgzboy, 2021-04-08:
Hello,
Thanks for reaching me, I'm a communication specialty student of nanchuang university in China,just interested on Verilog and wireless.
my email: qgzboy@163.com

JiaoXianjun, 2021-04-09:
Can you try our latest SD card image and the latest quick start? In the latest quick start, we use fosdem-11ag.sh

qgzboy, 2021-04-10:
hello:
After I do the following steps：

”Setup environment variables (use absolute path):
export XILINX_DIR=your_Xilinx_directory
export OPENWIFI_DIR=your_openwifi_directory
export BOARD_NAME=your_board_name
Get the latest FPGA bitstream from openwifi-hw, generate BOOT.BIN and transfer it on board via ssh channel:
$OPENWIFI_DIR/user_space/get_fpga.sh $OPENWIFI_DIR

For Zynq 7000:
$OPENWIFI_DIR/user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME“

generate new BOOT.BIN file,but after  I replace old BOOT.BIN file and run ‘fosdem-11ag.sh‘, above questions still exist。

Thanks!

JiaoXianjun, 2021-04-10:
What is the output of get_fpga.sh?
What is the output of boot_bin_gen.sh?

qgzboy, 2021-04-10:
the execution of get_fpga.sh is fetch the latest openwifi-hw contents.
the execution of boot_bin_gen.sh generate a new BOOT.BIN in folder of "output_boot_bin".

JiaoXianjun, 2021-04-10:
I know. 

You need to paste your command line output of those command.

Because you generate your own BOOT.BIN，this could be the issue. So I need to take a look at the procedure/output of generating BOOT.BIN (get_fpga.sh and boot_bin_gen.sh)。If you run quick start directly after flashing your SD card with our image (don't update FPGA), fosdem-11ag.sh it should work out of the box.

qgzboy, 2021-04-11:
I already send a email to you

JiaoXianjun, 2021-04-11:
can you run quick start directly after flashing your SD card with our image (don't update FPGA and copy boot.bin). In this case fosdem-11ag.sh is successful or not?

JiaoXianjun, 2021-04-11:
I just repeated the steps you took, and everything is OK at my side.
Would you please paste your complete output of fosdem-11ag.sh?

qgzboy, 2021-04-12:
hello:
Here is the commands/operation record:

root@ubuntu:/home/openwifi-master# export XILINX_DIR=/home/Xilinx

root@ubuntu:/home/openwifi-master# export BOARD_NAME=adrv9361z7035root@ubuntu:/home/openwifi-master# export OPENWIFI_DIR=/home/openwifi-master

root@ubuntu:/home/openwifi-master# ./user_space/get_fpga $OPENWIFI_DIR

bash: ./user_space/get_fpga: No such file or directory

root@ubuntu:/home/openwifi-master# ./user_space/get_fpga.sh $OPENWIFI_DIR

$OPENWIFI_DIR is found!

+ cd /home/openwifi-master/

+ git submodule init openwifi-hw

+ git submodule update openwifi-hw

+ cd openwifi-hw

+ git checkout master

Already on 'master'

Your branch is up-to-date with 'origin/master'.

+ git pull

Already up-to-date.

+ cd /home/openwifi-master

root@ubuntu:/home/openwifi-master# ./user_space/boot_bin_gen.sh $OPENWIFI_DIR $XILINX_DIR $BOARD_NAME

$OPENWIFI_DIR is found!

$XILINX_DIR is found!

$BOARD_NAME is found!

+ source /home/Xilinx/SDK/2018.3/settings64.sh

++ source /home/Xilinx/DocNav/.settings64-DocNav.sh

+++ '[' -n /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'

+++ export PATH=/home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

+++ PATH=/home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

++ source /home/Xilinx/Vivado/2018.3/.settings64-Vivado.sh

+++ export XILINX_VIVADO=/home/Xilinx/Vivado/2018.3

+++ XILINX_VIVADO=/home/Xilinx/Vivado/2018.3

+++ '[' -n /home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'

+++ export PATH=/home/Xilinx/Vivado/2018.3/bin:/home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

+++ PATH=/home/Xilinx/Vivado/2018.3/bin:/home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

++ source /home/Xilinx/SDK/2018.3/.settings64-SDK_Core_Tools.sh

+++ '[' -n /home/Xilinx/Vivado/2018.3/bin:/home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin ']'

+++ export PATH=/home/Xilinx/SDK/2018.3/bin:/home/Xilinx/SDK/2018.3/gnu/microblaze/lin/bin:/home/Xilinx/SDK/2018.3/gnu/arm/lin/bin:/home/Xilinx/SDK/2018.3/gnu/microblaze/linux_toolchain/lin64_le/bin:/home/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-linux-gnueabi/bin:/home/Xilinx/SDK/2018.3/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/home/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-linux/bin:/home/Xilinx/SDK/2018.3/gnu/aarch64/lin/aarch64-none/bin:/home/Xilinx/SDK/2018.3/gnu/armr5/lin/gcc-arm-none-eabi/bin:/home/Xilinx/SDK/2018.3/tps/lnx64/cmake-3.3.2/bin:/home/Xilinx/Vivado/2018.3/bin:/home/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
[... 520 lines cut ...]

Building '/hw_0'

+ cp ./boards/adrv9361z7035/build_boot_bin/build/sdk/fsbl/Release/fsbl.elf ./boards/adrv9361z7035/output_boot_bin/fsbl.elf

+ cp ./boards/adrv9361z7035/build_boot_bin/build/sdk/hw_0/system_top.bit ./boards/adrv9361z7035/output_boot_bin/system_top.bit

+ cd ./boards/adrv9361z7035/output_boot_bin

+ bootgen -arch zynq -image zynq.bif -o BOOT.BIN -w

****** Xilinx Bootgen v2018.3

  **** Build date : Dec  6 2018-23:41:49

    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

+ rm ./boards/adrv9361z7035/build_boot_bin -rf

+ cd /home/openwifi-master

root@ubuntu:/home/openwifi-master# 

Finally,generate a new BOOT.BIN in folder of "output_boot_bin"
Thanks!

qgzboy, 2021-04-12:
The following contents is complete output of fosdem-11ag.sh:

U-Boot 2014.07-dirty (Nov 20 2014 - 17:05:21)

Board:  Xilinx Zynq
I2C:   ready
DRAM:  ECC disabled 512 MiB
MMC:   zynq_sdhci: 0
SF: Detected N25Q256A with page size 256 Bytes, erase size 4 KiB, total 32 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   Gem.e000b000
Hit any key to stop autoboot:  0 
Device: zynq_sdhci
Manufacturer ID: 9c
OEM: 534f
Name: USD00 
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.7 GiB
Bus Width: 4-bit
reading uEnv.txt
392 bytes read in 8 ms (47.9 KiB/s)
Loaded environment from uEnv.txt
Importing environment from SD ...
Running uenvcmd ...
Copying Linux from SD to RAM...
reading uImage
4553040 bytes read in 445 ms (9.8 MiB/s)
reading devicetree.dtb
20715 bytes read in 24 ms (842.8 KiB/s)
reading uramdisk.image.gz
** Unable to read file uramdisk.image.gz **
## Booting kernel from Legacy Image at 03000000 ...
   Image Name:   Linux-4.14.0-gb6e379910a11-dirty
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4552976 Bytes = 4.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02a00000
   Booting using the fdt blob at 0x2a00000
   Loading Kernel Image ... OK
   Loading Device Tree to 1ed1b000, end 1ed230ea ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Booting Linux on physical CPU 0x0
Linux version 4.14.0-gb6e379910a11-dirty (jxj@xps15) (gcc version 7.3.1 20180314 (Linaro GCC 7.3-2018.04-rc3)) #1 SMP PREEMPT Thu Dec 17 17:41:33 CET 2020
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Analog Devices ADRV9361-Z7035 (Z7035/AD9361)
OF: fdt: earlycon: stdout-path /amba@0/uart@E0001000 not found
Memory policy: Data cache writealloc
cma: Reserved 128 MiB at 0x38000000
random: fast init done
percpu: Embedded 16 pages/cpu @ef7c8000 s33036 r8192 d24308 u65536
Built 1 zonelists, mobility grouping on.  Total pages: 260608
Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait cpuidle.off=1
PID hash table entries: 4096 (order: 2, 16384 bytes)
Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Memory: 895988K/1048576K available (7168K kernel code, 340K rwdata, 2376K rodata, 1024K init, 160K bss, 21516K reserved, 131072K cma-reserved, 131072K highmem)
Virtual kernel memory layout:
[... 653 lines cut ...]
[<c0611268>] (linkwatch_event) from [<c0133358>] (process_one_work+0x1d8/0x410)
[<c0133358>] (process_one_work) from [<c0134064>] (worker_thread+0x4c/0x598)
[<c0134064>] (worker_thread) from [<c0138de0>] (kthread+0x14c/0x154)
[<c0138de0>] (kthread) from [<c0107a30>] (ret_from_fork+0x14/0x24)
Code: e597371c e5911000 e0833182 e5910040 (e5934000) 
---[ end trace 7d5289f881892316 ]---
Kernel panic - not syncing: Fatal exception in interrupt
CPU1: stopping
CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D    O    4.14.0-gb6e379910a11-dirty #1
Hardware name: Xilinx Zynq Platform
[<c010edbc>] (unwind_backtrace) from [<c010b6f4>] (show_stack+0x10/0x14)
[<c010b6f4>] (show_stack) from [<c06f1e04>] (dump_stack+0x8c/0xa0)
[<c06f1e04>] (dump_stack) from [<c010d4dc>] (ipi_cpu_stop+0x64/0x74)
[<c010d4dc>] (ipi_cpu_stop) from [<c010dc1c>] (handle_IPI+0x6c/0x7c)
[<c010dc1c>] (handle_IPI) from [<c0101464>] (gic_handle_irq+0x8c/0x90)
[<c0101464>] (gic_handle_irq) from [<c010c20c>] (__irq_svc+0x6c/0xa8)
Exception stack(0xef05ff78 to 0xef05ffc0)
ff60:                                                       00000001 00000000
ff80: 00000001 c01179e0 ffffe000 c0c050f4 c0c050a8 c0b47a48 c0c54fa0 00000000
ffa0: 00000000 00000000 2ec97000 ef05ffc8 c0108480 c0108484 60030013 ffffffff
[<c010c20c>] (__irq_svc) from [<c0108484>] (arch_cpu_idle+0x38/0x3c)
[<c0108484>] (arch_cpu_idle) from [<c0150e14>] (do_idle+0x16c/0x1f8)
[<c0150e14>] (do_idle) from [<c0151148>] (cpu_startup_entry+0x18/0x1c)
[<c0151148>] (cpu_startup_entry) from [<0010174c>] (0x10174c)
---[ end Kernel panic - not syncing: Fatal exception in interrupt

JiaoXianjun, 2021-04-12:
According to your print, you are not using our latest SD card image. In this case, if you update FPGA, it will mean that you try to use the latest FPGA with the old driver and Linux kernel. You have 3 ways to try:

1. Use our latest SD card image. Please download it from the openwifi README. or
2. Not only update FPGA, but also driver: https://github.com/open-sdr/openwifi#update-driver or
3. Build your SD card image locally from scratch: https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch

qgzboy, 2021-04-13:
Hello：
Thanks very much，I tried second method，program normar  working。

closed 2021-04-13 by qgzboy