open-sdr/openwifi #61: Build Vivado Project

eljuligallego, 2021-04-09:
Hello,
after some time I decided to rebuild openwifi for new boards (zynq7020 mostly) but discover I'm not able now.

What should be the process to rebuild the Vivado project?.

I tried to compile it running the openwifi.tcl inside board projects but have some errors in project generation not able to read run some commands(set properties and reading the bd for now).

I have to do anything else?.

I see the Vivado version is the 2018.3 but I have the 2018.2 so expect minor changes, maybe not.

Thanks.

JiaoXianjun, 2021-04-09:
Hello,

We only do the full test on the vivado 2018.3. So I would suggest that you install 2018.3 and re-do everything from the scratch (with out latest code and following readme of openwifi-hw). If there is still issue, please paste the step/command you use and the error output here.

eljuligallego, 2021-04-09:
Thanks Jiao for reply,
was true for some reason the vivado 2018.2 does work to generate the project.
just follow the readme at openwifi-hw and the project was generated but I get stuck in the synthesis for now.

the openofdm_rx IP says IP license not found so can't update this core and give errors when compile. I guess I have to regenerate the openofdm_rx IP...

eljuligallego, 2021-04-09:
the vitervi IP license is installed

JiaoXianjun, 2021-04-09:
make sure the viterbi  license is not expired (0.5 year), and then delete the project, re-open vivado, re-generate again everything.

eljuligallego, 2021-04-09:
yes, was expired, expecting still works but not. regenerating is not working. going to start from scratch...

eljuligallego, 2021-04-10:
compiled, Thanks for the support!.

closed 2021-04-10 by eljuligallego