open-sdr/openwifi #62: zcu102 failed to connect fmcomms3

lihuanjing, 2021-04-09:
hello, i run your image on zcu102 , but it cannot connect to my fmcomms3, i have tried the image in two boards and all failed
the failed log is posted below
root@analog:~# ~/openwifi/fosdm-11ag.sh
-bash: /root/openwifi/fosdm-11ag.sh: No such file or directory
root@analog:~# ~/openwifi/fosdem-11ag.sh
hostapd: no process found
stop: Unknown instance:
num_arg  0
last_input xxxxxxxxx
test_mode 0
check tx_intf module is loaded or not
tx_intf is loaded!
check ad9361_drv module is loaded or not
ad9361_drv is loaded!
set RF frontend
Found openwifi_ad9361_fir.ftr
+ test -f /sys/bus/iio/devices/iio:device0/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device1/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device2/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device3/in_voltage_rf_bandwidth
+ test -f /sys/bus/iio/devices/iio:device4/in_voltage_rf_bandwidth
+ echo Can not find in_voltage_rf_bandwidth!
Can not find in_voltage_rf_bandwidth!
+ echo Check log to make sure ad9361 driver is loaded!
Check log to make sure ad9361 driver is loaded!
+ exit 1
check rx_intf module is loaded or not
rx_intf is loaded!
check openofdm_tx module is loaded or not
openofdm_tx is loaded!
check openofdm_rx module is loaded or not
openofdm_rx is loaded!
check xpu module is loaded or not
xpu is loaded!
rmmod: ERROR: Module sdr is not currently loaded
insert sdr.ko test_mode=0
check sdr module is loaded or not
sdr is loaded!
the end
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.0-gb6e379910a11 (jxj@xps15) (gcc version 7.3.1 20180314 (Linaro GCC 7.3-2018.04-rc3)) #1 SMP Mon Apr 5 14:05:03 CEST 2021
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: ZynqMP ZCU102 Rev1.0
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options '115200n8')
[    0.000000] bootconsole [cdns0] enabled
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x0000000070000000
[    0.000000] On node 0 totalpages: 1048576
[    0.000000]   DMA zone: 7168 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524288 pages, LIFO batch:31
[    0.000000]   Normal zone: 7168 pages used for memmap
[    0.000000]   Normal zone: 524288 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] random: fast init done
[    0.000000] percpu: Embedded 20 pages/cpu @ffffffc87ff61000 s44952 r8192 d28776 u81920
[    0.000000] pcpu-alloc: s44952 r8192 d28776 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: enabling workaround for ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1034240
[    0.000000] Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait cpuidle.off=1 root=/dev/mmcblk0p2 rw rootwait
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] software IO TLB [mem 0x6bfff000-0x6ffff000] (64MB) mapped at [ffffffc06bfff000-ffffffc06fffefff]
[... 575 lines cut ...]
[  215.247804] sdr,xpu dev_probe xpu_driver_api_inst 0x00ba7408
[  215.247806] sdr,xpu dev_probe             xpu_api 0x00ba7408
[  215.247807] sdr,xpu dev_probe reset tsf timer
[  215.347819] sdr,xpu dev_probe XPU_REG_TSF_RUNTIME_VAL_LOW_read 1 33000 99998us
[  215.347822] sdr,xpu dev_probe succeed!
[  215.347823] sdr,xpu hw_init mode 1
[  215.347835] sdr,xpu hw_init mode XPU_NORMAL
[  215.347839] sdr,xpu hw_init err 0
[  215.860604]
[  215.860610] sdr,sdr openwifi_dev_probe: match!
[  215.860643] sdr,sdr custom_match_spi_dev ad9361-phy ad9361-phy 1
[  215.860647] sdr,sdr bus_find_device ad9361-phy: ad9361. driver_data pointer           (null)
[  215.860649] sdr,sdr find_dev ad9361-phy failed. dev.driver_data == NULL
[  215.867291] sdr,sdr: probe of fpga-axi@0:sdr failed with error -12
SIOCSIFADDR: No such device
sdr0: ERROR while getting interface flags: No such device
SIOCADDRT: File exists
stop: Unknown instance:
isc-dhcp-server start/running, process 4664
Configuration file: hostapd-openwifi-11ag.conf
Could not read interface sdr0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
Could not read interface sdr0 flags: No such device
nl80211 driver initialization failed.
hostapd_free_hapd_data: Interface sdr0 wasn't started

JiaoXianjun, 2021-04-09:
Could because your fmcomms3 firmware issue? maybe try other RF card? (another fmcomms3/4/2)

You have this error: ad9361 spi1.0: ad9361_probe : Unsupported PRODUCT_ID 0xFF
```
[   50.503755] sdr,tx_intf hw_init mode TX_INTF_BW_20MHZ_AT_N_10MHZ_ANT1
[   50.503760] sdr,tx_intf hw_init err 0
[   51.167702] ad9361 spi1.0: ad9361_probe : enter (ad9361)
[   51.170639] ad9361 spi1.0: ad9361_probe : Unsupported PRODUCT_ID 0xFF
[   52.230503]
[   52.230508] sdr,rx_intf dev_probe match!
[   52.230528] sdr,rx_intf dev_probe io start 0xa0004000 end 0xa0004fff name /fpga-axi@0/rx_intf@a0004000 flags 0x00000200 desc 0x00000000
```
Normally is should be: ad9361_probe : AD936x Rev 2 successfully initialized

```
[   57.774555] sdr,tx_intf hw_init mode TX_INTF_BW_20MHZ_AT_N_10MHZ_ANT1
[   57.774560] sdr,tx_intf hw_init err 0
[   58.439653] ad9361 spi1.0: ad9361_probe : enter (ad9361)
[   58.659982] ad9361 spi1.0: ad9361_probe : AD936x Rev 2 successfully initialized
[   59.469344] cf_axi_adc 99020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x99020000 mapped to 0xffffff8009200000, probed ADC AD9361 as MASTER
[   59.489536] cf_axi_dds 99024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER (9.01.b) at 0x99024000 mapped to 0xffffff800906a000, probed DDS AD9361
[   66.671930] 
[   66.671935] sdr,rx_intf dev_probe match!
[   66.671956] sdr,rx_intf dev_probe io start 0xa0004000 end 0xa0004fff name /fpga-axi@0/rx_intf@a0004000 flags 0x00000200 desc 0x00000000

```

JiaoXianjun, 2021-04-17:
Have you resolved the issue?

lihuanjing, 2021-04-17:
The problem has been solved. The voltage of ZCU102 HPC0 is 0.0V, so the serial port is not active. We downloaded SCUI.exe and configured the voltage of HPC0.

JiaoXianjun, 2021-04-17:
Good to know!

closed 2021-04-17 by JiaoXianjun