open-sdr/openwifi #64: AIFS

Shangjie-Yang, 2021-04-13:
Hello,
first, thank you for this project,it provided me with a good platform to learn！

My questions are:
· Whether the time slice written is TXOP？（txop=（end-start）？）
· According to the EDCA，there has 4 queue for different business，and there should be 4 AIFS to backoff meanwhile，I would
like to know how to do the 4 queue backoff ？

Thanks，
Yang

weiliu1011, 2021-04-13:
Hi Shangjie,
1/ The time slice is not TXOP. TXOP is the duration that the station can access channel to send multiple frames after the initial CSMA procedure.  We don't disable CSMA procedure in a time slice, so we still sense the channel after a frame is transmitted to send another one.  However there is a register to configure the CSMA procedure, if you disable CSMA in a specific time slice for a specific queue, and there is no overlapping between the queues' time slice, then to my understanding you achieve TXOP.

2/ We now only support DIFS, not AIFS, though the 4 queue can have different contention windows. Please take a look at the openwifi_conf_tx function (https://github.com/open-sdr/openwifi/blob/master/driver/sdr.c#L1414) to understand how the contention window for different queues are set. 

It is possible to add AIFS, though before we further discuss, would you please introduce a bit yourself? Like company/university name, email, which country are your from, what is your role (student/employee) etc.? Can you please send the information to wei.liu@ugent.be? We would like to know who we are helping. 

Best,
Wei

Shangjie-Yang, 2021-04-13:
Thank you for your help and patience.

Best,
Yang

closed 2021-04-13 by Shangjie-Yang