open-sdr/openwifi #65: Unable to connect to WiFi

pengshitao, 2021-04-15:
Use openwifi as client to connect other AP ，After following the steps，Repeat output：“ sdr0: CTRL-EVENT-SCAN-STARTED
” 。And the WiFi scan was fruitless。

pengshitao, 2021-04-15:
root@analog:~/openwifi# wpa_supplicant -Dnl80211 -i sdr0 -c /root/openwifi/wpa-connect.conf
Successfully initialized wpa_supplicant
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
^Csdr0: CTRL-EVENT-TERMINATING
root@analog:~/openwifi# wpa_cli
wpa_cli v2.1
Copyright (c) 2004-2014, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Selected interface 'sdr0'

Interactive mode

> scan
OK
<3>CTRL-EVENT-SCAN-STARTED
<3>CTRL-EVENT-SCAN-RESULTS
> scan_results
bssid / frequency / signal level / flags / ssid
<3>CTRL-EVENT-SCAN-STARTED
<3>CTRL-EVENT-SCAN-RESULTS
<3>CTRL-EVENT-SCAN-STARTED
<3>CTRL-EVENT-SCAN-RESULTS
<3>CTRL-EVENT-SCAN-STARTED
<3>CTRL-EVENT-SCAN-RESULTS
> q
root@analog:~/openwifi# iwlist sdr0 scanning
sdr0      No scan results

root@analog:~/openwifi#

JiaoXianjun, 2021-04-15:
Hello,

Thanks for reaching us. We could take a look. Before that, would you please let us know whom we are helping? Could you send your information (name, organization/university/company, etc) from your organization email to my email xianjun.jiao@ugent.be?

Thanks!

pengshitao, 2021-04-15:
> Hello,
> 
> Thanks for reaching us. We could take a look. Before that, would you please let us know whom we are helping? Could you send your information (name, organization/university/company, etc) from your organization email to my email [xianjun.jiao@ugent.be](mailto:xianjun.jiao@ugent.be)?
> 
> Thanks!

pengshitao, 2021-04-15:
> > Hello,
> > Thanks for reaching us. We could take a look. Before that, would you please let us know whom we are helping? Could you send your information (name, organization/university/company, etc) from your organization email to my email [xianjun.jiao@ugent.be](mailto:xianjun.jiao@ugent.be)?
> > Thanks!

Not belong to any organizations, students ,
currently in China.

JiaoXianjun, 2021-04-15:
Which university/school? master student? PhD? Can you send me an email?

closed 2021-04-21 by JiaoXianjun

arhiv6, 2022-02-21:
@pengshitao , were you able to solve the problem?  I am build [d692cb8](https://github.com/open-sdr/openwifi/commit/d692cb8dcc40b8ce4844ede037dfde88c60534a8) on zcu102_fmcs2 board. I'm use "Quick start" instructions and run board in "Station mode", it is work. But, board do not connect to other AP - I'm use "Basic operations" instructions and getting infinite "sdr0: CTRL-EVENT-SCAN-STARTED". Command wpa_cli scan/scan_results always return empty.