open-sdr/openwifi #66: use openwifi_tx() to send a custom packet

18845725896peng, 2021-04-21:
hello,
    excuse me ,recently I read the openwifi document ,sdr.c and learn about linux mac80211 subsystem. I learn the packet transmission between mac80211 and sdr driver,but I do not know how to use openwifi_tx() to send a custom packet(I  konw APIS such as ieee80211_subif_start_xmit,drv_tx()... but I have no idea about where to encapsulate data and use mac80211 to send a packet)? 
   Our company wants to use openwifi in TSN.

 thanks

JiaoXianjun, 2021-04-21:
Hello,

Thanks for reaching us. We could take a look. Before that, would you please let us know whom we are helping? Could you send your information (name, organization/university/company, etc) from your organization email to my email xianjun.jiao@ugent.be?

Thanks!

18845725896peng, 2021-04-22:
hello,
We come from Chinese institute(771).The company's email can only be used through the intranet,sorry

thanks

JiaoXianjun, 2021-04-26:
Hello,

"but I do not know how to use openwifi_tx() to send a custom packet" -- We have show you the source code of openwifi_tx(). By reading the source code of openwifi_tx(), you will know everything you need.

You need to come up with specific/concrete question, not like just "I don't know how to use the source code".

18845725896peng, 2021-04-27:
thanks for your reply

closed 2021-04-27 by 18845725896peng