open-sdr/openwifi #69: How to utilize ad9361 in the new version?

frestuc, 2021-05-12:
Hi, I am not sure I understand how we can utilize ad9361 driver in the new version. I am running wgd.sh and getting the following:

```
[  931.095708] sdr,xpu dev_probe match!
[  931.095729] sdr,xpu dev_probe io start 0xa0006000 end 0xa0006fff name /fpga-axi@0/xpu@a0006000 flags 0x00000200 desc 0x00000000
[  931.095731] sdr,xpu dev_probe base_addr 0x09275000
[  931.095732] sdr,xpu dev_probe xpu_driver_api_inst 0x00c12408
[  931.095734] sdr,xpu dev_probe             xpu_api 0x00c12408
[  931.095736] sdr,xpu dev_probe reset tsf timer
[  931.195748] sdr,xpu dev_probe XPU_REG_TSF_RUNTIME_VAL_LOW_read 1 33000 99998us
[  931.195750] sdr,xpu dev_probe succeed!
[  931.195752] sdr,xpu hw_init mode 1
[  931.195763] sdr,xpu hw_init mode XPU_NORMAL
[  931.195767] sdr,xpu hw_init err 0
[  931.706562] sdr: Unknown symbol ad9361_spi_to_phy (err 0)
[  931.706602] sdr: Unknown symbol ad9361_set_tx_atten (err 0)
[  931.706614] sdr: Unknown symbol ad9361_spi_read (err 0)
[  931.706685] sdr: Unknown symbol ad9361_tx_mute (err 0)
[  931.706745] sdr: Unknown symbol ad9361_get_tx_atten (err 0)
[  931.706813] sdr: Unknown symbol ad9361_ctrl_outs_setup (err 0)
```

Any suggestion? Thanks!

JiaoXianjun, 2021-05-12:
We have restructured the procedure to generate new kernel after changing the way using ad9361.
I suggest you follow:
https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch
to rebuild your SD card image.

If you are not starting from a clean fresh git clone of openwifi, at least you need to reset the openwifi/adi-linux (or adi-linux-64) by:
```
git reset --hard
git clean -d -x -f

```
in adi-linux (or adi-linux-64) directory

frestuc, 2021-05-13:
I think I may have found the issue, when I git reset and clean like you said, and re-run prepare_kernel, this is what I get:

```/usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x10): multiple definition of `yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:111: scripts/dtc/dtc] Error 1
make[1]: *** [scripts/Makefile.build:573: scripts/dtc] Error 2
make: *** [Makefile:562: scripts] Error 2
```
If I go ahead and add `extern` to that variable, that screws up the kernel version and therefore the drivers mismatch...
Any hint/fix?

JiaoXianjun, 2021-05-13:
It is because you have gcc-10 (run: gcc --version).
Please install gcc-9, and make sure your default gcc points to gcc-9 instead of 10.

closed 2021-05-13 by frestuc