open-sdr/openwifi #71: 802.11 packet injection

18845725896peng, 2021-05-25:
hello,
    I ssh to the board and setup internet connection,when I input make,
   root@analog:~/openwifi/inject_80211# make
gcc  -Wall -Werror inject_80211.c -o inject_80211 -lpcap
In file included from inject_80211.c:22:0:
inject_80211.h:6:18: fatal error: pcap.h: No such file or directory
 #include <pcap.h>
                  ^
compilation terminated.
make: *** [inject_80211] Error 1

JiaoXianjun, 2021-05-25:
What is your board type?
Are you using our image? Or you create your own image?

18845725896peng, 2021-05-25:
I use ZC706 and ad-fmcomms2-ebz and your image

JiaoXianjun, 2021-05-25:
When do you download the image?

18845725896peng, 2021-05-25:
three months ago ,need I use the newest image?

JiaoXianjun, 2021-05-25:
yes. please

18845725896peng, 2021-05-25:
thank you, I will try

18845725896peng, 2021-05-25:
it works , thank you

closed 2021-05-25 by 18845725896peng