open-sdr/openwifi #72: How can I  make modified sdr.c work?

18845725896peng, 2021-05-25:
hello,
   when I modify the sdr.c ,how can I make it work?  Ssh to the board and compile it to generate sdr.ko, then insmod?

JiaoXianjun, 2021-05-25:
You need to follow: https://github.com/open-sdr/openwifi#update-driver

18845725896peng, 2021-05-25:
thank you ,I get it

closed 2021-05-25 by 18845725896peng

BZZDKBRS, 2022-06-25:
> 你好， 当我修改 sdr.c 的时候，我能怎么板子工作？ssh到，编译生成 sdr.ko，然后insmod？
Hello, where did you modify the code?
What I know is to download the code in the virtual machine. But I just started to migrate openwifi. What configurations do I need for the virtual machine? May I see your orders?

chenmo234, 2024-06-03:
I also need to modify sdr.c and perform the same operations as you. Can I discuss this with you? Here is my contact information: c2385405729.