open-sdr/openwifi #75: Management Entity

tommy782, 2021-06-08:
Hello,
IEEE 802.11 standard shows that the MAC and PHY Layers contains 2 management entities which are:
- MLME (MAC Layer Management Entity)
- PLME (Physical Layer Management Entity).

And 1 layer-independent entity which is the SME (Station Management Entity).

Entities interact via 3 SAPs (Service Access Point) which are:
- SME-MLME SAP
- SME-PLME SAP
- MLME-PLME SAP

SAPs include defined primitives which are exchanged. This definition includes the GET and SET operations between MLME, PLME, and SME as well as other individually defined service primitives.

Are those Entities and SAPs supported? If yes, would you please guide me where to find them in the Repo? 
Appreciate your help. 
 
Thanks,
Tommy

[image]

JiaoXianjun, 2021-06-09:
openwifi is 802.11 standard compliant, so YES, we support those Entities and SAPs.

All those Entities and SAPs are implemented in the Linux (mac80211 subsystem and hostapd/wpa_supplicant/etc), not by openwifi. 
(mac80211 subsystem introduction: https://www.kernel.org/doc/html/v4.9/80211/mac80211.html).

openwifi only implements the driver and the WiFi chip (like all other WiFi chip companies). And the interface between the driver and Linux are defined already by Linux mac80211 subsystem: https://www.kernel.org/doc/html/v4.9/80211/mac80211.html#c.ieee80211_ops

That's why openwifi can focus on the driver and chip, and doesn't need to care a lot about all the protocols (in the standard as you discovered). Because all those things can be implemented in software (Linux) and have already been implemented in Linux.

tommy782, 2021-06-09:
Hello,
Thanks for the clarification and the links. Really appreciated.

It seems that a lot of stuff is being handled behind which makes it more complicated than I thought.
One more thing please:
Is the Management Information Base (MIB) also included in Linux mac80211? Is it kind of a register file that you can read from or write to?

Thanks

JiaoXianjun, 2021-06-09:
Yes I think the MIB is inside the Linux mac80211 or application above (hostapd/wpa_supplicant). For the packet transmitted by openwifi, the openwifi driver only receives the packet from Linux, and the content (include the MIB I guess)  of the packet has already been filled by Linux.

tommy782, 2021-06-09:
Ok. Thank you for your help.

closed 2021-06-09 by tommy782