open-sdr/openwifi #76: mac80211 questions

hychiang73, 2021-06-23:
That's a great project indeed ! Thanks for your effort.

There are some questions around mac80211 in my mind. 

1. Did you ever modify mac80211 source code in order to fit some features with your FPGS? If so, is it possible to share with us?
2. How did you investigate the works between UMAC (upper mac) and LMAC that have to be done in mac80211 or in FPGA? Because it's kinda harder to make sure which of works related 802.11 has been implemented in mac80211 in my opinion.

Thanks in advance.

JiaoXianjun, 2021-06-23:
I am afraid that we are not the correct people to ask mac80211 question.
We focus on FPGA and driver, and make sure FPGA and driver are aligned with the Linux mac80211 API/requirements:
https://www.kernel.org/doc/html/v4.10/driver-api/80211/mac80211.html
https://01.org/linuxgraphics/gfx-docs/drm/driver-api/80211/mac80211-advanced.html

hychiang73, 2021-06-23:
I got it, thanks for your answer :)

closed 2021-06-23 by hychiang73