open-sdr/openwifi #77: Ping unsuccessful between two ad-hoc nodes

aaabbb414, 2021-06-25:
I am using Zedboard with FMCOMMS4 setup. I downloaded image in table with board_name: zed_fmcs2. 
I followed steps in https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/ad-hoc-two-sdr.md to configure two nodes. 
I could see both of them having same cell which I assume they are connected. However, "ping" keeps failing because "destination host unreachable". This happens for both nodes. 

I am wondering if there is any suggestion how to solve this? 

Thanks

JiaoXianjun, 2021-06-25:
Hello,

Thanks for report. We just updated an issue template. Would you please answer:

0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

JiaoXianjun, 2021-07-12:
Have you ping successfully?

closed 2021-07-13 by JiaoXianjun

srushti1763, 2025-09-09:
Same issue