open-sdr/openwifi #79: ADRV9364 configuration mode

black-pigeon, 2021-07-01:
Hello!
I recently recreated the openwifi hardware project. The  template project which I choose was adrv9364z7020.
I am a little confused about this project. The RF chip AD9364 should work in 1r1t mode. While in the openwifi vivado project, the AXI_AD9361 IP is configured as 2r2t mode.

https://github.com/open-sdr/openwifi-hw/blob/9e6e52df5f5589810bccaeb984d32571409cd180/boards/adrv9364z7020/src/system.bd#L2232

In addition，the devicetree is also enabled 2r2t mode.
https://github.com/open-sdr/openwifi/blob/3bd1e9ee68b02fd321d9f9a088c06cff40ff0189/kernel_boot/boards/adrv9364z7020/devicetree.dts#L215

Will ad9364 work properly？

JiaoXianjun, 2021-07-01:
yes. it does.

black-pigeon, 2021-07-01:
Thank for your reply!
But I see that the 2r2t mode is not enabled in the device tree of the official adrv9364.
If I change the properties in the project and in the device tree to 1r1t, will the openwifi project still work?

JiaoXianjun, 2021-07-01:
If you change our devicetree, we can't grantee openwifi will still work. But you can try.

black-pigeon, 2021-07-01:
OK, I will try and close this issue.

closed 2021-07-01 by black-pigeon