open-sdr/openwifi #80: How to set the baseband sample rate?

black-pigeon, 2021-07-02:
Hello.
I found that the openwifi project baseband sample rate is 40MSPS in the RX/TX clock chain and the file rf_init.sh shows that the sample rate is configured as 40MSPS.

https://github.com/open-sdr/openwifi/blob/3bd1e9ee68b02fd321d9f9a088c06cff40ff0189/user_space/openwifi_ad9361_fir.ftr#L4-L5

https://github.com/open-sdr/openwifi/blob/3bd1e9ee68b02fd321d9f9a088c06cff40ff0189/user_space/rf_init.sh#L42-L44

I am wondering that if I change the sampel rate to 20MSPS,  will this project still work?

JiaoXianjun, 2021-07-02:
I think it won't work. Sampling rate is the basic parameter, which affects many things.

black-pigeon, 2021-07-02:
Is that  to say the openwifi is working at the fixed samp_rate of 40MSPS? 
oops... That's sounds bad. 
I have a customized board but the hardware does not perform well in LVDS mode, it can't reach the sample rate above 40MSPS.
By the way, can openwifi project work when AXI_AD9361 is configured in CMOS 1R1T mode, my customized board performs well in this mode. I think this is the mode with the least hardware requirements.
Have you ever tried to implement openwifi on adrv9364z7020 ccbob_cmos, if this board can work, I think my customized board will work also.
Thanks for your quick reply!!!

JiaoXianjun, 2021-07-02:
openwifi can work under 20Msps, but you need to change the fpga code to support this. We use 40Msps to have more freedom to avoid self interference: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#analog-and-digital-frequency-design

we use adrv9364z7020 lvds for now and never try cmos on adrv9364z7020. But I think in principle it can be ported to 20Msps cmos mode.

black-pigeon, 2021-07-02:
Sure, I will try.
This is a really amzing project.

JiaoXianjun, 2021-07-12:
What about the result of your experiment?

black-pigeon, 2021-07-12:
Sorry for the late reply. For now, I am waiting for the new hardware design release.

closed 2021-07-21 by JiaoXianjun