open-sdr/openwifi #81: tx_intf module

Shangjie-Yang, 2021-07-07:
Hello,
       I notice that in tx_intf module has two data entry ports, s00_axis_tdata[63:0] and dma_data[63:0], If my understanding is correct, the data from s00_axis_tdata[63:0] send into queue by the queuing rules(linux prio) and then run into acc(openofdm), do ofdm modulate, finally the data from acc[31:0] and dma_data[63:0] do once mux,decide who run into dac_intf. so what is the difference between the data stored in s00_axis_tdata[63:0] and dma_data[63:0]？

Thanks

JiaoXianjun, 2021-07-07:
dma_data is from the original existing axi_ad9361_dac_dma in the original ADI reference design. It is useless for openwifi. We just keep the original connections for doing nothing.

Shangjie-Yang, 2021-07-07:
Thank you for your reply, whether the mux control signal from slv_reg7, also write nothing?

JiaoXianjun, 2021-07-07:
we keep the option to connect ad9361 to original axi_ad9361_dac_dma.
you can check what we set in the driver of tx_intf. tx_intf.c: TX_INTF_REG_MISC_SEL_write, and figure out.

Shangjie-Yang, 2021-07-08:
sure, thanks for your answer.

JiaoXianjun, 2021-07-12:
Do you still have other questions?

closed 2021-07-13 by JiaoXianjun