open-sdr/openwifi #82: How to transplant openwifi to zcu3cg device

feiger313, 2021-07-10:
0. I  am a digital ASIC engineer
1. Our image is used directly or you build your own image?
   own image.
2. What is your own modification?
   hardware:  modify some constraint ; device : ZCU9EG --> ZCU3CG
3. Versions: OS, Vivado, openwifi/openwifi-hw commit revision
  OS: ubuntu 18.04 ;vivado : 2018.3 ;openwifi : master 
4. Board/hardware type
  ALINX FPGA board from Chinese .
5. WiFi channel number

6. Steps to reproduce the issue

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

question:  
1.build u-boot：When compiling u-boot, does the config file use zcu102's config file directly? Or do we need to make changes? 
2. build device-tree : I generated the device tree in the SDK. There is a place to fill in bootargs. Do I need to fill in bootargs in this place separately。How do you set up bootargs？

JiaoXianjun, 2021-07-12:
Hello,

Since we don't have experience on porting to new hardware that we don't have access to the hardware and its reference design, we can only answer according to our case when we port it between Xilinx/ADI boards.

1. For the u-boot, it is not built by us. The bl31.elf and u-boot-zcu.elf are from the ADI SD card image of zcu102.
2. For the devicetree, according to: https://github.ugent.be/xjiao/openwifi#Porting-guide
```
We use dtc command to get devicetree.dts converted from devicetree.dtb in Analog Devices Linux image, then do modification according to what we have added/modified to the reference design.
```
In this case, we never touch the bootargs in the devicetree, because the devicetree is based on the orginal devicetree in the SD card image, and the bootargs is supposed to be correctly configured in the devicetree.

weiliu1011, 2021-07-12:
I think the bootarg is set in uEnv.txt file in the /BOOT partition of the openwifi Image. For instance for zcu102 it is: 
bootargs=console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon rootfstype=ext4 rootwait cpuidle.off=1

You can stop the auto boot process from uboot, define your own uboot command, environment variables including the bootargs.

feiger313, 2021-07-17:
> I think the bootarg is set in uEnv.txt file in the /BOOT partition of the openwifi Image. For instance for zcu102 it is:
> bootargs=console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlycon [...]

Thank you very much. I see this uenv.txt. I'll have a try

feiger313, 2021-07-17:
> Hello,
>
> Since we don't have experience on porting to new hardware that we don't have access to the hardware and its reference design, we can only answer according to our case when we port it between [...]

Thank you very much. I'll have a try

closed 2021-07-21 by JiaoXianjun