open-sdr/openwifi #83: intel 5300 can't get csi from openwifi AP.

huangjwDGUT, 2021-07-12:
1. The computer with 5300 NIC connect to a openwifi ap,  the computer can access internet.
2. It can not get csi from openwifi ap, but it can get csi from commercial wireless router.
3. When I debug, I found the program stop at recv(), this function do not return. 
4. Here is the FAQ from csi tool.
5.  For which packets is CSI measured? [−]
CSI is measured from a received packet under the following conditions:
• The packet is transmitted using an 802.11n "HT" rate.
• The receiver's logical configuration (e.g., channel type is HT, and channel width) and its physical configuration (e.g., correct number of antennas connected, all antennas work in the configured band, all antennas are enabled in the firmware) support the transmitted rate.
• The receiver decodes the packet correctly, with no errors.
AND depending on the mode:
• Client mode: The packet is transmitted by the connected AP, and the destination is either the local address or broadcast.
• AP mode: The packet is transmitted by a connected client.
• IBSS (ad-hoc) mode: The packet is transmitted by another station in the connected IBSS, and the destination is either the local address or broadcast.
• Monitor mode: The packet is addressed both to and from a hardcoded, fixed address 00:16:ea:12:34:56. See the injection scripts, using mode 1 only.
In other words, CSI is not recorded for overheard 802.11n conversations. It is also not recorded for beacons from a connected AP if they are sent a non-HT rate (e.g., the 802.11a/g 6 Mbps rate). I also note that there is only one CSI measurement per 802.11n A-MPDU (aggregation batch) since there is really only "one" preamble.
6. I do open the ht in file hostapd-openwifi-11an.conf
ieee80211n=1
ht_capab=[SHORT-GI-20]
require_ht=1

Has anyone ever encountered this problem? Any comments are welcome.

JiaoXianjun, 2021-07-12:
Hello,

When you run the internet traffic (online video, downloading, etc) over 5300 -- openwifi AP, could you please show me some openwifi dmesg printing according to the debug method: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#debug-methods

huangjwDGUT, 2021-07-14:
When the laptop with 5300 link to openwifi and download a file.
The dmesg from openwifi:

Welcome to Linaro 14.04 (GNU/Linux 4.14.0-gb6e379910a11-dirty armv7l)

 * Documentation:  https://wiki.analog.com/ https://ez.analog.com/
Last login: Thu Jan  1 00:06:16 1970 from 192.168.10.1
root@analog:~# dmesg
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/20 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/24 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
[... 324 lines cut ...]
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/0 PS frames since STA woke up
sdr0: STA 00:21:6a:65:21:90 aid 1 enters power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1: PS buffer for AC 2
sdr0: STA 00:21:6a:65:21:90 aid 1 exits power save mode
sdr0: STA 00:21:6a:65:21:90 aid 1 sending 0 filtered/1 PS frames since STA woke up
root@analog:~#

JiaoXianjun, 2021-07-14:
Hey, you need to read our debug method carefully. Please run:

```
./sdrctl dev sdr0 set reg drv_tx 7 7
./sdrctl dev sdr0 set reg drv_rx 7 7
```

Before you run dmesg while downloading.

huangjwDGUT, 2021-07-14:
The laptop with 5300 connect the openwifi and downloading the linux kernel, and here is the dmesg output.

root@analog:~# ./sdrctl dev sdr0 set reg drv_tx 7 7
-bash: ./sdrctl: No such file or directory
root@analog:~# cd openwifi/
root@analog:~/openwifi# ./sdrctl dev sdr0 set reg drv_tx 7 7
reg  cat: 8
reg addr: 0008001c
reg  val: 00000007
root@analog:~/openwifi# ./sdrctl dev sdr0 set reg drv_rx 7 7
reg  cat: 7
reg addr: 0007001c
reg  val: 00000007
root@analog:~/openwifi# dmesg
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr45 rd44 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2710 flag0001001e retr1 ack0 prio0 q0 wr45 rd44
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr46 rd45 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2720 flag0001001e retr1 ack0 prio0 q0 wr46 rd45
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr47 rd46 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC29d0 flag40000010 retr5 ack1 prio2 q2 wr39 rd38
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr40 rd39 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC29e0 flag40000010 retr5 ack1 prio2 q2 wr40 rd39
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC29f0 flag40000010 retr5 ack1 prio2 q2 wr41 rd39
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr42 rd40 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr42 rd41 num_rand_slot 0 cw 0 
sdr,sdr openwifi_rx_interrupt: WARNING loop_count 0
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0188 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd2f0 fcs1 buf_idx2 -50dBm
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0988 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd300 fcs1 buf_idx3 -49dBm
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a00 flag40000010 retr5 ack1 prio2 q2 wr42 rd41
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a10 flag40000010 retr5 ack1 prio2 q2 wr43 rd41
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr44 rd42 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr44 rd43 num_rand_slot 0 cw 0 
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0188 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd310 fcs1 buf_idx4 -50dBm
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a20 flag40000010 retr5 ack1 prio2 q2 wr44 rd43
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr45 rd44 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a30 flag40000010 retr5 ack1 prio2 q2 wr45 rd44
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a40 flag40000010 retr5 ack1 prio2 q2 wr46 rd44
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a50 flag40000010 retr5 ack1 prio2 q2 wr47 rd44
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr48 rd45 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr48 rd46 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr48 rd47 num_rand_slot 0 cw 0 
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0188 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd320 fcs1 buf_idx5 -50dBm
sdr,sdr openwifi_tx: 1474bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a60 flag40000010 retr5 ack1 prio2 q2 wr48 rd47
sdr,sdr openwifi_tx:   86bytes ht1 390M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2a70 flag40000010 retr5 ack1 prio2 q2 wr49 rd47
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr50 rd48 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio2 wr50 rd49 num_rand_slot 0 cw 0 
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 585M FC0988 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd330 fcs1 buf_idx6 -48dBm
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0188 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd340 fcs1 buf_idx7 -49dBm
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0188 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd350 fcs1 buf_idx8 -49dBm
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2730 flag0001001e retr1 ack0 prio0 q0 wr47 rd46
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr48 rd47 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2740 flag0001001e retr1 ack0 prio0 q0 wr48 rd47
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr49 rd48 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2750 flag0001001e retr1 ack0 prio0 q0 wr49 rd48
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr50 rd49 num_rand_slot 0 cw 0 
[... cut ...]
sdr,sdr openwifi_rx_interrupt:  90bytes ht1 650M FC0188 DI002c addr1/2/3:6655443322d8/00216a652190/6655443322d8 SCd5f0 fcs1 buf_idx4 -52dBm
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2850 flag0001001e retr1 ack0 prio0 q0 wr1 rd0
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr2 rd1 num_rand_slot 0 cw 0 
sdr,sdr openwifi_tx:  156bytes ht0  60M FC0080 DI0000 addr1/2/3:ffffffffffff/6655443322d8/6655443322d8 SC2860 flag0001001e retr1 ack0 prio0 q0 wr2 rd1
sdr,sdr openwifi_tx_interrupt: tx_result 00 prio0 wr3 rd2 num_rand_slot 0 cw 0 
root@analog:~/openwifi#

JiaoXianjun, 2021-07-14:
I don't see any issue with the dmesg. There are lots of
```
sdr,sdr openwifi_tx: 1474bytes ht1 520M FC0288 DI0000 addr1/2/3:00216a652190/6655443322d8/6655443322d8 SC2de0 flag40000010 retr5 ack1 prio2 q2 wr40 rd38
```
Which means openwifi is transmitting in "ht" rate to your 5300.

I suggest you try different modes to see whether some of them work for you:

1. ad-hoc mode. openwifi as one node, 5300 as another. https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/ad-hoc-two-sdr.md

2. 5300 as AP, and openwifi as client. https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/ap-client-two-sdr.md

3. packet injection from openwifi: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md , in this case, I think your 5300 should be set in monitor mode

JiaoXianjun, 2021-09-05:
Any progress?

huangjwDGUT, 2021-09-15:
No progress.

closed 2021-11-30 by JiaoXianjun