open-sdr/openwifi #84: RF chain and antenna offset when collect CSI

yujianyuanhaha, 2021-07-12:
Greetings,

Thanks for the solid work. While for CSI capture, does the platform has offset issues, like in  "[mD-Track: Leveraging Multi-Dimensionality in Passive Indoor Wi-Fi Tracking](https://wands.sg/publications/full_list/papers/MobiCom_19_1.pdf)", it include 5 type of offset calibration, especially different **RF chain** offset, the different **antenna offset** needed to be carefully measure and compensate.

Best

Jet

JiaoXianjun, 2021-07-12:
Hi,

We use single advanced RF chip, ad9361, to handle all intermediate steps in between antenna and baseband FPGA.

Although ad9361 has two RF chains (support 2*2 MIMO), now in baseband FPGA we only select one chain to work with. So current openwifi CSI only contains single antenna/chain result. Of course you can configure via FPGA register to select which chian/antenna will be used.

Regarding the offset (though I am not very clear about the definition, can you elaborate?) of antenna and RF chain, I don't think we will have big problem there. If you check the datesheet of ad9361, you will see that the chip is very advanced. It does many corrections on the dual channel analog signal and digital I/Q to achieve a very high EVM and 2*2 MIMO operation.

yujianyuanhaha, 2021-07-13:
@JiaoXianjun thanks, I agree it would not be a issue since it work in single chain.

Best

Jet

JiaoXianjun, 2021-07-13:
Currently we have trigger (WiFi baseband event) based dual antenna I/Q capture: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq_2ant.md

In the future dual antenna CSI will be supported.

huangjwDGUT, 2021-07-14:
> Greetings,
> 
> Thanks for the solid work @JiaoXianjun . While for CSI capture, does the platform has offset issues, like in "[mD-Track: Leveraging Multi-Dimensionality in Passive Indoor Wi-Fi Tracking](https://wands.sg/publications/full_list/papers/MobiCom_19_1.pdf)", it include 5 type of offset calibration, especially different **RF chain** offset, the different **antenna offset** needed to be carefully measured and compensated. Does open-wifi also met this issue?
> 
> Best
> 
> Jet

Hi Jet

Can you use computer with 5300 NIC to collect CSI from openwifi?

yujianyuanhaha, 2021-07-14:
@huangjwDGUT I don't know since I don't have open-wifi testbed at hand. But 5300 should be able to collect CSI from any device serving as AP, even a commercial router.

closed 2021-07-21 by JiaoXianjun