open-sdr/openwifi #85: How to modify RX/TX bandwidth under current modulation method

dz6242, 2021-07-13:
Dear Sir,

We use Analog device ADRV9361-Z7035 development board to evaluate and try to start the open-sdr related project. Everything work fine with the initial image downloaded from github, however, we meet some problem while trying to narrow down the TX/RX bandwidth (We need to modify the transmission Bandwidth in ad-hoc mode to meet requirement). My question is: Is it possible to modify RX/TX bandwidth under current modulation method, i.e. modify RX and TX from 17.5M (as shown in the following figure) to 3M under current design? just modify rf_init.sh or modify which file? and the throughput after narrowing down BW by iperf?

[image]

JiaoXianjun, 2021-07-13:
It is possible.
But it needs to change lots of files: drivers, .sh scripts and FPGA .v files, etc.
Sorry that I can't describe how to do it in several minutes. I might needs months's effort I guess.

I suggest you study all the code we published firstly to understand the design.
Then if you have concrete/specific question, you can ask.

Question like "can openwifi do this xxxx and how to do this xxxx" can't be answered by us. Because yes openwifi is based on SDR and in principle it can do many things. If "this xxxx" is not offered currently by us, you need to figure out how to do it by yourself.

dz6242, 2021-07-13:
Dear Sir,

Thank you very much for your quick response and very good suggestion. We will try to study further in the next step.

Best Regards,
TaoCheng

closed 2021-07-21 by JiaoXianjun