open-sdr/openwifi #87: Xianjun, could you help me solve this problem?

cnh19930108, 2021-07-14:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

JiaoXianjun, 2021-07-14:
We haven't tested 2.4GHz band a lot. Currently we recommend you try 5Ghz band firstly.
We will take a look at 2.4GHz, but not sure we can solve it in short term.

cnh19930108, 2021-07-14:
Hi Xianjun,

Thanks for your kind reply.

cnh19930108, 2021-07-14:
Hello,

In openwifi project, what is difference between 2.4G and 5G?

JiaoXianjun, 2021-07-14:
Hard to explain the differences in several sentences. We implement them according to the 802.11 standard. In the standard 2.4GHz (11g) and 5GHz (11a) are two difference chapters.

cnh19930108, 2021-07-14:
Sorry, I have some questions.
1. How can i test only PHY without MAC?
   I mean, send custom 80211 frames in one board and receive its in another one.

2. How can I check for ad9361 part?
   I mean, send frame via tx ant and  receive again from rx ant. 

How do I modify or set up openwifi project for these problem?
we use openwifi-1.2.0 now.

JiaoXianjun, 2021-07-14:
Hello,

If you check all our documents carefully, you will find this app note: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md

It allows you to do packet injection and monitor mode, where you have full control over openwifi.

cnh19930108, 2021-07-15:
thanks for you reply.

JiaoXianjun, 2021-07-15:
For that you need to check ad9361 datasheet. We never try ad9361 loop back.

closed 2021-07-15 by JiaoXianjun

cnh19930108, 2021-07-15:
For test inject_80211, i run follow commands.

Firstly, according to this instruction(https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md),
**cd openwifi
./wgd.sh**
/////////////////////////////////////////////////////////////////////////////////////////////////////
root@analog:~/openwifi# dmesg -c
xilinx-vdma 80400000.dma: Xilinx AXI DMA Engine Driver Probed!!
xilinx-vdma 80410000.dma: Xilinx AXI DMA Engine Driver Probed!!
tx_intf: loading out-of-tree module taints kernel.

sdr,tx_intf dev_probe match!
sdr,tx_intf dev_probe io start 0x83c0ffff83c00000 end 0x200ef7f4ec4 name (null) flags 0xbf18701c desc 0xbf18701c
sdr,tx_intf dev_probe base_addr 0xf3150000
sdr,tx_intf dev_probe tx_intf_driver_api_inst 0xbf187284
sdr,tx_intf dev_probe             tx_intf_api 0xbf187284
sdr,tx_intf dev_probe succeed!
sdr,tx_intf hw_init mode 6
sdr,tx_intf hw_init mode TX_INTF_BW_20MHZ_AT_N_10MHZ_ANT1
sdr,tx_intf hw_init err 0
ad9361 spi0.0: ad9361_probe : enter (ad9361)
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
SAMPL CLK: 25000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o o o o o o o o # #
1:o o o o o # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o # # # # # # # # # #
1:o o o o # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o # # # # # # # # # #
1:o o o o # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o O o o # # # # # # # # # #
1:o o o o # # # # # # # # # # # #
SAMPL CLK: 25000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o o o o o o
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o O o o o o # # # # #
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0xf3108000, probed ADC AD9361 as MASTER
cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER (9.01.b) at 0x79024000 mapped to 0xf314d000, probed DDS AD9361
random: crng init done
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o O o o # # # # # # # # # # #
1:o o o o # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o O o o o # # # # # #

sdr,rx_intf dev_probe match!
sdr,rx_intf dev_probe io start 0x83c20000 end 0x83c2ffff name /fpga-axi@0/rx_intf@83c20000 flags 0x00000200 desc 0x00000000
sdr,rx_intf dev_probe base_addr 0xf3240000
sdr,rx_intf dev_probe rx_intf_driver_api_inst 0xbf1b7284
[... 129 lines cut ...]
root@analog:~/openwifi# dmesg
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_config changed flag 00000100
sdr,sdr openwifi_config changed flag 00000008
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
root@analog:~/openwifi#
////////////////////////////////////////////////////////////////////////////////////////////

I can't send log such as 'sdr, sdr openwifi_tx ...'.
what is the cause?

i also tried follow method.
**cd openwifi
./wgd.sh
./monitor_ch.sh sdr0 11
./inject_80211/inject_80211 -m n -r 0 -n 10 -s 64 sdr0**

but result is same.
How can i see packet send log in sdr.c?

On the receiver side, we can see and receive nothing.

Looking for your kind reply.

JiaoXianjun, 2021-07-15:
Hello, to help you better, you need to answer according to our template:

0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

cnh19930108, 2021-07-15:
1. i used your image directly.
  In openwifi github page, according to follow instruction.
  [Build openwifi Linux img from scratch]
  i used 2019_R1-2020_02_04.img.xz from Analog Devices Wiki.

 2. My modification
  - I modified dts file in your project because our board is custom one, MicroZus ZYNQ 7020.
	    ad9361-phy@0 {
	    	...
	    	adi,lvds-bias-mV = <0x1c2>;
	    	...
	    	adi,dcxo-coarse-and-fine-tune = <0x8 0x1720>;
			/*adi,xo-disable-use-ext-refclk-enable;*/
	    }
	    clocks {

			clock@0 {
				...
				clock-frequency = <0x2dc6c00>;
				...
			}
		}
  - In openwifi-hw/board/adrv9364z7020, i modified zynqad9361 pin map in XDC file.

 3. Develop system : ubuntu 16.04.6 amd64
   OS: adi-kernel 2019_R1
   Vivado: 2018.3
   OpenWiFi: 1.2.0
   openwifi-hw: 1106691
   adi-hdl: f61d9707

 4. Board/hardware type: MicroZus+SRF_AD9361_V2.0
   MicroZus: MicroPhase zynq7020-1clg400
   SRF_AD9361_V2.0: ad9361 2R2T

 5. WiFi channel number:
    44, 5 in ap/station mode.
    11 in monitor mide.
  
 6. Steps to reproduce the issue
  - on sender side
   cd openwifi
   ./wgd.sh
   ------------------------------------------------------------------------------
    xilinx-vdma 80400000.dma: Xilinx AXI DMA Engine Driver Probed!!
	xilinx-vdma 80410000.dma: Xilinx AXI DMA Engine Driver Probed!!
	tx_intf: loading out-of-tree module taints kernel.

	sdr,tx_intf dev_probe match!
	sdr,tx_intf dev_probe io start 0x83c0ffff83c00000 end 0x200ef7f4ec4 name (null) flags 0xbf18701c desc 0xbf18701c
	sdr,tx_intf dev_probe base_addr 0xf3150000
	sdr,tx_intf dev_probe tx_intf_driver_api_inst 0xbf187284
	sdr,tx_intf dev_probe tx_intf_api 0xbf187284
	sdr,tx_intf dev_probe succeed!
	sdr,tx_intf hw_init mode 6
	sdr,tx_intf hw_init mode TX_INTF_BW_20MHZ_AT_N_10MHZ_ANT1
	sdr,tx_intf hw_init err 0
	ad9361 spi0.0: ad9361_probe : enter (ad9361)
	ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
	SAMPL CLK: 25000000 tuning: RX
	0:1:2:3:4:5:6:7:8:9🅰️b:c:d:e:f:
	0:o o o o o o o o o o o o o o # #
	1:o o o o o # # # # # # # # # # #
	SAMPL CLK: 40000000 tuning: RX
	0:1:2:3:4:5:6:7:8:9🅰️b:c:d:e:f:
	0:o o o o o o # # # # # # # # # #
	1:o o o o # # # # # # # # # # # #
	SAMPL CLK: 40000000 tuning: RX
	0:1:2:3:4:5:6:7:8:9🅰️b:c:d:e:f:
	0:o o o o o o # # # # # # # # # #
[... 185 lines cut ...]
   ./sdrctl dev sdr0 set reg drv_tx 7 7
   
   And i modified Source Address/Destination Address in inject_80211.c.
   then run this command.
   
   ./inject_80211/inject_80211 -m n -r 0 -n 10 -s 64 mon0
   ------------------------------------------------------------------------------
    sdr,sdr openwifi_prepare_multicast
	sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
	sdr,sdr openwifi_config changed flag 00000100
	sdr,sdr openwifi_config changed flag 00000008
	sdr,sdr openwifi_prepare_multicast
	sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
   ------------------------------------------------------------------------------

    I can't send log such as 'sdr, sdr openwifi_tx ...'.
	what is the cause?

  - on receiver side
   ./wgd.sh
   iw dev sdr0 interface add mon0 type monitor && ifconfig mon0 up
   ./sdrctl dev sdr0 set reg drv_rx 7 7
   tcpdump -i mon0 -w trace.pcap 'wlan addr1 66:55:44:33:22:57 and wlan addr2 66:55:44:33:22:77'

   On the receiver side, we can see and receive nothing.

JiaoXianjun, 2021-07-15:
Generally we don't know the situation of your modified SW and private hardware that we don't have.
From the log of your wgd.sh, you have:
```
SAMPL CLK: 25000000 tuning: RX
0:1:2:3:4:5:6:7:8:9🅰️b:c:d:e:f:
0:o o o o o o o o o o o o o o # #
1:o o o o o # # # # # # # # # # #
```

This means your LVDS connection between your FPGA and AD9361 has some issue. But since we only use the off the shelf Xilinx and Analgo Devices' boards, and never have issue like this, I am afraid that our help will be very limited.

cnh19930108, 2021-07-16:
Oh, you mean, this issue is board problem?

And, how about tuning log in the off the shelf xilinx you use?

JiaoXianjun, 2021-07-16:
Sorry we never see this log. Could be because everything is fine, so no such a error log. 

I think the best place to ask question about ad9361 lvds interface tuning is: https://ez.analog.com/sw-interface-tools/

cnh19930108, 2021-07-16:
Oh, thanks for your kind reply.
I'll do study it more.

But when i set sdr0 to monitor and run inject_80211,
why can't i see openwifi_tx log?

For seeing tx packet log, i set drv_tx reg to 7.
that is,
./sdrctl dev sdr0 set reg drv_tx 7 7

Why?

When i set sdr0 to ap with 'fosdem.sh' script, i saw such logs.
But i couldn't see it in monitor mode.

JiaoXianjun, 2021-07-16:
1. please don't do "And i modified Source Address/Destination Address in inject_80211.c."
2. run "iw dev sdr0 interface add mon0 type monitor && ifconfig mon0 up" 
3. run "sdrctl dev sdr0 set reg drv_tx 7 7"
4. run "./inject_80211/inject_80211 -m n -r 0 -n 9999 -s 64 mon0"  (change the -n 10 to -n 9999)

cnh19930108, 2021-07-16:
Yes. i will try it, now.

cnh19930108, 2021-07-16:
Hello, i tried it.
But result is same.

iw dev sdr0 interface add mon0 type monitor && ifconfig mon0 up
-------------------------------------------------------------------------
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_config changed flag 00000100
sdr,sdr openwifi_config changed flag 00000008
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
-------------------------------------------------------------------------

And then run this comman.
./sdrctl dev sdr0 set reg drv_tx 7 7
-------------------------------------------------------------------------
sdr,sdr recv set cmd reg cat 8 addr 0000001c val 00000007 idx 7
-------------------------------------------------------------------------

Then,
-------------------------------------------------------------------------
#./inject_80211/inject_80211 -m n -r 0 -n 9999 -s 64 mon0
DLT_IEEE802_11_RADIO Encap
mode = 802.11n, rate index = 0, SHORT GI = 0, number of packets = 9999 and packet size = 116 bytes, delay = 100000 usec
^Cmber of packets sent = 234
#dmesg
device mon0 entered promiscuous mode
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_tx: WARNING ieee80211_stop_queue prio 2 queue 2 no room flag 0 hw queue len 003d0000 request 13 wr 61 rd 0
device mon0 left promiscuous mode
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
-------------------------------------------------------------------------

In here, after send 234 packets, packet_injection is stopped automatically.
I also couldn't see tx log(in sdr.c).

cnh19930108, 2021-07-18:
Hello.
I am testing with inject_80211.

but whenever i sent 61th packets, ieee80211_stop_queue warning occurs.
What is the cause?
[image]

JiaoXianjun, 2021-07-18:
Have you seen all dmesg printing of packets before packet 61? Can you paste the dmesg log here?

Can you unplug tx and rx antenna and try again?

cnh19930108, 2021-07-19:
Yes.

but all dmesg printing of packets? That's all.

In openwifi 1.2.0,
///////////////////////////////////////// sdr.c ///////////////////////////////////////
if ( (!addr_flag) && (priv->drv_tx_reg_val[DRV_TX_REG_IDX_PRINT_CFG]&2) )
         printk("%s openwifi_tx: %4dbytes ht%d %3dM FC%04x DI%04x addr1/2/3:%04x%08x/%04x%08x/%04x%08x ...
//////////////////////////////////////////////////////////////////////////////////////
So no matter drv_tx set to 7, the log(openwifi_tx...) is not displayed.

I unplug tx and rx antenna, then tried again.
/////////////////////////////// dmesg log //////////////////////////////////////////
xilinx-vdma 80400000.dma: Xilinx AXI DMA Engine Driver Probed!!
xilinx-vdma 80410000.dma: Xilinx AXI DMA Engine Driver Probed!!
tx_intf: loading out-of-tree module taints kernel.

sdr,tx_intf dev_probe match!
sdr,tx_intf dev_probe io start 0x83c0ffff83c00000 end 0x200ef7f4ec4 name (null) flags 0xbf18d01c desc 0xc0c84830
sdr,tx_intf dev_probe base_addr 0xf3180000
sdr,tx_intf dev_probe tx_intf_driver_api_inst 0xbf18d284
sdr,tx_intf dev_probe             tx_intf_api 0xbf18d284
sdr,tx_intf dev_probe succeed!
sdr,tx_intf hw_init mode 6
sdr,tx_intf hw_init mode TX_INTF_BW_20MHZ_AT_N_10MHZ_ANT1
sdr,tx_intf hw_init err 0
ad9361 spi0.0: ad9361_probe : enter (ad9361)
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
SAMPL CLK: 25000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o o o o o o o o # #
1:o o o o o # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o # # # # # # # # # # #
1:o o o o o # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o # # # # # # # # # # #
1:o o o o o # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o O o o # # # # # # # # # # #
1:o o o o o # # # # # # # # # # #
SAMPL CLK: 25000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# o o o o o o o o o o o o o o o
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# o o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o O o o o o # # # # #
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0xf3138000, probed ADC AD9361 as MASTER
cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER (9.01.b) at 0x79024000 mapped to 0xf3135000, probed DDS AD9361
random: crng init done
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o O o o # # # # # # # # # # #
1:o o o o o # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
[... 104 lines cut ...]
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 0 M_BCN 1 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 0 CTL 0 BCN_PRP 0 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_prepare_multicast
IPv6: ADDRCONF(NETDEV_UP): sdr0: link is not ready
sdr,sdr openwifi_configure_filter MON 0 M_BCN 1 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 0 CTL 0 BCN_PRP 0 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_config changed flag 00000100
sdr,sdr openwifi_config changed flag 00000008
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr recv set cmd reg cat 8 addr 0000001c val 00000007 idx 7
device mon0 entered promiscuous mode
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
sdr,sdr openwifi_tx: WARNING ieee80211_stop_queue prio 2 queue 2 no room flag 0 hw queue len 003d0000 request 13 wr 61 rd 0
device mon0 left promiscuous mode
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 1 M_BCN 0 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 1 BCN_PRP 1 PCP_FL 0 FCS_FL 0 ALL_MUT 1
/////////////////////////////////////////////////////////////////////////////////////

In here, "sdr,sdr openwifi_tx: WARNING ieee80211_stop_queue" is always displayed whenever 61th packet is transmitted.
It does not change as you change the size of packet payload.

What is the cause?

cnh19930108, 2021-07-19:
I modified some FPGA code(cca.v in xpu) and tried again it .
My test was OK.

cnh19930108, 2021-07-26:
Hi Xianjun, I have some questions. It's strange for TX tuning.
I posted the log below.
I don't know why last tx tuning failed. Thank you in advances:

ad9361 spi0.0: ad9361_probe : enter (ad9361)
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
SAMPL CLK: 25000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o o o o o o o o o o
1:o o o o # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o O o o o # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 25000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o o o o o o
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o O o o o o # # # # #
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0xf                                  3128000, probed ADC AD9361 as MASTER
cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER                                   (9.01.b) at 0x79024000 mapped to 0xf3123000, probed DDS AD9361
random: crng init done
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o O o o # # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # # # # # # # # # # # # # # #  Tuning TX Failed!

JiaoXianjun, 2021-07-26:
The tuning issue is related to your modified sw and private hardware. We don’t have this issue on xilinx/adi original hardware. So, hard to help you.

cnh19930108, 2021-07-26:
i see, thanks for your kind reply.

cnh19930108, 2021-09-13:
Hello,
I want to know real openwifi test result.
for example,
"Performance (AP: openwifi at channel 44, client: TL-WDN4200 N900 USB Dongle):
AP --> client: 30.6Mbps(TCP), 38.8Mbps(UDP)
client --> AP: 17.0Mbps(TCP), 21.5Mbps(UDP)"
is this result the highest level or  on average?
When AP and Client are openwifi, what about average and highest performence?
Then I want to purchase ANTSDR E310.
How about the performance in the ANTSDR E310?