open-sdr/openwifi #89: i used z7020 custom board.I set an Zynq board as an AP by hostapd, the hostapd-openwifi.conf is as follows:

ruixingjin, 2021-07-26:


ruixingjin, 2021-07-26:
ad9361 spi0.0: ad9361_probe : enter (ad9361)
ad9361 spi0.0: ad9361_probe : AD936x Rev 2 successfully initialized
SAMPL CLK: 25000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o o o o o o o o o o
1:o o o o # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o o o o o # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o O o o o # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 25000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o o o o o o
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o o o o o o # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # o o o o O o o o o # # # # #
cf_axi_adc 79020000.cf-ad9361-lpc: ADI AIM (10.01.b) at 0x79020000 mapped to 0xf                                  3128000, probed ADC AD9361 as MASTER
cf_axi_dds 79024000.cf-ad9361-dds-core-lpc: Analog Devices CF_AXI_DDS_DDS MASTER                                   (9.01.b) at 0x79024000 mapped to 0xf3123000, probed DDS AD9361
random: crng init done
SAMPL CLK: 40000000 tuning: RX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:o o o O o o # # # # # # # # # #
1:o o o # # # # # # # # # # # # #
SAMPL CLK: 40000000 tuning: TX
  0:1:2:3:4:5:6:7:8:9:a:b:c:d:e:f:
0:# # # # # # # # # # # # # # # #
1:# # # # # # # # # # # # # # # #  Tuning TX Failed!

ruixingjin, 2021-07-26:
I don't know why last tx tuning failed. Thank you in advance.

closed 2021-07-26 by ruixingjin

ruixingjin, 2021-07-31:
Hello, Xianjun. could you help me?
I used adrv9364z7020 board, and I set an Zynq board as an AP.
Now, access point was enabled, and then i tried to connect to this AP using phone, but failed connecting.
I couldn't find needed wifi access point, because there was many wifi access point.
So i tested to check frequency(2422) with hackRF, but i couldn't see spectrum wave.
Now i couldn't check as if set frequency is correct.
So i am going to test with custom frequency.
In openwifi project, can i make custom frequency(500 ~ 600MHz)?

JiaoXianjun, 2021-07-31:
Please report new issue by our template.