open-sdr/openwifi #90: Build from scratch doesn't work

Eliad127, 2021-08-17:
0. The bug - When building image from scratch, wgd.sh fails (getting *.ko: Invalid module format)

1. I tried to build the image from scratch.

2. With my modification (removing Viterbi license) and without.

3. Versions
    - OS: Ubuntu 18.04.5 LTS
    - Vivado:  Vivado HLx 2018.3: WebPACK
    - openwif repo: commit 2742c7205ff6fbb6aaa28e55d4e54e6c72586ba1 (HEAD -> master, origin/master, origin/HEAD)

4. Board/hardware type - Xilinx zed board + FMCOMMS4 (zed_fmcs2)

5. WiFi channel number - Irrelevant

6. Steps to reproduce the issue
    - Run the "Build openwifi Linux img from scratch" process (stdout and stderr files attached)
    - On board: Run ./wgd.sh (output attached)
[logs.zip](https://github.com/open-sdr/openwifi/files/7001245/logs.zip)

JiaoXianjun, 2021-08-18:
I think maybe you don't need to build the whole image from scratch.

You only need to generate new FPGA image following openwifi-hw (with your viterbi), and then follow:

https://github.com/open-sdr/openwifi#update-fpga

Eliad127, 2021-08-18:
Tested it, and the results are different but still doesn't work.

When using SD card image (https://users.ugent.be/~xjiao/openwifi-1.2.0-leuven-2-32bit.img.xz) and then replacing the BOOT.BIN the following happens:
    -  When running wgd.sh the output says all is ok.
    - Hostapd process is running.
    - No SSID broadcast, means the AP is not really starting (tested with WiFi scanner to make sure).

Before changing the BOOT.BIN the AP was up and stable.
Same thing when using the original openwifi-hw project to build the BOOT.BIN

JiaoXianjun, 2021-08-18:
when issue is there:

1. cat /proc/interrupts
Execute the "cat ..." command for several times
You should see the number of interrupts of "sdr,tx_itrpt" (and related dma) grows, because it sends the "openwifi" beacon periodically)

2. can you also show me the demsg otuput?

JiaoXianjun, 2021-08-18:
While testing openwifi-hw master FPGA, could you also update onboard software to openwifi master by: https://github.com/open-sdr/openwifi#update-driver

Because the image might be not matched with the latest master.

Eliad127, 2021-08-19:
Sorry for the late response.

Looks like there is a problem with the update-driver part.

First test with the following steps:
    - Burn SD card with project image (https://users.ugent.be/~xjiao/openwifi-1.2.0-leuven-2-32bit.img.xz)
    - Generate new FPGA image, and replacing BOOT.BIN with new one (https://github.com/open-sdr/openwifi#update-fpga)
    - Power Cycle board
    - Updating onboard software drivers (https://github.com/open-sdr/openwifi#update-driver)
    - On board: Run fosdem-11ag.sh
    - Result, failed to start (output of process attached) 
[fosdem-11ag_output.txt](https://github.com/open-sdr/openwifi/files/7013014/fosdem-11ag_output.txt)

Second test with the following steps (like I have done before):
    - Burn SD card with project image (https://users.ugent.be/~xjiao/openwifi-1.2.0-leuven-2-32bit.img.xz)
    - Generate new FPGA image, and replacing BOOT.BIN with new one (https://github.com/open-sdr/openwifi#update-fpga)
    - Power Cycle board
    - On board: Run fosdem-11ag.sh
    - Result, no SSID broadcast, in /proc/interrupts sdr tx/rx are 0 (/proc/interrupts and demsg outputs attached as requested) 
[interrupts_output.txt](https://github.com/open-sdr/openwifi/files/7013009/interrupts_output.txt) 
[demsg_output.txt](https://github.com/open-sdr/openwifi/files/7013022/demsg_output.txt)

JiaoXianjun, 2021-08-19:
That error means that the rf front-end dac is not ready, so the tx chain is blocked.

Could you share me your FPGA image (without replacing viterbi, leave the fpga source code untouched)? You can pack the directory "openwifi-hw/boards/zed_fmcs2/sdk/" and attach it here, after you run (README of openwifi-hw):

In Vivado:
```
Open Block Design
Tools --> Report --> Report IP Status
Generate Bitstream
(Will take a while)
File --> Export --> Export Hardware... --> Include bitstream --> OK
File --> Launch SDK --> OK, then close SDK
```

In Linux:
```
cd openwifi-hw/boards
./sdk_update.sh board_name
```

Eliad127, 2021-08-19:
Attached  
[zed_fmcs2_sdk.zip](https://github.com/open-sdr/openwifi/files/7014311/zed_fmcs2_sdk.zip)

JiaoXianjun, 2021-08-19:
I compare your file with the file in master openwifi-hw, they are the same. So you didn't generate your own FPGA image? Have you run:

In Vivado:
```
Open Block Design
Tools --> Report --> Report IP Status
Generate Bitstream
(Will take a while)
File --> Export --> Export Hardware... --> Include bitstream --> OK
File --> Launch SDK --> OK, then close SDK
```

In Linux:
```
cd openwifi-hw/boards
./sdk_update.sh board_name
```

Eliad127, 2021-08-19:
Sorry, with all of this re-testing I forgot I have moved my own build and cloned again.

Attached my FPGA image with the original source code of openwifi-hw.

[zed_fmcs2_sdk.zip](https://github.com/open-sdr/openwifi/files/7015415/zed_fmcs2_sdk.zip)

JiaoXianjun, 2021-08-19:
I don't have any issue with your .zip. I do like this:

Put your system_top_hw_platform_0 into openwifi-my/openwifi-hw/boards/zed_fmcs2/sdk/ (replace the existing system_top_hw_platform_0)

Run in shell (form openwifi-my/user_space)
```
./boot_bin_gen.sh /directory-to/openwifi-my/ ~/Xilinx/ zed_fmcs2
```
(Use your own CORRECT directory!)

Now the BOOT.BIN is in openwifi-my/kernel_boot/boards/zed_fmcs2/output_boot_bin/

Run on computer:
```
cd  /directory-to/openwifi-my/kernel_boot/boards/zed_fmcs2/output_boot_bin/
scp ./BOOT.BIN root@192.168.10.122:
```
Run on board:
```
mount /dev/mmcblk0p1 /mnt
cp ~/BOOT.BIN /mnt
cd /mnt
sync
cd /
umount /mnt
reboot now
```
Power cycle the board, do the test and everything is OK.

Eliad127, 2021-08-22:
Ok, so when I run this steps it also works for me!
I'll try the same process with the modified Viterbi replacing code.

Still the build from scratch process doesn't work, but looks like there is a workaround..

JiaoXianjun, 2021-09-05:
Any news from your side?

Eliad127, 2021-09-06:
Had some bugs, think I'm almost done..
I will also test the upload/download speeds

JiaoXianjun, 2021-11-30:
If no further questions, I will close this issue.

closed 2022-04-01 by JiaoXianjun