open-sdr/openwifi #91: New SDR hardware platform support openwifi

black-pigeon, 2021-08-18:
1. We have released a new SDR platform called [ANTSDR-E310](https://github.com/MicroPhase/antsdr-fw). With a little modification on the openwifi project, the ANTSDR-E310 can support openwifi.

2. We forked the [openwifi](https://github.com/MicroPhase/openwifi/tree/antsdr) and [openwifi-hw](https://github.com/MicroPhase/openwifi-hw/tree/antsdr) project and create our new branch called antsdr. 

3. We think our SDR platform will be another choice for your project. We will be very pleased if you are interested in our hardware platform.

4. In the end, we would like to merge our branch of  [openwifi](https://github.com/MicroPhase/openwifi/tree/antsdr) and [openwifi-hw](https://github.com/MicroPhase/openwifi-hw/tree/antsdr) into your project.

JiaoXianjun, 2021-08-18:
Great!
Could you create pull requests towards our openwifi and openwifi-hw? Then we can review the changes and decide how to proceed.

JiaoXianjun, 2021-08-18:
I believe this is the purchase link for Chinese user (according to your github): https://item.taobao.com/item.htm?spm=a230r.1.14.16.34e21142YIlxqx&id=647986963313&ns=1&abbucket=2#detail

Seems like ~350USD. Very promising!

Do you have any plan to make it easier for user outside China to purchase?

helenleigh, 2021-08-18:
This is great! If you have any plans to bring a version of this to the US/EU please consider taking a look at Crowd Supply - we have a lot of open source SDR stuff! https://www.crowdsupply.com/

black-pigeon, 2021-08-19:
Sorry for the late reply, when I raised this issue, it was already night in China.  We are going to sell this product on AliExpress  for user outside China , and we are preparing for this now. 
Sure, I will  pull requests to your project later.

black-pigeon, 2021-08-19:
> This is great! If you have any plans to bring a version of this to the US/EU please consider taking a look at Crowd Supply - we have a lot of open source SDR stuff! https://www.crowdsupply.com/
We've looked at your site before, and it does have a lot of interesting projects. We think your website is also a good choice.

black-pigeon, 2021-08-19:
> I believe this is the purchase link for Chinese user (according to your github): https://item.taobao.com/item.htm?spm=a230r.1.14.16.34e21142YIlxqx&id=647986963313&ns=1&abbucket=2#detail
> 
> Seems like ~350USD. Very promising!
> 
> Do you have any plan to make it easier for user outside China to purchase?

 The product you mentioned is version AD9363, and it can also support openwifi, just declare the device as AD9361 in the device tree. In addition, we have another AD9361 version.
Although the official datasheet states that the AD9363 has a working range of 325M to 3.8GHz. But when I read the user manual, which is also called UG1040, it says ad9363 can work at  a full frequency of 70M ~ 6GHz.

For the RX datapath:
> The AD9363 LNA devices (1A, 1B, 1C, 2A, 2B, 2C) are
functional for the full 70 MHz to 6.0 GHz receive frequency
range. When operating at or below 3 GHz, any LNA input port
provides optimal performance. When operating above 3 GHz,
utilize the Rx1A and Rx2A LNA input ports for optimal performance.

For the Tx datapath:
>The AD9363 transmit path covers a full 70 MHz to 6.0 GHz
transmit frequency range and the two Tx outputs exhibit similar
performance. The Tx outputs are differential and require an
external output bias.

AllenWuhan, 2021-08-19:
It is a good news.  Can Open Wi-Fi  work over ANDSDR-E310, now?

black-pigeon, 2021-08-19:
Yes, we have successfully run the Openwifi project on ANTSDR.

screwer, 2021-08-20:
> The product you mentioned is version AD9363, and it can also support openwifi, just declare the device as AD9361 in the device tree. In addition, we have another AD9361 version.
> Although the official datasheet states that the AD9363 has a working range of 325M to 3.8GHz. But when I read the user manual, which is also called UG1040, it says ad9363 can work at a full frequency of 70M ~ 6GHz. You can find these statements easily in the UG1040.

i'm confused. Then which option is best to buy ? AD9361 much expensive, but you claim AD9363 have the same features...

black-pigeon, 2021-08-20:
> i'm confused. Then which option is best to buy ? AD9361 much expensive, but you claim AD9363 have the same features...

If you have used [ADI PlutoSDR](https://wiki.analog.com/university/tools/pluto/users/customizing#updating_to_the_ad9364) before, you may not have this confusion.
For individual users, the AD9363 can provide almost the same performance as the AD9361. For example, the frequency turange, maximum sampling rate and the maximum band width.

HC153Yuanyuan, 2021-08-21:
what about ANDSDR-B220 board ?

JiaoXianjun, 2021-09-02:
Today I received the HW and did a quick test. Everything seems OK.
Thanks for the quick HW support for the main line merging!

I have merged antsdr support to openwifi and openwifi-hw master.
Looking forward to your online sale solution for abroad users!

black-pigeon, 2021-09-03:
Nice to hear that.  I hope ANTSDR can meet your expectations.

By the way, the ANTSDR is available on AliExpress now.

Follow this link [ANTSDR on AliExpress](https://www.aliexpress.com/item/1005003181244737.html?spm=5261.ProductManageOnline.0.0.15744edfIp5RL0), users outside of China can easily get this product.

closed 2021-10-11 by JiaoXianjun

alexbvo, 2022-10-22:
Dear all, I am planning to use ANTSDR E310 with GNSS SDR software (gnss-sdr.org). Anyone is able to help or give an advice on this?